espressif/arduino-esp32 #397: analogRead not working on pins 27,28 (ESP32 Thing)

miguelgg, 2017-05-25:
Hi all

I'm trying to read pins 37 and 38 but always return 0. Other pins like 36, 39 are fully accesible.

Please, can you give some help?

Thanks in advance.

copercini, 2017-09-08:
These pins connects some 270pF capacitors which determinate the gain of internal Low Noise Amplifier, I think most boards don't expose GPIO37 and 38 for this reason.

[image]

Here you can see some details about it: https://espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf  session 3.1.6

http://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf  25.4 Low-Noise Amplifier

Akskaypohankar, 2017-10-26:
@miguelgg have you found any solution to read from this pins.  I am facing the same issue.

miguelgg, 2017-10-26:
@Akskaypohankar nop, sorry, if you finally can find it, share please.

Akskaypohankar, 2017-10-26:
i am currently reading about 

esp_err_t adc1_config_channel_atten(adc1_channel_t channel, adc_atten_t atten)
on 
https://esp-idf.readthedocs.io/en/v2.0/api/peripherals/adc.html

looks like something related let's see if it helps. 
 by the way, i found out that
gpio 37 and gpio 38 are not even declared as analog pin A1, A2  in esp32things

Akskaypohankar, 2017-10-28:
@miguelgg I found some solution to this problem try this code it is working for me on both gpio 37 and 38.
//
#include <driver/adc.h>
void setup()
{ Serial.begin(115200);
adc1_config_width(ADC_WIDTH_12Bit);
adc1_config_channel_atten(ADC1_CHANNEL_1, ADC_ATTEN_6db);//channel_1 for GPIO37 and 2 for 38.  
}
void loop() {
    int value = adc1_get_voltage(ADC1_CHANNEL_1);
    Serial.println(value);
    delay(1000);
}

let me know if you try it.

copercini, 2017-11-18:
Fixed on https://github.com/espressif/arduino-esp32/pull/822  =)

closed 2017-11-18 by copercini

berniv, 2019-03-10:
Hello, is it any update since 2017? I would like to read from a piezoresistive bridge sensor, such amplifier looks ideal for that (differential signal, low voltage around 50mV full scale, common mode near VCC/2).
I do not find any API ready to configure and read the ADC and LNA (Low Noise Amplifier). 
I see that ESP WROOM32 has this amplifier hardware ready. (image below)
[image]

pylorak, 2020-04-08:
> Hello, is it any update since 2017? I would like to read from a piezoresistive bridge sensor, such amplifier looks ideal for that (differential signal, low voltage around 50mV full scale, common [...]

Descriptions of the LNA and built-in temperature sensor have been removed from the documentation and have become unsupported features. I know this was done to the temperature sensor because it was unusable in practice, and Espressif decided to remove the feature instead of fixing it in silicon. I can only guess the LNA suffered from a similar fate.

berniv, 2020-04-09:
Hi Pylorak, thanks for the update! Then the only way is using an external converter. Personnaly I added HX711, a good chip for low cost.