espressif/arduino-esp32 #399: ESP32 - doesn't start

smily77, 2017-05-26:
The ordinary blink example  doesn't run on one of my ESP32 (thing) but runs on the other !?! The problems appears to be the same as in #346 but there isn't any mistake in the sketch otherwise it wouldn't run on the other ESP32.

I use the Arduino IDE with the standard Bilnk example make pin 5 to blink:

`void setup() {
  pinMode(5, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(5, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(5, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
}`

I'm getting the message on the serial: 
`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:2016
load:0x40078000,len:7780
ho 0 tail 12 room 4
load:0x40080000,len:252
entry 0x40080034`

but the LED doesn't blink. Then I connect the other ESP32. Same code, same output and the LED blinks.

The other differences: the arduino IDE takes the first ESP32 (which has some other electronics attached) on COM 6: while it has the 2nd ESP32 on Com 9

Of cores this doesn't only applies to the blink example. So it isn't a matter of the LED.

lonerzzz, 2017-08-25:
@smily did you manage to resolve the issue? It sounds like a hardware problem.

copercini, 2017-10-06:
Closed due no response =/

closed 2017-10-06 by copercini