espressif/arduino-esp32 #402: SD card library

lutfisan, 2017-05-28:
I try SD_Test example,
when I want to write string to sd card it become error
`void loop(){
   // make a string for assembling the data to log:
  String dataString = "";

  // read three sensors and append to the string:
  for (int analogPin = 0; analogPin < 3; analogPin++) {
    int sensor = analogRead(analogPin);
    dataString += String(sensor);
    if (analogPin < 2) {
      dataString += ",";
    }
  }

  Serial.println(dataString);
  writeFile(SD, "/log.txt", dataString);
}`

jestebanes, 2017-05-29:
Hi, without reading the error message, I think that you problem could be the way you are passing the argument dataString, becouse the writeFile funtion signature is:

void writeFile(fs::FS &fs, const char * path, const char * message)

and your third argument is a String not a const char*, try

 writeFile(SD, "/log.txt", dataString.c_str());

Bye

lutfisan, 2017-05-30:
thanks it's works

but when i try this code
```
#include "FS.h"
#include "SD.h"
#include "SPI.h"

void setup(){
    Serial.begin(115200);
    if(!SD.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD card attached");
        return;
    }

    Serial.print("SD Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);

}

void loop(){
  // make a string for assembling the data to log:
  String dataString = "";

  // read three sensors and append to the string:
  for (int analogPin = 0; analogPin < 3; analogPin++) {
    int sensor = analogRead(analogPin);
    dataString += String(sensor);
    if (analogPin < 2) {
      dataString += ",";
    }
  }
  
  File file = fs.open("/log.txt", FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }
    if(file.println(dataString)){
        Serial.println("File written");
    } else {
        Serial.println("Write failed");
    }

}
```
[image]

me-no-dev, 2017-05-30:
SD.open() not fs.open()

lutfisan, 2017-05-31:
I have tried use SD.open() and dataFile.close() in code above but file cannot open

```
File dataFile = SD.open("log.txt", FILE_WRITE);

  // if the file is available, write to it:
  if (dataFile) {
    dataFile.println(dataString);
    dataFile.close();
    // print to the serial port too:
    Serial.println(dataString);
  }
  // if the file isn't open, pop up an error:
  else {
    Serial.println("error opening datalog.txt");
  }
```

copercini, 2017-10-06:
@lutfisan try with `File dataFile = SD.open("/log.txt", FILE_WRITE);` =)

copercini, 2017-10-18:
@lutfisan is it working now?

copercini, 2017-10-31:
Closed due no response =/
Can be reopened if the OP returns to live =)

closed 2017-10-31 by copercini