espressif/arduino-esp32 #405: SimpleBLE.cpp:19:21: fatal error: bta_api.h: No such file or directory

ledav-net, 2017-05-30:
Hi,

I just updated the IDF and Arduino components from git and had this error when compiling my code:

<project>/components/arduino/libraries/SimpleBLE/src/SimpleBLE.cpp:19:21: fatal error: bta_api.h: No such file or directory

I don't need the Bluetooth so it was disabled in menuconfig (Component config --> Bluetooth). However when enabled, the compilation error is gone... As a work around, it's ok... But how to get rid of it without including all the bluetooth stuff ?

Thanks for your feedback.

Regards,
David.

lacyrhoades, 2017-07-26:
I noticed this, too.

Following the "Using as ESP-IDF component" steps in the README leads to the error `fatal error: bta_api.h`

macOS 10.12 + LOLIN32 + very basic project modified from `hello_world`

Enabling bluetooth with `make menuconfig` did seem to do the trick, allowing compilation.

Although it makes no difference to me at this point if BT is enabled (I'm not trying to save space or anything) that step is NOT necessary before installing the Arduino component.

Here's the `sdkconfig` diffs, working and not working:

## sdkconfig.bad vs sdkconfig.good

```
$ diff -u sdkconfig.bad sdkconfig.good
--- sdkconfig.bad   2017-07-25 22:58:13.000000000 -0400
+++ sdkconfig.good  2017-07-25 23:00:19.000000000 -0400
@@ -126,8 +126,20 @@
 CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL=1
 # CONFIG_ARDUHAL_LOG_COLORS is not set
 # CONFIG_AWS_IOT_SDK is not set
-# CONFIG_BT_ENABLED is not set
-CONFIG_BT_RESERVE_DRAM=0
+CONFIG_BT_ENABLED=y
+CONFIG_BLUEDROID_ENABLED=y
+CONFIG_BTC_TASK_STACK_SIZE=3072
+# CONFIG_BLUEDROID_MEM_DEBUG is not set
+# CONFIG_CLASSIC_BT_ENABLED is not set
+# CONFIG_BT_DRAM_RELEASE is not set
+CONFIG_GATTS_ENABLE=y
+CONFIG_GATTC_ENABLE=y
+CONFIG_BLE_SMP_ENABLE=y
+# CONFIG_BT_STACK_NO_LOG is not set
+CONFIG_BT_ACL_CONNECTIONS=4
+CONFIG_BTDM_CONTROLLER_RUN_CPU=0
+CONFIG_SMP_ENABLE=y
+CONFIG_BT_RESERVE_DRAM=0x10000
 
 #
 # ESP32-specific
@@ -197,6 +209,7 @@
 CONFIG_ESP32_XTAL_FREQ_AUTO=y
 CONFIG_ESP32_XTAL_FREQ=0
 CONFIG_WIFI_ENABLED=y
+# CONFIG_SW_COEXIST_ENABLE is not set
 CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=10
 CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32
 # CONFIG_ESP32_WIFI_STATIC_TX_BUFFER is not set
```

## Bad (fatal error: bta_api.h after install steps)

```diff
diff --git a/sdkconfig b/sdkconfig
index bfb289f..e7a06c1 100644
--- a/sdkconfig
+++ b/sdkconfig
@@ -106,6 +106,25 @@ CONFIG_ESP32_APPTRACE_DEST_NONE=y
 #
 # FreeRTOS SystemView Tracing
 #
+
+#
+# Arduino Configuration
+#
+CONFIG_ENABLE_ARDUINO_DEPENDS=y
+# CONFIG_AUTOSTART_ARDUINO is not set
+# CONFIG_DISABLE_HAL_LOCKS is not set
+
+#
+# Debug Log Configuration
+#
+# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_NONE is not set
+CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_ERROR=y
+# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_WARN is not set
+# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_INFO is not set
+# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_DEBUG is not set
+# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_VERBOSE is not set
+CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL=1
+# CONFIG_ARDUHAL_LOG_COLORS is not set
 # CONFIG_AWS_IOT_SDK is not set
 # CONFIG_BT_ENABLED is not set
 CONFIG_BT_RESERVE_DRAM=0
@@ -195,7 +214,11 @@ CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE=y
 # CONFIG_ESP32_PHY_INIT_DATA_IN_PARTITION is not set
 CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER=20
 CONFIG_ESP32_PHY_MAX_TX_POWER=20
-# CONFIG_ETHERNET is not set
+CONFIG_ETHERNET=y
+CONFIG_DMA_RX_BUF_NUM=10
+CONFIG_DMA_TX_BUF_NUM=10
+# CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE is not set
+CONFIG_EMAC_TASK_PRIORITY=20
 
 #
 # FAT Filesystem support
@@ -272,7 +295,7 @@ CONFIG_LOG_COLORS=y
 CONFIG_LWIP_MAX_SOCKETS=10
 CONFIG_LWIP_THREAD_LOCAL_STORAGE_INDEX=0
 # CONFIG_LWIP_SO_REUSE is not set
-# CONFIG_LWIP_SO_RCVBUF is not set
+CONFIG_LWIP_SO_RCVBUF=y
 CONFIG_LWIP_DHCP_MAX_NTP_SERVERS=1
 # CONFIG_LWIP_IP_FRAG is not set
 # CONFIG_LWIP_IP_REASSEMBLY is not set
```

## Good (compiles and uploads)

```diff
diff --git a/sdkconfig b/sdkconfig
index bfb289f..b57ac47 100644
--- a/sdkconfig
+++ b/sdkconfig
@@ -106,9 +106,40 @@ CONFIG_ESP32_APPTRACE_DEST_NONE=y
 #
 # FreeRTOS SystemView Tracing
 #
+
+#
+# Arduino Configuration
+#
+CONFIG_ENABLE_ARDUINO_DEPENDS=y
+# CONFIG_AUTOSTART_ARDUINO is not set
+# CONFIG_DISABLE_HAL_LOCKS is not set
+
+#
+# Debug Log Configuration
+#
+# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_NONE is not set
+CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_ERROR=y
+# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_WARN is not set
+# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_INFO is not set
+# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_DEBUG is not set
+# CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_VERBOSE is not set
+CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL=1
+# CONFIG_ARDUHAL_LOG_COLORS is not set
 # CONFIG_AWS_IOT_SDK is not set
-# CONFIG_BT_ENABLED is not set
-CONFIG_BT_RESERVE_DRAM=0
+CONFIG_BT_ENABLED=y
+CONFIG_BLUEDROID_ENABLED=y
+CONFIG_BTC_TASK_STACK_SIZE=3072
+# CONFIG_BLUEDROID_MEM_DEBUG is not set
+# CONFIG_CLASSIC_BT_ENABLED is not set
+# CONFIG_BT_DRAM_RELEASE is not set
+CONFIG_GATTS_ENABLE=y
+CONFIG_GATTC_ENABLE=y
+CONFIG_BLE_SMP_ENABLE=y
+# CONFIG_BT_STACK_NO_LOG is not set
+CONFIG_BT_ACL_CONNECTIONS=4
+CONFIG_BTDM_CONTROLLER_RUN_CPU=0
+CONFIG_SMP_ENABLE=y
+CONFIG_BT_RESERVE_DRAM=0x10000
 
 #
 # ESP32-specific
@@ -178,6 +209,7 @@ CONFIG_ESP32_DEEP_SLEEP_WAKEUP_DELAY=2000
 CONFIG_ESP32_XTAL_FREQ_AUTO=y
 CONFIG_ESP32_XTAL_FREQ=0
 CONFIG_WIFI_ENABLED=y
+# CONFIG_SW_COEXIST_ENABLE is not set
 CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=10
 CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=32
 # CONFIG_ESP32_WIFI_STATIC_TX_BUFFER is not set
@@ -195,7 +227,11 @@ CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE=y
 # CONFIG_ESP32_PHY_INIT_DATA_IN_PARTITION is not set
 CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER=20
 CONFIG_ESP32_PHY_MAX_TX_POWER=20
-# CONFIG_ETHERNET is not set
+CONFIG_ETHERNET=y
+CONFIG_DMA_RX_BUF_NUM=10
+CONFIG_DMA_TX_BUF_NUM=10
+# CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE is not set
+CONFIG_EMAC_TASK_PRIORITY=20
 
 #
 # FAT Filesystem support
@@ -272,7 +308,7 @@ CONFIG_LOG_COLORS=y
 CONFIG_LWIP_MAX_SOCKETS=10
 CONFIG_LWIP_THREAD_LOCAL_STORAGE_INDEX=0
 # CONFIG_LWIP_SO_REUSE is not set
-# CONFIG_LWIP_SO_RCVBUF is not set
+CONFIG_LWIP_SO_RCVBUF=y
 CONFIG_LWIP_DHCP_MAX_NTP_SERVERS=1
 # CONFIG_LWIP_IP_FRAG is not set
 # CONFIG_LWIP_IP_REASSEMBLY is not set
```

copercini, 2017-09-22:
Fixed here: https://github.com/espressif/arduino-esp32/commit/c66a020e8067f657e730d018700e7de5b23175d4

=)

closed 2017-09-22 by copercini