espressif/arduino-esp32 #406: analog to digital not correct

hsgentry, 2017-05-31:
I have been using the Arduino IDE and ESP32 analog to digital for a while and either it does not return the correct digital value or I am missing something.  The digital value is non linear and lower than expected.  The non linear and scale errors are different for each of the attenuation settings (0,2.5,6,11).

I did a piecewise linear interpolation and was able to get each attenuation correct.  However, this only works on A0,A3,A4,A5,A6,A7 (ADC1).  A10-A19 (ADC2) does not work at all  (always returns 4095).

I am reading the values with:
x=analogRead(A0);

Is there anything else that needs to be done to make the ADC2 lines work?  Do I need to do something else to make the data on the ADC1 set correct?

me-no-dev, 2017-05-31:
see https://github.com/espressif/arduino-esp32/issues/92#issuecomment-266648072

hsgentry, 2017-05-31:
I am not sure what I was smoking when I tested this, but there is no issue unique to ADC2.  I must have been counting pins incorrectly.  All analog ports work, just non linear and scale error.  

I see that this is marked as a duplicate.  I looked but did not see another related issue before posting.  Maybe this is just not my day!!

me-no-dev, 2017-05-31:
I marked it as duplicate because of the non-linearity issues. The other issues you already found :)

hsgentry, 2017-05-31:
Thanks, I found the dup this time. Don't know how I missed it first time.

copercini, 2017-09-16:
Let's concentrate this issue in https://github.com/espressif/arduino-esp32/issues/92 to keep the things organized =)

closed 2017-09-16 by copercini