espressif/arduino-esp32 #407: Serial1 or Serial2 in external library reference

hicotton02, 2017-05-31:
This very well could be from lack of programming skills, but I am having an issue. In another Arduino Library, there is reference to Serial2

#define mySerial Serial2 

but in the esp32 code, there is no reference to such serial. if, in my sketch, I declare

HardwareSerial Serial2(2);

I get this error:

error: 'Serial2' was not declared in this scope (from their h and cpp files)

if I try to manually declare it in their library, I either get the same error, or "redefinition of Serial2". They have this:

 #include "Arduino.h"

should I change that to something else? Basically my question is "How do deal with external libraries that reference another serial port (either hardware or software) other than "Serial"?

jestebanes, 2017-05-31:
I can't see your code but Serial2 works for me

in the cpp class add as a global variable

#include "HardwareSerial.h"
HardwareSerial Serial2(2);

and then you can use as a serial port in Arduino way, for instance

void Myclass::setSerial(int speed){
   Serial2.begin(speed, SERIAL_8N1);
]

hicotton02, 2017-05-31:
Here is the library in question:

https://github.com/itead/ITEADLIB_Arduino_Nextion

in the NexConfig.h, there is this:

#define nexSerial Serial2

in that config file, it references "NexHardware.h" and in NexHardware.cpp I added the bits above:

`#include "NexHardware.h"
#include "HardwareSerial.h"
HardwareSerial Serial2(2);

#define NEX_RET_CMD_FINISHED            (0x01)
#define NEX_RET_EVENT_LAUNCHED          (0x88)`

and in my sketch, I have within setup():

`void setup() { 
  Serial.begin(115200);
  Serial2.begin(115200);
  //Serial.flush();
  //Serial2.flush();
}
  `

the IDE fails with:
Main:5: error: 'Serial2' was not declared in this scope

me-no-dev, 2017-05-31:
Here is what you can do: add to [HardwareSerial.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.h#L78) the following line:
```Arduino
extern HardwareSerial Serial1;
extern HardwareSerial Serial2;
```
Then keep ```HardwareSerial Serial2(2);``` in your sketch.
No need to ```#include "HardwareSerial.h"``` in your sketch also.

hicotton02, 2017-05-31:
ok that worked!

i tried adding extern HardwareSerial Serial2; to HardwareSerial.h and it didnt work. i did not know I needed to also add it in my sketch, I thought it was one or the other. Thank you so much!

me-no-dev, 2017-05-31:
The one in ```HardwareSerial.h``` only declares that somewhere Serial2 will be implemented and with adding it to your sketch, you actually implement/instantiate the class.

hicotton02, 2017-05-31:
Gotcha!

I appreciate all the effort you guys are putting in on this.

closed 2017-05-31 by me-no-dev

tablatronix, 2018-03-06:
This throws an error for me

tablatronix, 2018-03-06:
ohh you have to add serial1 support?

tablatronix, 2018-03-06:
nm, #47 

`HardwareSerial Serial1(2);`

JacoFourie, 2018-03-06:
I am using all 3 UARTS at the same time.

````
HardwareSerial MySerial(1);
HardwareSerial ModbusSerial(2);

//In the set-up function
Serial.begin(115200);
MySerial.begin(115200, SERIAL_8N1, 16, 17);
ModbusSerial.begin(9600 ,SERIAL_8E1, 4, 15);

````

maxim25111, 2018-05-14:
I have solution for stm32F1 communicating with nextion display. This is extremely difficult to find that solution. But it is extrimly easy to set. 
I was trying many configurations from 2 days. Until I opened this file and try change Serial to Serial2
Find in folder NexConfig.h and open it.
```
/** 
 * Define DEBUG_SERIAL_ENABLE to enable debug serial. 
 * Comment it to disable debug serial. 
 */
//#define DEBUG_SERIAL_ENABLE //    <-- coment that

/**
 * Define dbSerial for the output of debug messages. 
 */
//#define dbSerial Serial      // <--- Coment that.

/**
 * Define nexSerial for communicate with Nextion touch panel. 
 */
#define nexSerial Serial2 // <--- Set hire from Serial to Serial2 for stm32 pins are RX2 = PA3 TX2 = PA2
                          // <--- basic setings for arduino is SoftwareSerial... and redirect SoftwareSerial to digital 
// Serial (Serial1) is comunicating with PC USB and will make conflict with Nextion display.
//pins other than RX and TX.
// Arduino Uno have only one Serial port. 

```

> #define nexSerial Serial2 // <--- Set hire from Serial to Serial2 for stm32 pins are RX2 = PA3 TX2 = PA2
>                           // <--- basic setings for arduino is SoftwareSerial... and redirect SoftwareSerial to digital 
> // Serial (Serial1) is comunicating with PC USB and will make conflict with Nextion display.
> //pins other than RX and TX.

On Your file set Serial.begin() and Serial2.begin().
I am proposing only short example in *.INO file in void setup(); Pins are from STM32F103.....
```
void setup(void) {
  dht.begin();   // <-- This is my temperature & humidity sensor
  //pinMode(rxPin, INPUT);  // <-- This line is only needed if is using Arduino Uno SoftwareSerial
  //pinMode(txPin, OUTPUT);   // <-- This line is only needed if is using Arduino Uno  SoftwareSerial
  Serial2.begin(9600); // For stm32 pin PA3-RX2 PA2-TX2 In nexConfig.h "#define nexSerial Serial2"
  Serial.begin(9600); // For PC Serial conection = TX1 RX1 - PB6 PB7 and PA9 PA10
  //myserial.begin(9600);  // <--- If You use Arduino Uno to simulate Serial2
  // You might need to change NexConfig.h file in your ITEADLIB_Arduino_Nextion folder
  nexInit();
```

SteveAmor, 2018-07-29:
I had the same error with Serial2(2) so I named it Serial3.  ```HardwareSerial Serial3(2)```  in the Arduino IDE worked for me.

jestebanes, 2018-07-29:
After the last reviews of the library, they create the Serial2 and Serial1 objects, you only need to remove the declaration of
HardwareSerial Seiral2(2)
Because it's already made
Regards

SpoturDeal, 2018-10-27:
With using the standard Nextion library version 0.7.0

First in the Nextion Editor
Set a preinitialise event in user code 

`bauds=115200`

I made some changes in the library
line 31 in nexHardware.h

Before
`bool nexInit();`

Updated
`bool nexInit(int Baud = 115200, int Rx = 16, int Tx = 17);`

Then in nexHardware.cpp adapted function nexInt() starting at line 224

Before

```
bool nexInit()
{
    bool ret1 = false;
    bool ret2 = false;
	
    
    dbSerialBegin(9600);
    nexSerial.begin(9600);
    sendCommand("");
    sendCommand("bkcmd=1");
    ret1 = recvRetCommandFinished();
    sendCommand("page 0");
    ret2 = recvRetCommandFinished();
    return ret1 && ret2;
}
```

Updated 
```
bool nexInit(int Baud, int Rx, int Tx )
{
    bool ret1 = false;
    bool ret2 = false;
	
    dbSerialBegin(9600);
    nexSerial.begin(Baud, SERIAL_8N1 ,Rx, Tx);
    sendCommand("");
    sendCommand("bkcmd=1");
    ret1 = recvRetCommandFinished();
    sendCommand("page 0");
    ret2 = recvRetCommandFinished();
    return ret1 && ret2;
}
```

In the sketch you **remove** HardwareSerial Serial2(2);  (is not needed)
And change the nexInit(115200,16,17);

Now you can set the speed and select the pins on the ESP32