espressif/arduino-esp32 #408: Using as ESP-IDF component : how to create blank project ?

picpic020960, 2017-05-31:
Bonjour ,
After do what i read in "using as ESP-IDF component":the make flash is ko as (see below)
Program is the "loop" example as main.cpp
What is my error ? Noblank project ?
Thanks for help
Regards
-------------------------------------------
$ make flash
CC cores/esp32/esp32-hal-timer.o
D:/ESP32/test/components/arduino/cores/esp32/esp32-hal-timer.c: In funct
D:/ESP32/test/components/arduino/cores/esp32/esp32-hal-timer.c:201:9: eron of function 'DPORT_SET_PERI_REG_MASK' [-Werror=implicit-function-decl
         DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_TIMERGROU
         ^
D:/ESP32/test/components/arduino/cores/esp32/esp32-hal-timer.c:202:9: eron of function 'DPORT_CLEAR_PERI_REG_MASK' [-Werror=implicit-function-de
         DPORT_CLEAR_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_TIMERGR
         ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/d/ESP32/esp32_win32_msys2_environment_and_toolchain-20170ent_wrapper.mk:177: cores/esp32/esp32-hal-timer.o] Error 1
make: *** [D:/ESP32/esp32_win32_msys2_environment_and_toolchain-20170330k:386: arduino-build] Error 2

me-no-dev, 2017-05-31:
Old version of ESP-IDF. Update your IDF

picpic020960, 2017-06-01:
Thanks for reply
install made with https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170330.zip
(all include in one zip = easy for newbie)
how to update ?
thanks

picpic020960, 2017-06-03:
bonjour
ok fine after reload ESP-IDF
but heavy !
thanks

closed 2017-06-03 by picpic020960