espressif/arduino-esp32 #409: Uploading stub... Timed out waiting for packet header

opichals, 2017-05-31:
I am trying to flash this board:
https://www.aliexpress.com/item/ESP-32S-Development-Board-Esp32-Module-2-4-GHz-dual-mode-Wi-Fi-and-Bluetooth-chips/32803195774.html

The esptool has the following trouble:
```
Arduino: 1.8.2 (Mac OS X), Board: "ESP32 Dev Module, 40MHz, 921600, Verbose"

Archiving built core (caching) in: /var/folders/kd/dms1b9n94m3_5fn8rg8s_k7c0000gn/T/arduino_cache_973448/core/core_espressif_esp32_esp32_FlashFreq_40,UploadSpeed_921600,DebugLevel_verbose_284d0d6d1a9f5d714c68e4f94b1da0ff.a
Sketch uses 120768 bytes (9%) of program storage space. Maximum is 1310720 bytes.
Global variables use 9332 bytes (3%) of dynamic memory, leaving 285580 bytes for local variables. Maximum is 294912 bytes.
/Users/standa/Documents/Arduino/hardware/espressif/esp32/tools/esptool --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_freq 40m --flash_mode dio --flash_size 4MB 0x1000 /Users/standa/Documents/Arduino/hardware/espressif/esp32/tools/sdk/bin/bootloader.bin 0x8000 /var/folders/kd/dms1b9n94m3_5fn8rg8s_k7c0000gn/T/arduino_build_102618/BareMinimum.ino.partitions.bin 0xe000 /Users/standa/Documents/Arduino/hardware/espressif/esp32/tools/partitions/boot_app0.bin 0x10000 /var/folders/kd/dms1b9n94m3_5fn8rg8s_k7c0000gn/T/arduino_build_102618/BareMinimum.ino.bin 
esptool.py v2.0-beta3
Connecting....
Uploading stub...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header
```

I have tried different baud rates as well as flash_freq changes to no help.

opichals, 2017-05-31:
The effect appears very similar to what @architmuchhal12 reports https://github.com/espressif/arduino-esp32/issues/333#issuecomment-303515054

I have also installed the CP2102 drivers from http://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

opichals, 2017-05-31:
Eh, just found out that the flash goes through ok if I pull the board from the breadboard and just connect it to the USB port.

closed 2017-05-31 by opichals

opichals, 2017-06-01:
Double checked the breadboard and it doesn't seem to have any issues while I have the board in (no shorts or any other noticeable obvious stuff). It still gives the same trouble.

projectgus, 2017-06-02:
Breadboards are often a source of high capacitance (it may not be obvious with a multimeter, but it causes problems for high speed signals on the TX/RX pins.) If you lower the baud rate to 9600bps while connected to the breadboard, does flashing succeed?

opichals, 2017-06-02:
Nope, tried that.

sandman72, 2017-06-05:
OK, I can verify this behavior. I cannot program the two unmodified DOIT ESP32 dev boards I have here if they are plugged into a breadboard. The type of breadboard doesn't matter. Unplugging the dev board and then programming it works fine.
My current solution is to clip off pins TX0 and RX0 so that they do not connect to the breadboard. With this modification the boards flash fine while being plugged in.

zorglub76, 2017-06-07:
The same happened to me today. DOIT ESP32 DEVKIT V1 board. I was working normally, flashing several times during the day, and then at some point flashing stopped with the error above.
I was using Eclipse on Win10.
When the board was removed from breadboard, everything worked fine.

hs06, 2018-01-04:
Have you tried flashing with `--no-stub`? This helped for me.

projectgus, 2018-01-04:
Hi @hs06 ,

> Have you tried flashing with --no-stub? This helped for me

What hardware & baud rate did you have where `--no-stub` was required?

Angus

hs06, 2018-01-04:
A Sonoff basic with ESP8266EX. The baudrate did not matter - tried several. The USB/serial is a standard FTDI adapter.

projectgus, 2018-01-04:
@hs06 Ah, OK. Please consider opening an Issue at https://github.com/espressif/esptool/issues/new with the command line log output with & without --no-stub and any other details you think are relevant. It would be good to track down whatever the difference is.

ilxj, 2019-06-02:
Hello  everyone,
I meet the trouble ，I had a ESP-WROVER-KIT ,and I work in macOS system.when I run the cmd ,`make flash` to upload my firmware ，it's fail ;
BUT ,the same board I make in windows7 it work well.

the follow is the error log in macOS ,waiting for help ...;

```
WARNING: Toolchain version is not supported: 1.22.0-75-gbaf03c2
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: 1.22.0-75-gbaf03c2
Expected to see version: 1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Flashing binaries to serial port /dev/cu.usbserial-00002014B (app at offset 0x10000 )...
esptool.py v2.6
Serial port /dev/cu.usbserial-00002014B
Connecting.....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, BLK3 partially reserved, Coding Scheme 3/4
MAC: 30:ae:a4:c9:f8:b0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 21248 bytes to 12534...
Writing at 0x00001000... (100 %)
A fatal error occurred: Timed out waiting for packet header
make: *** [flash] Error 2
```

kbickham, 2019-07-31:
I  too am experiencing this problem. None of the solutions worked.