espressif/arduino-esp32 #410: Aynchronous I2C support

lonerzzz, 2017-06-01:
I noticed that there is a work item on the NodeMCU code for the ESP32 to support asynchronous I2C communication (https://github.com/nodemcu/nodemcu-firmware/issues/1663). Is anyone familiar with such an implementation on any Arduino hardware in the past? Given the connected nature of this chips, it seems to make sense that such a capability would be present to avoid having the MCU sitting waiting for bits.

I am not familiar with the underlying code in the Wire library but will take a look.

lonerzzz, 2017-06-01:
Actually, just found that someone was already doing this for the ESP8266. My question is whether it makes sense to extend the Wire library for the ESP32 to incorporate this?

https://github.com/cskarai/asynci2cmaster

me-no-dev, 2017-06-01:
You can run async but you need to alter the driver. Arduino and the Wire library are not meant to work Async so the driver is not written in that way.

lonerzzz, 2017-06-01:
I am aware that the Wire library does not currently support asynchronous behavior, hence why I created this item. The question is whether the addition of the functionality makes sense. I would argue it does on chips with the processing capability so that they are not sitting idly waiting for a response. The question is whether a new library has to be created or whether Wire can be extended with the functionality. When you say 'driver', what are you referring to, the underlying TwoWire class? Lastly, when you say that the Arduino is not meant to work Async, are you referring to some standard or just commonly how the implementations have worked? I looked but did not find anything to reference.

me-no-dev, 2017-06-01:
For chips that can do so much more, you have threads ;) any time you delay, you allow other threads to run and do other things. Making Arduino Wire support Async is something that will bring more issues that else. You need to watch out for mutexes and semaphores and whatnot to ensure multicore operability.
Surely you are more than welcome to create an Async "Wire" library, I have such network libs for ESP8266, but that is not the general use.

lonerzzz, 2017-06-01:
Understood threading is a possibility but it is again not a normal Arduino usage pattern from what I have read so was thinking to use an interrupt model without full blown threads. I guess the question now comes down to whether extending the i2c hal code for async support would fit with the intended architecture to allow creation of a parallel Async "Wire" library. I am familiar with embedded development just not the Arduino internals. Any thoughts on that?

kriswiner, 2017-06-04:
We have an async Wire.h library for the STM32L4 Arduino core. See: https://github.com/GrumpyOldPizza/arduino-STM32L4/tree/master/libraries/Wire

Not sure how this will help you with the ESP32, but asynch I2C is very useful.

lonerzzz, 2017-06-04:
@kriswiner Thanks for that. I will take a look.

stickbreaker, 2019-05-19:
@lonerzzz Are you still interested in this feature? how would you use it?

Chuck.

lonerzzz, 2019-08-01:
@stickbreaker Sorry for the delay. I have not been on the board for a while and was not getting notifications. I have worked around the issue at this stage with core threading so don't think that this functionality is a need. I would have used it for a send and forget model of polling I2C devices with a callback in the event that the device had something to report.

closed 2019-08-01 by lonerzzz

michox, 2021-02-06:
hi @lonerzzz can you describe a little bit more how you worked around this issue? I would like to do something similar

atlaste, 2022-08-10:
I just noticed this thread because ASync I2C would be useful for me as well.

I've browsed through the async I2C implementation that was mentioned above. It doesn't use the hardware facilities, and basically does I2C by bit banging. While that might work for some, it's definitely no good enough for me. So I've dug a bit deeper...

It would be really, really good if async I2C would be built into the framework. Without it, we would have to copy-paste the I2C implementation and everything below it right into the HAL. If it's in the framework, the change would be nearly trivial.

Basically, the code will eventually go to I2C.c (https://github.com/espressif/esp-idf/blob/master/components/driver/i2c.c), registers an ISR and then run a loop. In short, an ISR will throw things on a queue, and the loop will pick it off. Making it async is actually very easy - it's just a matter of making the queue a bit bigger and splitting the loop up in 2 functions. The loop itself in I2C.c looks like this:

```c++
    i2c_cmd_evt_t evt;
    while (1) {
        TickType_t wait_time = xTaskGetTickCount();
        if (wait_time - ticks_start > ticks_to_wait) { // out of time
            wait_time = I2C_CMD_ALIVE_INTERVAL_TICK;
        } else {
            wait_time = ticks_to_wait - (wait_time - ticks_start);
            if (wait_time < I2C_CMD_ALIVE_INTERVAL_TICK) {
                wait_time = I2C_CMD_ALIVE_INTERVAL_TICK;
            }
        }
        ...
```

Anyways, I hope it gets picked up...

ozdemirsalih, 2024-10-04:
An async i2c is a hundred percent necessary since there're async web server libraries with which if you use the "Wire" library it causes unintended behavior. Especially if you're developing a websocket server and you need an i2c device connected to the ESP8266 or ESP32, you'll see that some pretty interesting behavior among GPIOs start to happen. For now the only solution is to use a microcontroller with a hardware i2c capability and interface with that device thorough other means(SPI, UART etc...). I'll try to work on it if I can.