espressif/arduino-esp32 #413: WiFi CPU halted

bbasil2012, 2017-06-02:
Arduini IDE 1.8.2 OSX /fresh install and fresh git clone/
Every examples with WiFi causes CPU halted:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
E (915) wifi: wifi_init 1333 ret=4357
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:715 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008418b on core 1
Guru Meditation Error: Core  1 panic'ed (abort)

Backtrace: 0x40008155:0x3ffce350 0x40007d16:0x3ffce370 0x40082484:0x3ffce3b0 0x400f6248:0x3ffce400 0x400ef584:0x3ffce420 0x400d12a8:0x3ffce450 0x400d12b6:0x3ffce4a0 0x400d12ce:0x3ffce4d0 0x400d133c:0x3ffce4f0 0x400d1397:0x3ffce510 0x400d0ad8:0x3ffce610 0x40113f9a:0x3ffce640

CPU halted.

bbasil2012, 2017-06-02:
The same on Win7(64) fresh installation :(

lowcid, 2017-06-02:
I might have the same issue. Also using arduino core and 1.8.2. Using NodeMCU-ESP32S board with an extra 5v external power supply.
For me CPU halts after/at the WiFi.begin(ssid,pass); command. 

```
Connecting to WiFi network: My wifi
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x40080cf8  PS      : 0x00060d30  A0      : 0x800d0ef0  A1      : 0x3ffc78f0  
A2      : 0x00000007  A3      : 0x00000001  A4      : 0x00000013  A5      : 0x00000017  
A6      : 0x000000ff  A7      : 0xff000000  A8      : 0x3f400ed4  A9      : 0x0009a800  
A10     : 0x0009a800  A11     : 0x00000064  A12     : 0x00000040  A13     : 0x00000000  
A14     : 0x3ffc11a0  A15     : 0x00000006  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0009a800  LBEG    : 0x400d1b34  LEND    : 0x400d1b96  LCOUNT  : 0x00000002  

Backtrace: 0x40080cf8:0x3ffc78f0 0x400d0ef0:0x3ffc7910 0x400d1340:0x3ffc7930 0x400d08d0:0x3ffc7960 0x400d06b2:0x3ffc7980 0x400d073c:0x3ffc79c0 0x400d9272:0x3ffc79e0

CPU halted.
```

lowcid, 2017-06-02:
I just found out what the issue was on my side.
Turns out arduino chose the standard wifi lib and not this one. This was because i didn't follow the installation guidelines EXACTLY. I didn't rename the project folder name from "arduino-esp32" to "esp32".

After renaming the folder to : arduino-1.8.2/hardware/espressif/**esp32**  I can connect to wifi.

haroonahmed97, 2017-07-10:
In my case same error was coming. The problem was that the WiFi setup was not the first thing it was written after some code in void setup( ). Problem was solved by writing WiFi setup at start of void setup( ) function.

haroonahmed97, 2017-07-17:
In my case CPU is halted whenever I attach touch interface to my code which used WiFi, pubSubClient and preferences library.

copercini, 2017-09-22:
The OP issue looks like this one: https://github.com/espressif/arduino-esp32/issues/207 and seems to be solved due his other posts with others issues....

CPU halted is a very generic error, and can happen for several reasons... so if anyone more is having problems, please open a new topic to keep the things organized

closed 2017-09-22 by copercini