espressif/arduino-esp32 #414: OTA Update From AWS S3

arvindr21, 2017-06-03:
@me-no-dev In continuation to our discussion on Gitter, I have run the below code to perform an OTA, hosting a bin file on AWS S3 & making it public download. Here is the code

```
#include <WiFi.h>
#include <Update.h>
#include <Preferences.h>

WiFiClient client;
Preferences preferences;

bool isOTADone = false;
int contentLength = 0;
bool isValidContentType = false;
char* SSID = "SSID";
char* PSWD = "PSWD";
void setup() {
  Serial.begin(115200);
  delay(10);

  Serial.println("Connecting to " + String(SSID));
  WiFi.begin(SSID, PSWD);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.println("Connected to " + String(SSID));

  preferences.begin("OTA", false);
  isOTADone = preferences.getBool("isOTADone", false);
  if (!isOTADone) {
    execOTA();
  } else {
    // Test the rollback to ping pong
    // between apps/bins
    if (Update.canRollBack()) {
      Serial.println("Rolling back bins");
      Update.rollBack();
    } else {
      Serial.println("Another Update");
      execOTA();
    }
  }
}

void loop() {
  // chill
}

void execOTA() {
  String host = "bucket-name.s3.ap-south-1.amazonaws.com";
  int port = 80;
  String url = "/sketchname.ino.bin";
  Serial.println("Connecting to: " + host);

  if (client.connect(host.c_str(), port)) {
    Serial.println("Fetching Bin: " + url);

    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Cache-Control: no-cache\r\n" +
                 "Connection: close\r\n\r\n");

// Check what is being sent
//    Serial.print(String("GET ") + url + " HTTP/1.1\r\n" +
//                 "Host: " + host + "\r\n" +
//                 "Cache-Control: no-cache\r\n" +
//                 "Connection: close\r\n\r\n");

    delay(100);
    while (client.available()) {
      String line = client.readStringUntil('\n');
      line.trim();

      if (!line.length()) {
        //headers ended
        break; // and get the OTA started
      }

      if (line.startsWith("HTTP/1.1")) {
        if (line.indexOf("200") < 0) {
          Serial.println("Got a non 200 status code from server. Exiting OTA Update.");
          break;
        }
      }

      // extract headers here
      // Start with content length
      if (line.startsWith("Content-Length: ")) {
        contentLength = atoi((getHeaderValue(line, "Content-Length: ")).c_str());
        Serial.println("Got " + String(contentLength) + " bytes from server");
      }

      // Next, the content type
      if (line.startsWith("Content-Type: ")) {
        String contentType = getHeaderValue(line, "Content-Type: ");
        Serial.println("Got " + String(contentType) + " payload.");
        if (contentType == "application/octet-stream") {
          isValidContentType = true;
        }
      }
    }
  } else {
    Serial.println("Connection to " + host + " failed. Please check your setup");
    // retry??
    // execOTA();
  }

  Serial.println("contentLength : " + String(contentLength) + ", isValidContentType : " + String(isValidContentType));

  // check contentLength and content type
  if (contentLength && isValidContentType) {
    bool canBegin = Update.begin(contentLength);
    
    if (canBegin) {
      Serial.println("Begin OTA. Things might be quite for a while.. Patience!");
      size_t written = Update.writeStream(client);
      Serial.println("written : " + String(written));
      if (Update.end()) {
        Serial.println("OTA done!");
        if (Update.isFinished()) {
          Serial.println("Update successfully completed. Rebooting.");
          preferences.putBool("isOTADone", true);
          ESP.restart();
        }
      } else {
        Serial.println("Error Occurred. Error #: " + String(Update.getError()));
      }
    }
  } else {
    Serial.println("There was no content in the response");
    client.flush();
  }
}

// Utility to extract header value from headers
String getHeaderValue(String header, String headerName) {
  return header.substring(strlen(headerName.c_str()));
}
```
And this is my Serial out

```
Connected to SSID
Connecting to: bucket-name.s3.ap-south-1.amazonaws.com
Fetching Bin: /sketchname.ino.bin
Got application/octet-stream payload.
Got 761328 bytes from server
contentLength : 761328, isValidContentType : 1
Begin OTA. Things might be quite for a while.. Patience!
written : 761328
Error Occurred. Error #: 3
```
I ran the same code twice. The first time I got error # 8 (UPDATE_ERROR_MAGIC_BYTE ) and now I got error # 3 (UPDATE_ERROR_READ). 

I tried re-uploading the above sketch a couple of times. But I keep getting the same error # 3. 

Also, noticed that my 32 dev module gets heated up while the OTA update is in progress.

Any help? 

Thanks,
Arvind.

arvindr21, 2017-06-03:
Thanks @me-no-dev. https://github.com/espressif/arduino-esp32/commit/bf60ca7e3b4a0639a9f243dbe97972a5ad0194ed push fixed it!

closed 2017-06-03 by arvindr21

vvs551, 2017-06-05:
Thank you. It is working

SulaimanEfendi, 2018-05-24:
When I try to compile AWS_S3_OTA.info I get the fatal error: Update.h: No such file or directory
How I can all the header files and libraries need in this example?

Thanks

SviFi, 2019-04-01:
> When I try to compile AWS_S3_OTA.info I get the fatal error: Update.h: No such file or directory
> How I can all the header files and libraries need in this example?
> 
> Thanks

It is important to put the name of the bucket at the beginning of the host string. AWS by default gives path with bucket name like this s3.eu-north-1.amazonaws.com/BUCKET/folder/file.bin and you need to but bucket name in front of host, separating by dot e.g.:
`String host =  "BUCKET.s3.eu-north-1.amazonaws.com";`

And then in the bin string include folder (if any), but not the bucket name:
`String bin = "/folder/file.bin";`

Another thing is it is important to make sure permissions of public access are set. By default, AWS disables public access permissions on every re-upload, even if you set them before. So you would have to make it public again after every re-upload.