espressif/arduino-esp32 #416: Get not the HIGH Level on GPIO16, GPIO17

ESP32DE, 2017-06-03:
hi
not sure ( i am not an constant Arduino User )
why i get not the HIGH Level 3.3/1.8   ( only  0.0012V ~  0.0055V )

Modul: Wrover

```
void setup() {
Serial.begin(115200);
pinMode(16, OUTPUT);
digitalWrite(16,LOW);
}

void loop() {
delay(3000);
Serial.println("PIN HIGH");
digitalWrite(16,HIGH);

delay(3000);
Serial.println("PIN LOW");
digitalWrite(16,LOW);
}
```
same on GPIO17

```

void setup() {
Serial.begin(115200);
pinMode(17, OUTPUT);
digitalWrite(17,LOW);
}

void loop() {
delay(3000);
Serial.println("PIN HIGH");
digitalWrite(17,HIGH);

delay(3000);
Serial.println("PIN LOW");
digitalWrite(17,LOW);
}
```

if i use example GPIO 4 or 5
then i get HIGH Level 3.3V

txs

ESP32DE, 2017-06-03:
sry - i forget that they not led out.
i use an adapter where on the silk is 16, 17 
its an wroom-32 / wrover adapter.
txs!..
for controll measuring i have to connect on the SRAM pins directly.
all ok - 

[image]

closed 2017-06-03 by ESP32DE

me-no-dev, 2017-06-04:
Rudi, did you get the LAN chip working fine? I got an "original" ethernet board but that seemed to have gone bad :(

ESP32DE, 2017-06-05:
@me-no-dev 

Hi Hristo, which type you get?
LAN8720 LAN8710 TLK110 or other?
Is the Board EVB from manufacturer or Breakout?
The most mistake is in setup the SMI address
example:
Oli_LAN8710 0x0,  
WaveShare LAN8720 0x0 or 0x1
TLK110 EVB Board from TI has 0x31
( TLK110CUSEVM ), is in MII mode
must change to RMII Mode,
removing the 25MHz crystal, 
ewmoving R71 and R72. 
place a zero ohm resistor on R73 
and added a 50MHz oscillator.
then you have SMI address 0x1
https://esp32.com/viewtopic.php?f=12&t=1178&hilit=LAN8720&start=10#p5710

LAN8720A EVB Board from MC has 0x0

the second is gpio pin setup
#define PIN_PHY_POWER CONFIG_PHY_POWER_PIN
#define PIN_SMI_MDC   CONFIG_PHY_SMI_MDC_PIN
#define PIN_SMI_MDIO CONFIG_PHY_SMI_MDIO_PIN

let me know which board you get,
do you need the arduino code or esp-idf ( eclipse ) code?

btw CAN is ready for esp32 arduino ide.
you get next time a PR CAN + LAN :)

best wishes
rudi ;-)

btw:
example code for ethernet is here
https://github.com/espressif/esp-idf/tree/master/examples/ethernet/ethernet
you can take it like it is, and run menuconfig
setup the board ( LAN8720/LAN8710 or TLK110 )
config the pins
config the address
and ready to go
for first start i would use DHCP -
you need not change IP or so.
be sure you use cross over cable if you use between pc and board or enable autosense
and DHCP must run on PC
if you connect to a router, there must run DHCP then on router.
you can setup IP fix too without DCP later.
hope this helps 
let me know your board and how you go on with this.
sry for delay - rock with pSRAM the weekend :)
best wishes
rudi ;-) 
sry for the big book text :)

me-no-dev, 2017-06-05:
Hi Rudi, my board is made by Espressif and is based on the TLK110 chip. I had it working just fine, I even made some modifications to the network functions in Arduino to prepare for integration, but all in a sudden I can no longer get it to work... unfortunately I do not have much time to deal with it either, but will need it soon, so will report on the situation.

me-no-dev, 2017-06-05:
```cpp
#include "esp_eth.h"
#include "eth_phy/phy.h"
#include "eth_phy/phy_tlk110.h"
#include "eth_phy/phy_lan8720.h"

#define ETH_PHY_CONF  phy_tlk110_default_ethernet_config
#define ETH_PHY_ADDR  PHY31
#define PIN_PHY_POWER 17
#define PIN_SMI_MDC   23
#define PIN_SMI_MDIO  18

static eth_config_t eth_config = ETH_PHY_CONF;

static void phy_rmii_config_gpio(void){
    // RMII data pins are fixed:
    // CRS_DRV = GPIO27
    // TXD0 = GPIO19
    // TXD1 = GPIO22
    // TX_EN = GPIO21
    // RXD0 = GPIO25
    // RXD1 = GPIO26
    // CLK == GPIO0
    phy_rmii_configure_data_interface_pins();
    // MDC is GPIO 23, MDIO is GPIO 18
    phy_rmii_smi_configure_pins(PIN_SMI_MDC, PIN_SMI_MDIO);
}

static void phy_power_enable(bool enable){
    pinMode(PIN_PHY_POWER, OUTPUT);
    digitalWrite(PIN_PHY_POWER, enable);
    delay(1);// Allow the power up/down to take effect, min 300us
}

extern void tcpipInit();

void ETH_begin(){
    eth_config.phy_addr = ETH_PHY_ADDR;
    eth_config.gpio_config = phy_rmii_config_gpio;
    eth_config.tcpip_input = tcpip_adapter_eth_input;
    eth_config.phy_power_enable = phy_power_enable;
    tcpipInit();
    if(!esp_eth_init(&eth_config)){
        esp_eth_enable();
    }
}

IPAddress ETH_localIP()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.ip.addr);
}

IPAddress ETH_subnetMask()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.netmask.addr);
}

IPAddress ETH_gatewayIP()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.gw.addr);
}

bool ETH_setHostname(const char * hostname)
{
    return tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_ETH, hostname) == 0;
}

bool ETH_enableIpV6()
{
    return tcpip_adapter_create_ip6_linklocal(TCPIP_ADAPTER_IF_ETH) == 0;
}

bool ETH_fullDuplex()
{
    return eth_config.phy_get_duplex_mode();
}

uint8_t ETH_linkSpeed()
{
    return eth_config.phy_get_speed_mode()?100:10;
}

IPv6Address ETH_localIPv6()
{
    static ip6_addr_t addr;
    if(tcpip_adapter_get_ip6_linklocal(TCPIP_ADAPTER_IF_ETH, &addr)){
        return IPv6Address();
    }
    return IPv6Address(addr.addr);
}

String ETH_macAddress(void)
{
    uint8_t mac[6];
    char macStr[18] = { 0 };
    esp_eth_get_mac(mac);
    sprintf(macStr, "%02X:%02X:%02X:%02X:%02X:%02X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
    return String(macStr);
}
```

ESP32DE, 2017-06-05:
have a note to the things are //***
( btw how you have done format color code? ) 

```
#include "esp_eth.h"
#include "eth_phy/phy.h"
//*** if you use tlk110
#include "eth_phy/phy_tlk110.h"
//*** if you use LAN8720
//*** #include "eth_phy/phy_lan8720.h"

#define ETH_PHY_CONF  phy_tlk110_default_ethernet_config
#define ETH_PHY_ADDR  PHY31
#define PIN_PHY_POWER 17
#define PIN_SMI_MDC   23
#define PIN_SMI_MDIO  18

static eth_config_t eth_config = ETH_PHY_CONF;

static void phy_rmii_config_gpio(void){
    // RMII data pins are fixed:
    // CRS_DRV = GPIO27
    // TXD0 = GPIO19
    // TXD1 = GPIO22
    // TX_EN = GPIO21
    // RXD0 = GPIO25
    // RXD1 = GPIO26
    // CLK == GPIO0
    phy_rmii_configure_data_interface_pins();
    // MDC is GPIO 23, MDIO is GPIO 18
    phy_rmii_smi_configure_pins(PIN_SMI_MDC, PIN_SMI_MDIO);
}

static void phy_power_enable(bool enable){
    pinMode(PIN_PHY_POWER, OUTPUT);
	//*** enable means HIGH 1 or
    digitalWrite(PIN_PHY_POWER, enable);
    delay(1);// Allow the power up/down to take effect, min 300us
}

extern void tcpipInit();

void ETH_begin(){
    eth_config.phy_addr = ETH_PHY_ADDR;
    eth_config.gpio_config = phy_rmii_config_gpio;
    eth_config.tcpip_input = tcpip_adapter_eth_input;
    eth_config.phy_power_enable = phy_power_enable;
    tcpipInit();
    if(!esp_eth_init(&eth_config)){
        esp_eth_enable();
    }
}

IPAddress ETH_localIP()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.ip.addr);
}

IPAddress ETH_subnetMask()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.netmask.addr);
}

IPAddress ETH_gatewayIP()
{
    tcpip_adapter_ip_info_t ip;
    tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_ETH, &ip);
    return IPAddress(ip.gw.addr);
}

bool ETH_setHostname(const char * hostname)
{
    return tcpip_adapter_set_hostname(TCPIP_ADAPTER_IF_ETH, hostname) == 0;
}

bool ETH_enableIpV6()
{
    return tcpip_adapter_create_ip6_linklocal(TCPIP_ADAPTER_IF_ETH) == 0;
}

bool ETH_fullDuplex()
{
    return eth_config.phy_get_duplex_mode();
}

uint8_t ETH_linkSpeed()
{
    return eth_config.phy_get_speed_mode()?100:10;
}

IPv6Address ETH_localIPv6()
{
    static ip6_addr_t addr;
    if(tcpip_adapter_get_ip6_linklocal(TCPIP_ADAPTER_IF_ETH, &addr)){
        return IPv6Address();
    }
    return IPv6Address(addr.addr);
}

String ETH_macAddress(void)
{
    uint8_t mac[6];
    char macStr[18] = { 0 };
    esp_eth_get_mac(mac);
    sprintf(macStr, "%02X:%02X:%02X:%02X:%02X:%02X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
    return String(macStr);
}

```