espressif/arduino-esp32 #417: CPU Halted on Wifi client

jestebanes, 2017-06-04:
Hello, I'm working on FTP client, to record on external flash files downloaded from a FTP server and I found some strange things that I can't understand, I was using without troubles the first version in an ESP2866 system

I've two versions of the critical code with the same result
first, reading bytes one by one:
        while (dclient.connected()) {
          int j=0;
          while (dclient.available()) {
            buf[j++] = dclient.read();
            if(j>=256){
              file.write((uint8_t *)buf, 256);
              j = 0;            
            }
          }
          if(j>0){
            file.write((uint8_t *)buf, j);
          }
       }
second, reading bytes in blocks
       while (dclient.connected()) {
          while ((dispo = dclient.available()) > 0) {
            int t;

            (dispo > BLOCKFTP)?t = BLOCKFTP:t=dispo;
            if((readed = dclient.read(buf, t))>0){
              file.write(buf, readed);
            }else{
              dbg->printf("Fail reading from ftp client %d bytes\n", t);
              break;
            }
          }
      }

when I work with little files the ESP32 works fine, but when I use big files 8M or above, the ESp32 crash, first I think maybe was overheating but now I've got the ESP32 working with a cooler and the same trouble happens, any ideas?

king regards

Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40087899  PS      : 0x00060333  A0      : 0x80086285  A1      : 0x3ffe3710  
A2      : 0x3ffdfe0c  A3      : 0x00060323  A4      : 0x00000000  A5      : 0x3ffc05c0  
A6      : 0x3ff000e0  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffdfe0c  
A10     : 0x3ffdfe0c  A11     : 0x00060323  A12     : 0x00060323  A13     : 0x3ffc05a0  
A14     : 0x3ffc1464  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x40087899:0x3ffe3710 0x40086285:0x3ffe3730 0x40085780:0x3ffe3750 0x401511ec:0x3ffe3790 0x401781d1:0x3ffe37c0 0x4017d87c:0x3ffe37f0 0x40178c58:0x3ffe38b0 0x40179620:0x3ffe38f0 0x40179729:0x3ffe3920 0x4017dbd5:0x3ffe3950

CPU halted.

me-no-dev, 2017-06-07:
Have you looked into https://github.com/me-no-dev/EspExceptionDecoder ? It might help with deciphering the errors

jestebanes, 2017-06-07:
Thanks for the help, but I arrive at an end point in the freertos library, apparently there is a trouble in any point of the wifi client library, because related to the error I found into the rtos code the next message

                       1) Stack overflow -
			   see http://www.freertos.org/Stacks-and-stack-overflow-checking.html
			2) Incorrect interrupt priority assignment, especially on Cortex-M
			   parts where numerically high priority values denote low actual
			   interrupt priorities, which can seem counter intuitive.  See
			   http://www.freertos.org/RTOS-Cortex-M3-M4.html and the definition
			   of configMAX_SYSCALL_INTERRUPT_PRIORITY on
			   http://www.freertos.org/a00110.html
			3) Calling an API function from within a critical section or when
			   the scheduler is suspended, or calling an API function that does
			   not end in "FromISR" from an interrupt.
			4) Using a queue or semaphore before it has been initialised or
			   before the scheduler has been started (are interrupts firing
			   before vTaskStartScheduler() has been called?).

The values returned by the exception decoder tool, there are two different failures at similar parts of the rtos code

0x40087899: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 218
0x40087899: uxListRemove at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 218
0x40086285: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4420
:  (inlined by) xTaskRemoveFromEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 3046
0x40085780: xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 2483
0x401513c0: coex_wifi_force_stop at ?? line ?
0x401783a5: lmacRecycleMPDU at ?? line ?
0x4017da50: ppResortTxAMPDU at ?? line ?
0x40178e2c: lmacTxFrame at ?? line ?
0x401797f4: lmacProcessTxSuccess at ?? line ?
0x401798fd: lmacProcessTxComplete at ?? line ?
0x4017dda9: ppTask at ?? line ?
0x4008786c: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 188 (discriminator 1)
0x4008786c: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 188 (discriminator 1)
0x40086cf7: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4420
:  (inlined by) vTaskPlaceOnEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 2857
0x40085b43: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 2483
0x40151379: coex_wifi_force_stop at ?? line ?
0x401783a5: lmacRecycleMPDU at ?? line ?
0x4017da50: ppResortTxAMPDU at ?? line ?
0x40178e2c: lmacTxFrame at ?? line ?
0x401797f4: lmacProcessTxSuccess at ?? line ?
0x401798fd: lmacProcessTxComplete at ?? line ?
0x4017dda9: ppTask at ?? line ?

King regards

me-no-dev, 2017-06-07:
this looks like it could be causing by bad power. Make sure the USB cable is good and not too long, or try external power and larger capacitor to coop with current spikes when the radio is transmitting.

jestebanes, 2017-06-08:
Hi, thanks for the support, at the end the trouble was an interference between BLE and WiFi, when I stop the BLE while the files are downloading, then the trouble disappear.
I also follow your advice and reinforce the FDA until all spikes disappear, to avoid other troubles.

King regards

hpsaturn, 2017-06-22:
Similar problem in my `esp32` over `platformio` Arduino, but in my case here is **aleatory**: 
(Sorry for nest output but I don't have `EspExceptionDecoder` for `platformio` via commandline, only I found for old ESP8266)

**Code**:
```Python
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
```
**Output**:
```Python
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x400d0b64  PS      : 0x00060034  A0      : 0x40081f71  A1      : 0x3ffc0be0  
A2      : 0x00000003  A3      : 0x3ffc2dcc  A4      : 0x00000008  A5      : 0x40086238  
A6      : 0x00000003  A7      : 0x00060f23  A8      : 0x80080e5e  A9      : 0x04000c08  
A10     : 0x00000008  A11     : 0x00000006  A12     : 0x00060023  A13     : 0x3ffc0bc0  
A14     : 0x3ffc0c08  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x400d0b64:0x3ffc0be0 0x40081f71:0x3ffc0c00

CPU halted.
```
----
**Hardware**:
* WeMOS ESP32 OLED
* Lipo 300mAh
* Lipo Battery Charger module 
* Input voltage: `~3.265-3.178`

Many Thanks.

yaqwsx, 2017-07-30:
I have the exactly same problem as @hpsaturn - I try to call WiFi.begin() from a new task. When I call exactly the same code from main task, everything works.

lonerzzz, 2017-07-30:
@yaqwsx In my code, I hit this as well. Instead of trying to get it working in the other thread, I retained HTTP communication on the main thread and just passed messages to the main thread on a queue. As such, my main thread is responsible for all wireless communications and interacting with the UART. This works fine for me.

dmody, 2017-08-01:
@me-no-dev 
I tried installing and running the exception decoder, but when I run the decoder it asks me choose to open an .elf file.  I'm not sure what an ELF file is, or where to find it.

copercini, 2017-08-01:
@dmody just compile and upload your code again, so exception decoder will work fine

dmody, 2017-08-01:
Seems like I need to close the Serial Monitor too?

lonerzzz, 2017-09-28:
@jestebanes, @dmody  Is this still an issue or can the issue be closed?

jestebanes, 2017-10-15:
Finally, I came up with the solution, the problem was that WIFICLIENT.available() returns a number of bytes that you can read, but when you really read they, the number of bytes read by WIFICLIENT.read could be less.
If you write the code with this thing in mind the code doesn't fail.
Thanks for the help

lonerzzz, 2017-10-15:
Glad you found out how to work around, but that certainly isn't what you would want for functionality. Probably need to keep this open to make available() mean what a person would expect.

dmody, 2017-10-15:
I'm finding very long run times, but still the occasional hang that causes the radios to stay on and power to drain.  You can see I've had two such cases since July from the following graph.
[image]

closed 2019-08-15 by stale[bot]