espressif/arduino-esp32 #420: Can't use two timer in code

maxpromer, 2017-06-06:
Hello !, I use NodeMCU-32S board. I can't use 2 timer in code, the second timer not callback via timerAlarmWrite() but the first timer can callback and work.
My code :

```Arduino
#define LED1 LED_BUILTIN
#define LED2 23

hw_timer_t *timerOne = NULL;
hw_timer_t *timerTwo = NULL;

void onTimerOne(){
  digitalWrite(LED1, !digitalRead(LED1));
}

void onTimerTwo(){
  digitalWrite(LED2, !digitalRead(LED2));
}

void setup() {
  pinMode(LED1, OUTPUT);
  pinMode(LED2, OUTPUT);
  
  timerOne = timerBegin(0, 80, true);
  timerAttachInterrupt(timerOne, &onTimerOne, true);
  timerAlarmWrite(timerOne, 500000, true);
  timerAlarmEnable(timerOne);
  timerStart(timerOne);

  timerTwo = timerBegin(1, 80, true);
  timerAttachInterrupt(timerTwo, &onTimerTwo, true);
  timerAlarmWrite(timerTwo, 300000, true);
  timerAlarmEnable(timerTwo);
  timerStart(timerTwo);
}

void loop() {}
```

LED on board blink but LED on GPIO23 not blink.

Thank for reply.

lrmoreno007, 2017-06-06:
Try with timerAlarmEnable after timerStart:

```
#define LED1 LED_BUILTIN
#define LED2 23

hw_timer_t *timerOne = NULL;
hw_timer_t *timerTwo = NULL;

void onTimerOne(){
	digitalWrite(LED1, !digitalRead(LED1));
}

void onTimerTwo(){
	digitalWrite(LED2, !digitalRead(LED2));
}

void setup() {
	Serial.begin(115200);
	pinMode(LED1, OUTPUT);
	pinMode(LED2, OUTPUT);

	timerOne = timerBegin(0, 80, true);
	timerAttachInterrupt(timerOne, &onTimerOne, true);
	timerAlarmWrite(timerOne, 500000, true);
	timerStart(timerOne);
	timerAlarmEnable(timerOne);

	timerTwo = timerBegin(3, 80, true);
	timerAttachInterrupt(timerTwo, &onTimerTwo, true);
	timerAlarmWrite(timerTwo, 300000, true);
	timerStart(timerTwo);
	timerAlarmEnable(timerTwo);

}

void loop() {}

```

Or with double timerAlarmEnable (I don't know why...):
```
#define LED1 LED_BUILTIN
#define LED2 23

hw_timer_t *timerOne = NULL;
hw_timer_t *timerTwo = NULL;

void onTimerOne(){
	digitalWrite(LED1, !digitalRead(LED1));
}

void onTimerTwo(){
	digitalWrite(LED2, !digitalRead(LED2));
}

void setup() {
	Serial.begin(115200);
	pinMode(LED1, OUTPUT);
	pinMode(LED2, OUTPUT);

	timerOne = timerBegin(0, 80, true);
	timerAttachInterrupt(timerOne, &onTimerOne, true);
	timerAlarmWrite(timerOne, 500000, true);
	timerAlarmEnable(timerOne);
	timerAlarmEnable(timerOne);
	timerStart(timerOne);

	timerTwo = timerBegin(3, 80, true);
	timerAttachInterrupt(timerTwo, &onTimerTwo, true);
	timerAlarmWrite(timerTwo, 300000, true);
	timerAlarmEnable(timerTwo);
	timerAlarmEnable(timerTwo);
	timerStart(timerTwo);

}

void loop() {}

```

lonerzzz, 2017-08-15:
@maxpromer Did the suggested fix address your issue? Can we close this item?

closed 2017-08-17 by maxpromer