espressif/arduino-esp32 #421: WPS issue

bestpika, 2017-06-07:
I try below code, but I can't connect wps every time.
Could anyone give me some advice?
Thanks.
```cpp
#include <esp_wps.h>

esp_wifi_wps_enable(WPS_TYPE_PBC);
esp_wifi_wps_start(60000);
```

me-no-dev, 2017-06-07:
I do not see wifi being started at all in this code.

closed 2017-06-07 by me-no-dev

bestpika, 2017-06-07:
full code
```cpp
#include <esp_task_wdt.h>
#include <WiFi.h>
#include <esp_wps.h>

void setup() {
  esp_task_wdt_init();

  Serial.begin(115200);
  Serial.println();
  Serial.println();

  WiFi.begin();
  WiFi.mode(WIFI_STA);
  int wait = 0;
  while (!WiFi.isConnected() && wait++ < 30) {
    Serial.print(".");
    delay(1000);
  }
  if (!WiFi.isConnected()) {
    esp_wifi_wps_enable(WPS_TYPE_PBC);
    delay(1000);
    esp_wifi_wps_start(30000);
  }
  wait = 0;
  while (!WiFi.isConnected() && wait++ < 15) {
    Serial.print(".");
    delay(1000);
  }
  Serial.println();
  if (WiFi.isConnected()) {
    Serial.println(WiFi.localIP().toString() + " (" + WiFi.macAddress() + ") (" + WiFi.SSID() + ")");
  } else {
    Serial.println(WiFi.status());
    ESP.restart();
  }
}

void loop() {
  esp_task_wdt_feed();
}
```

me-no-dev, 2017-06-07:
Do you notice a pattern? like it works every other time, or is it random when it would work or not?

bestpika, 2017-06-07:
If wps can't connect then always happened.