espressif/arduino-esp32 #424: ESP.restart() and CPU halted

bestpika, 2017-06-07:
Below code caused CPU halted. (WiFi.status() will show 255);
```cpp
void setup() {
  esp_task_wdt_init();

  Serial.begin(115200);
  Serial.println();
  Serial.println();

  WiFi.mode(WIFI_STA);
  WiFi.begin();
  int wait = 0;
  while (!WiFi.isConnected() && wait++ < 5) {
    Serial.print(".");
    delay(1000);
  }
  if (!WiFi.isConnected()) {
    esp_wifi_wps_enable(WPS_TYPE_PBC);
    esp_wifi_wps_start(15000);
  }
  wait = 0;
  while (!WiFi.isConnected() && wait++ < 10) {
    Serial.print(".");
    delay(1000);
  }
  Serial.println();
  if (WiFi.isConnected()) {
    Serial.println(WiFi.localIP().toString() + " (" + WiFi.macAddress() + ") (" + WiFi.SSID() + ")");
  } else {
    Serial.println(WiFi.status());
    ESP.restart(); // here
  }
}
```

me-no-dev, 2017-06-07:
Maybe attach some log? Also please ask for WPS in IDF's repository. Arduino does not extend an API yet.

bestpika, 2017-06-07:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034

79956516
.............................................
255
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40118996  PS      : 0x00060930  A0      : 0x8011b310  A1      : 0x3ffd5e50  
A2      : 0x0000010f  A3      : 0x00000003  A4      : 0x401320b0  A5      : 0x00000000  
A6      : 0x3ffc7780  A7      : 0x3ffc7358  A8      : 0x8011898b  A9      : 0x3ffd5e30  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc0560  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000d8  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40118996:0x3ffd5e50 0x4011b310:0x3ffd5e70 0x4013216c:0x3ffd5e90 0x4013dfe9:0x3ffd5eb0 0x4013e064:0x3ffd5ed0

CPU halted.
```

bestpika, 2017-06-07:
```cpp
...
if (WiFi.isConnected()) {
  Serial.println(WiFi.localIP().toString() + " (" + WiFi.macAddress() + ") (" + WiFi.SSID() + ")");
} else {
  Serial.println(WiFi.status());
  esp_wifi_wps_disable() // add this, okay
  ESP.restart();
}
...
```