espressif/arduino-esp32 #425: ESP32 webserver

skateone, 2017-06-07:
Has anyone implemented a decent web server yet like the esp8266 webserver solution.

PhilColbert, 2017-06-07:
ESP8266WebServer works a treat :)

lonerzzz, 2017-06-08:
@PhilColbert Translate to English please... :) Is this a vote for the ESP8266WebServer or have you got something that has worked?

salqadri, 2017-06-10:
Created a PR for this: #430

yeralin, 2017-06-26:
This SimpleWiFiServer is crazy confusing.

What's kinda gypsy magic is going on there?

```
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character
```

Will there be more adequate implementation of WebServer on EPS32?

I'm thinking of returning my ESP32 and just get old ESP8266 :/

fesch, 2017-07-09:
I have copied and renamed the library ESP8266WebServer to ESP32WebServer. Only a very view adaptations were needed. So for those who need this right now and can't wait, this might be an easy way to go.

But as mentioned by @me-no-dev in issue #430, it makes non sense to have the same code in two places. I strongly agree on that!

fesch, 2017-07-16:
http://www.fisch.lu/junk/ESP32-WebServer.zip

Copy the library to the "libraries" folder of you expressif/esp32 installation folder.

Jimbobnz, 2017-07-17:
Downloaded the ESP32-WebServer.zip file but the example WebServer examples reference ESP8266 files and won't compile.

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

```

Jimbobnz, 2017-07-17:
I believe got the necessary correction needed.

```
#include <WiFi.h>
#include <WiFiClient.h>
#include <ESP32WebServer.h>
#include <ESPmDNS.h>

const char* ssid = "........";
const char* password = "........";

ESP32WebServer server(80);
```

fesch, 2017-07-17:
Yes, you are right. I forgot yesterday to update the examples ...

bbx10, 2017-07-17:
This project roadmap may be helpful.

https://github.com/espressif/arduino-esp32/projects/1

lonerzzz, 2017-07-17:
@fesch, @bbx10 I added a work item to update the example code into the road map. If you own any of the tasks, please indicate that. If any taske are still missing, feel free to add or I can do it if you tell me what it is.

gusantor, 2017-08-29:
I've just downladed .zip file posted by fesch

Copied WebServer folder to ~/Arduino/hardware/espressif/esp32/libraries

open arduino ide 1.8

open examples -> Esp32WebServer -> SimpleWebServer

compile, receive multiple errors , as

arduino-1.8.0/libraries/Ethernet/src/Dhcp.h:47:19: error: expected identifier before string constant
 #define HOST_NAME "WIZnet"

Arduino/hardware/espressif/esp32/tools/sdk/include/lwip/apps/dhcpserver.h:58:5: note: in expansion of macro 'HOST_NAME'
     HOST_NAME = 12,
.
.
.
Arduino/hardware/espressif/esp32/tools/sdk/include/lwip/apps/dhcpserver.h:154:1: error: expected declaration before '}' token
 } dhcp_msg_option;

please, any advise ? thanks

MarkyAD, 2017-08-30:
@gusantor If you're just looking to use wifi, comment out these lines:

// #include <Ethernet.h>
// ESP32Ethernet ethernet;

and change:
 ethernet.begin();
to:
 server.begin();

gusantor, 2017-08-30:
Thank you @MarkyAD , it works great

Greeniemax, 2017-09-01:
http://www.fisch.lu/junk/ESP32-WebServer.zip isn't working as link... 

I'm so stuck, I'm stuck on server.streamFile not working.

Greeniemax, 2017-09-01:
http://www.fisch.lu/junk/ESP32-WebServer.zip link isn't working.

I could see the webserver is still under development, when will it be ready, got projects in pipeline where I would replace ESP8266 with ESP32.

fesch, 2017-09-01:
The link should still work ...

kendo55, 2017-10-08:
thanks !!!
i tested your lib under: http://www.fisch.lu/junk/ESP32-WebServer.zip
with my example, that runs on the  ESP826.
____________________
webserver.send(200, "text/html", sHTML);     works fine !!
_________________________________________
  HTTPUpload& upload = webserver.upload();     works fine !!
_________________________________________
Download from the SPIFFS does not work very well.
It works, but the downloaded content ist wrong--> it begins with the header.....

 File downfile = SPIFFS.open(webserver.arg("filename"),"r"); 
        webserver.sendHeader("Content-Disposition"," attachment; filename=\"" + sendname + "\"");
        size_t sent = webserver.streamFile(downfile,"application/octet-stream");
_________________________________________
i think the method  webserver.streamFile()  is the problem.
To stream a .gif or .jpg-file does not work

 File pageFile = SPIFFS.open("/logo.gif");
 size_t sent = webserver.streamFile(pageFile, "image/gif");

mihailescu2m, 2017-10-22:
Not even html or simple text file won't work for me, since the function in WiFiClient::write(Stream) is not implemented.

kendo55, 2017-10-22:
webserver.streamFile in this lib:  http://www.fisch.lu/junk/ESP32-WebServer.zip
does not work!! see my last post.
I found another library:
https://github.com/bbx10/WebServer_tng
it works very well!!!!

tobiilg, 2018-01-27:
@kendo55 thank you! The WebServer_tng by @bbx10 was the one I was looking for!

pipi61, 2018-01-27:
http://www.fisch.lu/junk/ESP32-WebServer.zip not wifi, but for wired ethernet ...

sunepedersen, 2018-04-04:
The WebServer_tng streamfile does also not work for me.

skyladle, 2018-04-12:
root CA upload to SPIFF through HTTPS not successfully.
I try to upload a root CA ( the root CA was from AWS symantec) to SPIFFS through HTTPs, however I can not upload this root CA successful, only root CA can not successful, others private key and public keys are ok to upload through HTTPS.  I try to trace the code found the code never reach phase of _currentUpload.status = UPLOAD_FILE_END. it hang at status=UPLOAD_FILE_WRITE.

in WebServer::_parseForm() of parsing.cpp
content of endBuf is always weird that while run till 
 if (strstr((const char*)endBuf, boundary.c_str()) != NULL) it always can not fulfill this condition.
always can not find boundary in endBuf.
can somebody teach me how to solve ?

debug information as below 

Parse Form: Boundary: ----WebKitFormBoundaryx8LmZvvONNsnhaKX Length: 1963
PostArg FileName: /symtec_G5_CA.pem
PostArg Name: data
PostArg Type: application/octet-stream
Start File: /symtec_G5_CA.pem Type: application/octet-stream
handleFileUpload Name: /symtec_G5_CA.pem
Quit FILE START phase
UPLOAD_FILE_WRITE PHASE
total size =1731
content of endBuf=**---END CERTIFICATE-----
------WebKitF⸮?L2⸮?d2⸮?**
content of boundary=**----WebKitFormBoundaryx8LmZvvONNsnhaKX**
boundary length=38

[image]

kendo55, 2018-04-19:
@ sunepedersen 

it is essential, that you copy the library (WebServer_tng) in directory:
[Arduino]\hardware\espressif\esp32\libraries

for example: D:\arduino_1_8_5_ESP32\hardware\espressif\esp32\libraries

copercini, 2018-06-27:
Please update to last code, it's now supporting webserver =)
https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer

closed 2018-06-27 by copercini