espressif/arduino-esp32 #426: Adafruit FONA 808: couldn't find fona

architmuchhal12, 2017-06-07:
Hello,
I am using Espressif ESp32 along with Adafruit FONA808. I am able to compile the code and upload too.
After uploading, I get a message on Serial Com as -

`Couldn't find FONA`

I have set the baud rate as 115200 and also attached a battery. I am attaching the screenshots of the Serial Com.

[image]
[image]

The connections which I've made are as-

VIO ---> 3V3
GND ---> GND
RST ---> Pin 4 (GPIO4)
TX ---> Pin 16 (U2_RXD)
RX ---> Pin 17 (U2_TXD)
Key ---> GND

Any suggestions are welcome.
Thanks!

me-no-dev, 2017-06-07:
How do you initialize Serial2? Show some code :)

architmuchhal12, 2017-06-08:
This is the code which I used. It's a basic FONAtest example from the Adafruit FONA Library - 
```

#include <SoftwareSerial.h>

/***************************************************
  This is an example for our Adafruit FONA Cellular Module

  Designed specifically to work with the Adafruit FONA
  ----> http://www.adafruit.com/products/1946
  ----> http://www.adafruit.com/products/1963
  ----> http://www.adafruit.com/products/2468
  ----> http://www.adafruit.com/products/2542

  These cellular modules use TTL Serial to communicate, 2 pins are
  required to interface
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  BSD license, all text above must be included in any redistribution
 ****************************************************/

/*
THIS CODE IS STILL IN PROGRESS!

Open up the serial console on the Arduino at 115200 baud to interact with FONA

Note that if you need to set a GPRS APN, username, and password scroll down to
the commented section below at the end of the setup() function.
*/
#include "Adafruit_FONA.h"

#define FONA_TX 16 
#define FONA_RX 17 
#define FONA_RST 4

// this is a large buffer for replies
char replybuffer[255];

// We default to using software serial. If you want to use hardware serial
// (because softserial isnt supported) comment out the following three lines 
// and uncomment the HardwareSerial line
SoftwareSerial fonaSS = SoftwareSerial(FONA_TX, FONA_RX);
SoftwareSerial *fonaSerial = &fonaSS;

// Hardware serial is also possible!
//  HardwareSerial *fonaSerial = &Serial1;

// Use this for FONA 800 and 808s
Adafruit_FONA fona = Adafruit_FONA(FONA_RST);
// Use this one for FONA 3G
//Adafruit_FONA_3G fona = Adafruit_FONA_3G(FONA_RST);

uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);

uint8_t type;

void setup() {
  while (!Serial);

  Serial.begin(115200);
  Serial.println(F("FONA basic test"));
  Serial.println(F("Initializing....(May take 3 seconds)"));
  delay(1000);

  fonaSerial->begin(115200);
  if (! fona.begin(*fonaSerial)) {
    Serial.println(F("Couldn't find FONA"));
[... 1680 lines cut ...]
      //Serial.print(c, HEX); Serial.print("#"); Serial.println(c);

      if (c == '\r') continue;
      if (c == 0xA) {
        if (buffidx == 0)   // the first 0x0A is ignored
          continue;

        timeout = 0;         // the second 0x0A is the end of the line
        timeoutvalid = true;
        break;
      }
      buff[buffidx] = c;
      buffidx++;
    }

    if (timeoutvalid && timeout == 0) {
      //Serial.println(F("TIMEOUT"));
      break;
    }
    delay(1);
  }
  buff[buffidx] = 0;  // null term
  return buffidx;
}
```

me-no-dev, 2017-06-08:
Aaahhh SoftwareSerial ;) that is probably your problem ;) lib might need rework to be able to work with one of ESP32's hardware serial interfaces

architmuchhal12, 2017-06-08:
I had found this library for ESP32 Software Serial -
[https://github.com/jdollar/espsoftwareserial/]

So, should I use this or there is some other way in which I should proceed?

Thanx!

andrei-ivanov, 2017-06-08:
```
// Hardware serial is also possible!
//  HardwareSerial *fonaSerial = &Serial1;
```

Doesn't that work? It's commented there in the code.

me-no-dev, 2017-06-08:
ESP32 has 3 hardware serials that you can attach to pretty much any pin (pins below 34), So no need to try SoftSerial at all

architmuchhal12, 2017-06-08:
Hey, thanks for the suggestion.
So, I did some changes to use the HardwareSerial.
But, I think, I am missing something here.
Could you just help me with that?
This is the code -

```
#include "Adafruit_FONA.h"
//#include <SoftwareSerial.h>

//#define FONA_TX 16 
//#define FONA_RX 17 
#define FONA_RST 4

// this is a large buffer for replies
char replybuffer[255];

// We default to using software serial. If you want to use hardware serial
// (because softserial isnt supported) comment out the following three lines 
// and uncomment the HardwareSerial line

//SoftwareSerial fonaSS = SoftwareSerial(FONA_TX, FONA_RX);
//SoftwareSerial *fonaSerial = &fonaSS;

// Hardware serial is also possible!
  HardwareSerial *fonaSerial = &Serial2;

// Use this for FONA 800 and 808s
Adafruit_FONA fona = Adafruit_FONA(FONA_RST);
// Use this one for FONA 3G
//Adafruit_FONA_3G fona = Adafruit_FONA_3G(FONA_RST);

uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);

uint8_t type;

void setup() {
  while (!Serial);

  Serial.begin(115200);
  Serial.println(F("FONA basic test"));
  Serial.println(F("Initializing....(May take 3 seconds)"));
  delay(1000);

  fonaSerial->begin(115200);
  if (! fona.begin(*fonaSerial)) {
    Serial.println(F("Couldn't find FONA"));
    while (1);
  }
  type = fona.type();
  Serial.println(F("FONA is OK"));
  Serial.print(F("Found "));
  switch (type) {
    case FONA800L:
      Serial.println(F("FONA 800L")); break;
    case FONA800H:
      Serial.println(F("FONA 800H")); break;
    case FONA808_V1:
      Serial.println(F("FONA 808 (v1)")); break;
    case FONA808_V2:
      Serial.println(F("FONA 808 (v2)")); break;
    case FONA3G_A:
      Serial.println(F("FONA 3G (American)")); break;
    case FONA3G_E:
      Serial.println(F("FONA 3G (European)")); break;
    default: 
      Serial.println(F("???")); break;
  }
  
  // Print module IMEI number.
  char imei[16] = {0}; // MUST use a 16 character buffer for IMEI!
  uint8_t imeiLen = fona.getIMEI(imei);
  if (imeiLen > 0) {
    Serial.print("Module IMEI: "); Serial.println(imei);
  }
printMenu();
}

```

I am getting error as -

```
FONAtest:45: error: 'Serial2' was not declared in this scope

   HardwareSerial *fonaSerial = &Serial2;

                                 ^
```

I really appreciate the quick response!
Thanks a lot!

me-no-dev, 2017-06-08:
```cpp
HardwareSerial *fonaSerial = new HardwareSerial(2);
```
OR
```cpp
HardwareSerial Serial2(2);
HardwareSerial *fonaSerial = &Serial2;
```

architmuchhal12, 2017-06-08:
Thanks a lot! I have got it to work!
You guys are doing great work! Cheers!

mswmIoT, 2017-06-13:
hello,
am using adafruit fona32u4 am getting the following error while running Fonatest from example library, i have tried both hardware and software serial. Can anybody help please.
FONA basic test
Initializing....(May take 3 seconds)
Attempting to open comm with ATs
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
Timeout: No response to AT... last ditch attempt.
	---> AT
	<--- 
	---> AT
	<--- 
	---> AT
	<--- 
	---> ATE0
	<--- 
	---> ATE0
	<--- 
Couldn't find FONA

forthlightning, 2017-06-21:
Is there a way to use FONA with TLS and MQTT? I'm trying to publish messages to a MQTT broker, but it requires TLS encryption

copercini, 2017-06-21:
@forthlightning I know 2 methods (there may be more):

- Using the old stuff provided by SIM808: SSL2.0, SSL3.0, TLS1.0 (more details in http://simcom.ee/documents/SIM800x/SIM800%20Series_SSL_Application%20Note_V1.01.pdf)

- PPPoS: ESP32 manage the cryptographic stuff and SIM808 just provide a transparent communication, here is a example for IDF: https://github.com/loboris/ESP32-PPPOS-EXAMPLE

hevnsnt, 2017-07-16:
@architmuchhal12 do you mind posting your final code?  I am also having the same problem. Thanks!

architmuchhal12, 2017-07-17:
@hevnsnt You need to use HardwareSerial. That's the only change I made - 
Here is the code -

```
#include "Adafruit_FONA.h"

#define FONA_RST 25

// this is a large buffer for replies
char replybuffer[255];

// Hardware serial is also possible!
HardwareSerial Serial2(2);
HardwareSerial *fonaSerial = &Serial2;

// Use this for FONA 800 and 808s
Adafruit_FONA fona = Adafruit_FONA(FONA_RST);

uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);

uint8_t type;

void setup() {
  while (!Serial);

  Serial.begin(115200);
  Serial.println(F("FONA basic test"));
  Serial.println(F("Initializing....(May take 3 seconds)"));
  delay(1000);

  fonaSerial->begin(115200);
  if (! fona.begin(*fonaSerial)) {
    Serial.println(F("Couldn't find FONA"));
    while (1);
  }
  type = fona.type();
  Serial.println(F("FONA is OK"));
  Serial.print(F("Found "));
  switch (type) {
    case FONA800L:
      Serial.println(F("FONA 800L")); break;
    case FONA800H:
      Serial.println(F("FONA 800H")); break;
    case FONA808_V1:
      Serial.println(F("FONA 808 (v1)")); break;
    case FONA808_V2:
      Serial.println(F("FONA 808 (v2)")); break;
    case FONA3G_A:
      Serial.println(F("FONA 3G (American)")); break;
    case FONA3G_E:
      Serial.println(F("FONA 3G (European)")); break;
    default:
      Serial.println(F("???")); break;
  }
```

carlosbvz, 2017-08-05:
Hi Guys, 

Same error here and I tried to replace the softSerial with those lines in order to use the hardwareSerial. 

I got this error: (Any help would be appreciated)

`
FONAtest:46: error: no matching function for call to 'HardwareSerial::HardwareSerial(int)'
 HardwareSerial Serial2(2);
                         ^
/var/folders/tp/7_mqsr6d4t39s0pj11bnng69_rx_kz/T/arduino_modified_sketch_589308/FONAtest.ino:46:25: note: candidates are:
In file included from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:232:0,
                 from sketch/FONAtest.ino.cpp:1:
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:117:12: note: HardwareSerial::HardwareSerial(volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*, volatile uint8_t*)
     inline HardwareSerial(
            ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:117:12: note:   candidate expects 6 arguments, 1 provided
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:93:7: note: constexpr HardwareSerial::HardwareSerial(const HardwareSerial&)
 class HardwareSerial : public Stream
       ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:93:7: note:   no known conversion for argument 1 from 'int' to 'const HardwareSerial&'
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:93:7: note: constexpr HardwareSerial::HardwareSerial(HardwareSerial&&)
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/HardwareSerial.h:93:7: note:   no known conversion for argument 1 from 'int' to 'HardwareSerial&&'
exit status 1
no matching function for call to 'HardwareSerial::HardwareSerial(int)'`

me-no-dev, 2017-08-05:
You selected the wrong board in the menu.

closed 2017-08-05 by me-no-dev

mobimation, 2018-03-01:
If you are using the sketch provided for Fona 32u4 they have a part to confuse the enemy:

You'll need to make a simple change to the sketch. At the top find these lines:
 Download file Copy Code
#include "Adafruit_FONA.h"
 
#define FONA_RX 2
#define FONA_TX 3
#define FONA_RST 4
and change them to:
 Download file Copy Code
#define FONA_RX  9
#define FONA_TX  8
#define FONA_RST 4
#define FONA_RI  7

A subtle mapping difference think you not ?  :)
Well once changed the sketch works.
No doubt many moons have passed since and there have had plenty of time to modifying that sketch instead of leaving it up to the developer.
But maybe it is about picking out the cherry developers.

beegee-tokyo, 2018-03-02:
Added your findings to the [Wiki](http://desire.giesecke.tk/index.php/2018/03/02/problem-with-adafruit-fona-library/)

AlphaMinor, 2018-05-14:
@architmuchhal12  did u use a hardware serial library?

RicoJia, 2018-05-18:
@architmuchhal12  @AlphaMinor  Hi all, Try to check the physical connections: Fona_TX to your arduino pins! I am using a shield and had EXACTLY the same problem. However, I found out this bug and now it's working totally fine.

AustinGregory223, 2018-06-24:
Hey Guys,
I have a DFRobot Sim 7000c and am getting the same issue every time "Couldn't find FONA"
This is my Code
Much appreciated if anyone could help becasue im  building a GPS Tracker

AustinGregory223, 2018-06-24:
/***************************************************
  This is an example for our Adafruit FONA Cellular Module

  Designed specifically to work with the Adafruit FONA
  ----> http://www.adafruit.com/products/1946
  ----> http://www.adafruit.com/products/1963
  ----> http://www.adafruit.com/products/2468
  ----> http://www.adafruit.com/products/2542

  These cellular modules use TTL Serial to communicate, 2 pins are
  required to interface
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  BSD license, all text above must be included in any redistribution
 ****************************************************/

/*
THIS CODE IS STILL IN PROGRESS!

Open up the serial console on the Arduino at 115200 baud to interact with FONA

Note that if you need to set a GPRS APN, username, and password scroll down to
the commented section below at the end of the setup() function.
*/
#include "Adafruit_FONA.h"

#define FONA_RX 2
#define FONA_TX 3
#define FONA_RST 4

// this is a large buffer for replies
char replybuffer[255];

// We default to using software serial. If you want to use hardware serial
// (because softserial isnt supported) comment out the following three lines 
// and uncomment the HardwareSerial line
#include <SoftwareSerial.h>
SoftwareSerial fonaSS = SoftwareSerial(FONA_TX, FONA_RX);
SoftwareSerial *fonaSerial = &fonaSS;

// Hardware serial is also possible!
//  HardwareSerial *fonaSerial = &Serial1;

// Use this for FONA 800 and 808s
Adafruit_FONA fona = Adafruit_FONA(FONA_RST);
// Use this one for FONA 3G
//Adafruit_FONA_3G fona = Adafruit_FONA_3G(FONA_RST);

uint8_t readline(char *buff, uint8_t maxbuff, uint16_t timeout = 0);

uint8_t type;

void setup() {
  while (!Serial);

  Serial.begin(115200);
  Serial.println(F("FONA basic test"));
  Serial.println(F("Initializing....(May take 3 seconds)"));

  fonaSerial->begin(4800);
  if (! fona.begin(*fonaSerial)) {
    Serial.println(F("Couldn't find FONA"));
    while (1);
  }
  type = fona.type();
  Serial.println(F("FONA is OK"));
  Serial.print(F("Found "));
[... 788 lines cut ...]

      //Serial.print(c, HEX); Serial.print("#"); Serial.println(c);

      if (c == '\r') continue;
      if (c == 0xA) {
        if (buffidx == 0)   // the first 0x0A is ignored
          continue;

        timeout = 0;         // the second 0x0A is the end of the line
        timeoutvalid = true;
        break;
      }
      buff[buffidx] = c;
      buffidx++;
    }

    if (timeoutvalid && timeout == 0) {
      //Serial.println(F("TIMEOUT"));
      break;
    }
    delay(1);
  }
  buff[buffidx] = 0;  // null term
  return buffidx;
}

AustinGregory223, 2018-06-24:
It says to add your APN details but i am not sure where

AustinGregory223, 2018-06-24:
I am trying to follow this guys tutorial
http://www.instructables.com/id/LTE-NB-IoT-Shield-for-Arduino/