espressif/arduino-esp32 #433: How to select pins_arduino.h when using as an esp32-Idf component

NickWaterton, 2017-06-11:
I'm using Arduino-esp32 as a component (in eclipse), but I'm not sure how to set the correct pins_arduino.h for my board.

I've included the correct pins_arduino.h for my board, but the pin mapping is still wrong. For I2C SCL is still showing as pin 19 (I need 22). I can manually override this in the Arduino libraries, but this is a cludge.

Is there a proper, programatic way of doing this?

me-no-dev, 2017-06-21:
IDF uses the Dev Board config by default. I'll leave this open as it's a good idea to make it configurable in some way :)

aureq, 2018-10-11:
Has there been any progress on this please ? I ran into the same issue and I'm trying to find a clean path to resolve this. Thank you.

closed 2025-01-16 by Parsaabasi