espressif/arduino-esp32 #434: Odd problem with eclipse

NickWaterton, 2017-06-11:
I'm using Arduino-esp32 as a component in eclipse. I know this isn't really an issue here, but was hoping for some pointers.

If I run make from the mingGW command line, all builds correctly, if I run it from eclipse, everything compiles, but the linker complains about an undefined reference to setup() and loop() - which are defined in my program (a .cpp file). The make being run from eclipse is exactly the same make as is run from the command line (as far as I can tell).

Once I've built from the command line, I can build/flash from eclipse (unless I make clean).

I'm missing something here, there are many tutorials on how to get esp32 to build in eclipse, and they are all out of date or different. Has anyone seen this before?

I notice that setup() and loop() are defined as extern "C", and I'm defining them in a .cpp file - is that a factor? Do I need some preprocessor's or something?

If I disable the Arduino configuration in menuconfig, I get the same problem with loopTask(), even though I have defined loopTask() (I comment it out normally, and use the built in). This all compiles/links/flashes properly from the command line, it's just eclipse that has the problem.

Any suggestions as to where the problem is? I'm using the Arduino version of eclipse (and if I set up a project as an Arduino type, it works just fine, this is only a problem if I use Arduino-esp32 as an esp32-Idf component).

Thanks.

me-no-dev, 2017-06-21:
Oh this is so strange... and I had this issue at some point, but can not remember what I did to make it work. There was something missing or misconfigured on my end. Do your other IDF projects compile fine? This does seem a bit like linking issue

far5893, 2017-10-03:
Hi can you explai how to install esp32 to slober eclipse ?
I'm using esp8266 and all work right but with esp8266 i have json link  esp32 require git clone?

me-no-dev, 2017-10-04:
yes, you clone it to a folder and have the parent forder set in the settings as location for platform files. works fine from then on

closed 2018-04-19 by everslick