espressif/arduino-esp32 #435: Two Stop Bit RX Problem

claudeheintz, 2017-06-12:
Serial settings with two stop bits do not work properly with UART RX.

Is it possible to modify uartBegin in esp32-hal-uart.c to include the new workaround for two stop bits in the latest IDF, something like this.

324  uartFlush(uart);
325  uartSetBaudRate(uart, baudrate);
326  UART_MUTEX_LOCK();
327  
328  uart->dev->conf0.val = config;
329
330  //begin mod ****************************************************
331  // workaround for two stop bits hardware issue
332  // if 2 stop bits, set conf0.stop_bit_num to 1 and set conf delay 1 bit
333  // see https://github.com/espressif/esp-idf/blob/master/components/driver/uart.c  lines 118-127
334
335	 if ( uart->dev->conf0.stop_bit_num == TWO_STOP_BITS_CONF) {
336	    uart->dev->conf0.stop_bit_num = ONE_STOP_BITS_CONF;
337	    uart->dev->rs485_conf.dl1_en = 1;
338	 }
339
340	 //end mod   ****************************************************
341
342	 UART_MUTEX_UNLOCK();

me-no-dev, 2017-06-21:
can you submit a PR for this? I can also do it manually

claudeheintz, 2017-06-22:
I haven't forked/done a pull request.  But, I'll try it later today

lonerzzz, 2017-08-15:
@claudeheintz I believe that this fix is in place in the current code. Can you confirm? Is the issue fixed for you?

claudeheintz, 2017-08-16:
Hi Jason,

I am using a modified version of the esp32-hal-uart.c file in order to use a custom ISR.  However, except for that, the file is identical to esp32-hal-uart.c in Arduino core, including the two stop bit fix.

Thanks,

-Claude Heintz 
 

On Tue, 15 Aug 2017 15:23:14 +0000 (UTC), "Jason K."  wrote:
 
@claudeheintz I believe that this fix is in place in the current code. Can you confirm? Is the issue fixed for you?

claudeheintz, 2017-08-17:
Hi,

In testing the latest ESP32 Arduino Core, I notice that the uartEnd function was changed.  But, this causes a problem if the uart is re-used because the queue is deleted but not set to NULL.  I'd propose the following fix:

void uartEnd(uart_t* uart) {
    if(uart == NULL) {
        return;
    }

    UART_MUTEX_LOCK();
    if(uart-&gt;queue != NULL) {
        uint8_t c;
        while(xQueueReceive(uart-&gt;queue, &amp;c, 0));
        vQueueDelete(uart-&gt;queue);
//begin mod   ****************************************************
        uart-&gt;queue = NULL;
//end mod   ****************************************************
    }

    uart-&gt;dev-&gt;conf0.val = 0;

    UART_MUTEX_UNLOCK();

    uartDetachRx(uart);
    uartDetachTx(uart);
}
 

Thanks,

-Claude Heintz

On Tue, 15 Aug 2017 15:23:14 +0000 (UTC), "Jason K."  wrote:
 
@claudeheintz I believe that this fix is in place in the current code. Can you confirm? Is the issue fixed for you?

me-no-dev, 2017-08-17:
good catch @claudeheintz

closed 2017-08-17 by me-no-dev