espressif/arduino-esp32 #436: FreeRTOS CPU halted

lutfisan, 2017-06-12:
i try FreeRTOS with this code but get CPU halted
```
// define two tasks for Blink & AnalogRead
void TaskAnalogReadA0( void *pvParameters );
void TaskAnalogReadA3( void *pvParameters );

void setup() {
  Serial.begin(115200);
  
  while (!Serial) {
    ; 
  }

  xTaskCreate(
    TaskAnalogReadA0
    ,  "AnalogReadA0"   // A name just for humans
    ,  128  // This stack size can be checked & adjusted by reading the Stack Highwater
    ,  NULL
    ,  2  // Priority, with 3 (configMAX_PRIORITIES - 1) being the highest, and 0 being the lowest.
    ,  NULL );

  xTaskCreate(
    TaskAnalogReadA3
    ,  "AnalogReadA3"
    ,  128  // Stack size
    ,  NULL
    ,  1  // Priority
    ,  NULL );

  // Now the task scheduler, which takes over control of scheduling individual tasks, is automatically started.
}

void loop()
{
  // Empty. Things are done in Tasks.
}

void TaskAnalogReadA0(void *pvParameters)  // This is a task.
{
  (void) pvParameters;

  for (;;) // A Task shall never return or exit.
  {
    int sensorValueA0 = analogRead(A0);
    Serial.println(sensorValueA0);
    vTaskDelay(100); 
  }
}

void TaskAnalogReadA3(void *pvParameters)  // This is a task.
{
  (void) pvParameters;
 
  for (;;)
  {
    int sensorValueA3 = analogRead(A3);
    Serial.println(sensorValueA3);
    vTaskDelay(10); 
  }
}
```

[image]
[image]

```
Decoding 17 results
0x40080000: _iram_start at ?? line ?
0x40080034: _iram_start at ?? line ?
0x40085cda: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 188 (discriminator 1)
0x400821c8: _malloc_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c line 26
0x40085cda: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./list.c line 188 (discriminator 1)
0x4008520b: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4420
:  (inlined by) vTaskPlaceOnEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 2857
0x400840a3: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1992
0x400d0c94: uartWriteBuf at C:\Users\lutfi\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-uart.c line 277 (discriminator 1)
0x400d0840: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\lutfi\Documents\Arduino\hardware\espressif\esp32\cores\esp32/HardwareSerial.cpp line 47
0x400d0959: Print::write(char const*) at C:\Users\lutfi\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Print.cpp line 85
0x400d09b7: Print::printNumber(unsigned long, unsigned char) at C:\Users\lutfi\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Print.cpp line 85
0x400d0a37: Print::print(long, int) at C:\Users\lutfi\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Print.cpp line 85
0x400d0a4b: Print::print(int, int) at C:\Users\lutfi\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Print.cpp line 85
:  (inlined by) Print::println(int, int) at C:\Users\lutfi\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Print.cpp line 196
0x400d05c4: TaskAnalogReadA0(void*) at D:\PENS\ta\arduino esp32\_20170601-freertos1/_20170601-freertos1.ino line 81 (discriminator 1)
```

me-no-dev, 2017-06-21:
This is an issue with Serial in some certain case. Since you did not specify where to run the tasks they prolly got in a weird step over each other and the serial driver went crazy. Try different delay in both tasks and see if that "fixes" the behavior.

lonerzzz, 2017-08-30:
@lutfisan Have you resolved this issue? Were you able to use the approach from me-no-dev?

lutfisan, 2017-08-31:
yes I have solve this problem, 
i change xTaskCreate() to xTaskCreatePinnedToCore()
source : https://www.youtube.com/watch?v=i8ffEbyU-7w

closed 2017-08-31 by copercini