espressif/arduino-esp32 #437: usage of Uart1 and Uart2

eroom1966, 2017-06-12:
Hi

I have been struggling to get the 2 UART interfaces working reliably, and have now seen a strange issue which seems to be related to my core issue

I have an external GPS receiver running at 9600 baud configured as follows 
HardwareSerial SerGPS(1);
...
SerGPS(9600, SERIAL_8N1, 33, 32);

All works well, I stream the data from the GPS device and no issues, I can read and parse the messages

If I reconfigure to use UART2 (everything else remains the same), thus
HardwareSerial SerGPS(2);
...
SerGPS(9600, SERIAL_8N1, 33, 32);

Now using UART2 it becomes very unstable - the checksums on the messages fail and does not work reliably, this is very confusing from a S/W perspective, nothing has changed, only the internal configuration of the IO to UART mapping.

I will try to reduce to a simple testcase and share on this forum, but I am wondering has anybody else experienced a similar issue when trying to use UART2

Kind Regards
E

copercini, 2017-06-12:
Quite strange, For me it works fine:

HardwareSerial SerialGPS(2);
SerialGPS.begin(9600,SERIAL_8N1,32,33);

eroom1966, 2017-06-12:
@claudeheintz 
Hi Claude, 
That is usefule information, 
Can you provide a link to the reference that UART2 is muxed to GPIO 16 & 17 ?
Regards
E

eroom1966, 2017-06-12:
@copercini 
When you say 
> it works fine
Can you tell me what you mean ?

I am guessing that individual characters are incorrect due to the fact the calculated and provided checksums are differing.

Regards 
E

copercini, 2017-06-12:
@eroom1966 
I just did some superficial tests with a SIM808, using AT commands

Using Serial2 or Serial1 with pins 33 and 32 the ESP32 send AT commands and SIM808 receive and respond it without any problems.

Did you tried with **.begin** in SerialGPS.begin(9600,SERIAL_8N1,33,32); ?

eroom1966, 2017-06-12:
@claudeheintz 
Follow up
https://www.espressif.com/sites/default/files/documentation/esp_wroom_32_datasheet_en.pdf
This document indicates that UART0 & UART1 can be muxed to any GPIO pins

eroom1966, 2017-06-12:
@copercini 
> Did you tried with .begin in SerialGPS.begin(9600,SERIAL_8N1,33,32); ?
my posting was a typo, this is in fact what I have 
sorry for the confusion
E

claudeheintz, 2017-06-12:
In the technical reference manual it refers to the ability to connect pins in two ways.  I'm not sure I completely understand.  But, there is through the GPIO mux and through the I/O mux.  With the latter, only specific signals can be assigned to a pin.  The technical reference manual (https://espressif.com/en/support/download/documents?keys=&field_type_tid%5B%5D=13) indicates that using the I/O mux has better performance for high frequency signals (whatever that means).

I'm suggesting that if there is a corruption issue, even at 9600 baud, that using the pins specifically designated for U2TX and U2RX (rather than using the GPIO mux to assign any random pins) may make a difference.

eroom1966, 2017-06-12:
@claudeheintz 

> I'm suggesting that if there is a corruption issue, even at 9600 baud, that using the pins specifically designated for U2TX and U2RX (rather than using the GPIO mux to assign any random pins) may make a difference.

This is definitely worth investigating, thanks for the pointer
Regards 
E

closed 2017-06-12 by eroom1966

llewellynnu, 2017-08-25:
@eroom1966 Did using the Pins designated for UART2 Solve the issue you were facing? 
I am asking since I am facing the exact same issue. 

Regards

eroom1966, 2017-08-29:
@llewellynnu 
Yes I used the designated pins
Regards
E

llewellynnu, 2017-08-29:
@eroom1966 Thanks a lot. 
The only issue with using the designated pins is that ESP32-WROVER module GPIO 16 and 17 are not to be connected as its connected to PSRAM.

claudeheintz, 2017-08-30:
Here is a useful article about re-assigning the pins for the serial ports on ESP32 boards:
http://hackaday.com/2017/08/17/secret-serial-port-for-arduinoesp32/

Vorms, 2018-01-05:
I use the UART 2 on pin 16 and 17 and it is working well.
When I setup the UART1  with 
HardwareSerial MP3Player
MP3Player.begin(9600, SERIAL_8N1, MP3PLAYER_RX, MP3PLAYER_TX);

The UART2 will be miss fonctioning.

Best regards

Thierry Vorms
vormsty@gmail.com

tgmaxx, 2018-02-28:
UART 1 and 2 not working. Seems to me the ESP32 is a POS.

Vorms, 2018-02-28:
Whats about Expressif response?
Whats about serius irremote library using rmt register?

Best regards
Thierry Vorms

Thierry Vorms Saint-Cyrille-de-Lessard G0R 2W0