espressif/arduino-esp32 #451: touchRead(T1); and touchRead(T2); always give low values.

dmarkey, 2017-06-17:
Perhaps I'm doing it wrong?

`int threshold = 40;

void setup() {
  Serial.begin(115200);

}

void loop(){
  int read = touchRead(T0);
  if (read < threshold){
    Serial.println("T0 touched");
    Serial.println(read);
  }
  delay(1000);
  read = touchRead(T1);
  if (read < threshold){
    Serial.println("T1 touched");
    Serial.println(read);
  }
  delay(1000);
  read = touchRead(T2);
  if (read < threshold){
    Serial.println("T2 touched");
    Serial.println(read);
  }
  delay(1000);
  read = touchRead(T3);
  if (read < threshold){
    Serial.println("T3 touched");
    Serial.println(read);
  }
  delay(1000);
  read = touchRead(T4);
  if (read < threshold){
    Serial.println("T4 touched");
    Serial.println(read);
  }
  delay(1000);
  read = touchRead(T5);
  if (read < threshold){
    Serial.println("T5 touched");
    Serial.println(read);
  }
  delay(1000);
  read = touchRead(T6);
  if (read < threshold){
    Serial.println("T6 touched");
  }
  delay(1000);
  read = touchRead(T7);
  if (read < threshold){
    Serial.println("T7 touched");
    Serial.println(read);
  }

}
`

dmarkey, 2017-06-17:
Output (not touching the module):

T2 touched
0
T1 touched
1
T2 touched
0
T1 touched
1
T2 touched
0
T1 touched
1
T2 touched
0
T1 touched
1
T2 touched
0
T1 touched
1
T2 touched
0
T1 touched
1
T2 touched
0
T1 touched
1
T2 touched
0
T0 touched

me-no-dev, 2017-06-21:
yup ;) bootstrap pins that have been pulled by resistors externally ;) do not use those

closed 2017-06-21 by me-no-dev