espressif/arduino-esp32 #453: UDP send packages crashes

ImanolGo, 2017-06-18:
I'm Using Sparkfun ESP32 Thing and sending simple udp messages to a given IP Address. 

```
 //Send a packet
      udp.beginPacket(ipRemote,udpPort);
       for(int i = 0; i< NUM_TOUCH_PINS; i++)
      {
         //Serial.print(touchRead(touchPins[i]));  // get value using Ti
          //Serial.print(" ");
          udp.print(touchRead(touchPins[i]));  // get value using Ti
          udp.print(" ");
      }  
      udp.endPacket();
```

The connection works fine and it's starts sending packages right away. But after a few seconds of sending different values it always crashes, showing:

`
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
`

I'm not even reading udp messages nor using udp.begin(). 

Maybe you have some insights or workaround to solve this issue.

ImanolGo, 2017-06-20:
I changed the board to the HIMALAYA ESP32 Dev Board and that solved the issue.

me-no-dev, 2017-06-21:
that does not explain the reason though... interesting. I have never seen this error

lonerzzz, 2017-08-25:
@ImanolGo Is this issue still present in the latest software version? Are you able to confirm?

ImanolGo, 2017-08-29:
@lonerzzz I returned the Sparkfun ESP32 modules so unfortunately I cannot test it against the newest software version.

closed 2017-10-06 by copercini