espressif/arduino-esp32 #455: Can't connect to Erlang MQTT

architmuchhal12, 2017-06-19:
Hello,

I am using Arduino Espressif ESP32 along with DHT22 to connect to Erlang MQTT http://emqtt.io/.
On the serial console, I am getting an output as -

```
Attempting MQTT Connection...failed, rc= -4
 try again in 5 seconds
Attempting MQTT Connection...failed, rc= -4
 try again in 5 seconds
Attempting MQTT Connection...failed, rc= -4
 try again in 5 seconds
```

This is the code that I have used -

```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>
#include <PubSubClient.h>
#include "DHT.h"

#define wifi_ssid "****"
#define wifi_password "*****"

#define mqtt_server "******"
#define mqtt_user "*****"
#define mqtt_password "*****"

#define DHTPIN 5          //define as DHTPIN the Pin 3 used to connect the Sensor
#define DHTTYPE DHT22     // DHT 22  (AM2302), AM232
DHT dht(DHTPIN, DHTTYPE); //create an instance of DHT

#define humidity_topic "sensor/humidity"
#define temperature_topic "sensor/temperature"

WiFiClient espClient;
PubSubClient client(espClient);

void setup() {
  Serial.begin(115200);
  dht.begin();
  Serial.println("DHT Started");
  setup_wifi();
  client.setServer(mqtt_server, 8883);
}

void setup_wifi() {
  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to: ");
  Serial.println(wifi_ssid);

  WiFi.begin(wifi_ssid, wifi_password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT Connection...");
    // Attempt to connect
    // If you do not want to use a username and password, change next line to
    // if (client.connect("ESP32Client")) {
    if (client.connect("ESP32Client", mqtt_user, mqtt_password)) {
      Serial.println("connected");
    } else {
      Serial.print("failed, rc= ");
      Serial.println(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

bool checkBound(float newValue, float prevValue, float maxDiff) {
  return !isnan(newValue) &&
         (newValue < prevValue - maxDiff || newValue > prevValue + maxDiff);
}

long lastMsg = 0;
float temp = 0.0;
float hum = 0.0;
float diff = 1.0;

void loop() {
  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  long now = millis();
  if (now - lastMsg > 1000) {
    lastMsg = now;

    float newTemp = dht.readTemperature();
    float newHum = dht.readHumidity();

    if (checkBound(newTemp, temp, diff)) {
      temp = newTemp;
      Serial.print("New temperature: ");
      Serial.println(String(temp).c_str());
      client.publish(temperature_topic, String(temp).c_str(), true);
    }

    if (checkBound(newHum, hum, diff)) {
      hum = newHum;
      Serial.print("New humidity: ");
      Serial.println(String(hum).c_str());
      client.publish(humidity_topic, String(hum).c_str(), true);
    }
  }
}

```
Is the **port number 8883** correct for MQTT over SSL?
Please suggest me the changes!
Thank you

copercini, 2017-06-19:
If you want a non SSL connection, choose WiFiClient and port 1883
If you want a SSL try with WiFiClientSecure and port 8883

architmuchhal12, 2017-06-19:
Thanks a lot!I was using a wrong library altogether!
It worked!

closed 2017-06-19 by architmuchhal12