espressif/arduino-esp32 #456: CPU Halted when using softAP and WiFiClient

androidcho, 2017-06-20:
Ok, so it seems that's not just me with this issue. I've searched google and issues in this repo and everyone seems to fix it's problem in a different way.
Everything worked on me until something updated(not sure what, but as I can remember it was the PlatformIO core). Tryed reinstalling the "Espresiff 32" package, but it doesn't help.

Everything works, softAP creates AP and I can connect to it, it won't disconnect. At the moment I try to communicate with it by WiFiClient library (serving a webpage), it halts.
I think my problem is with messed-up partition table, but I can't do "erase_flash", it just does nothing..
Here's the debug log:
==========================
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400839a3  PS      : 0x00060833  A0      : 0x80083b00  A1      : 0x3ffcd0d0
A2      : 0x3ffd7690  A3      : 0x00000000  A4      : 0xff000000  A5      : 0x80ffffff
A6      : 0x3ffc3820  A7      : 0xff000000  A8      : 0x00e006c0  A9      : 0x00e006c0
A10     : 0x3ffc13bc  A11     : 0x00000030  A12     : 0x3ffcc140  A13     : 0x00000000
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00e006c0  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x400839a3:0x3ffcd0d0 0x40083b00:0x3ffcd0f0 0x40086b9b:0x3ffcd110 0x40086bdc:0x3ffcd130 0x4008202c:0x3ffcd150 0x4000beb2:0x3ffcd170 0x400d1460:0x3ffcd190 0x400d14ab:0x3ffcd1b0 0x400d1591:0x3ffcd1d0 0x400d1
5ec:0x3ffcd1f0 0x400d0fca:0x3ffcd220 0x400d0c3e:0x3ffcd2a0 0x400ddb24:0x3ffcd2c0

CPU halted.

copercini, 2017-06-20:
Please, download and decode your logs with: https://github.com/me-no-dev/EspExceptionDecoder

androidcho, 2017-06-23:
Here it is:
Decoding 15 results
```
0x400839a3: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1992
:  (inlined by) xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 876
0x400839a3: xPortGetCoreID at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1992
:  (inlined by) xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 876
0x40083b00: xQueueGenericSendFromISR at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1992
0x40086b9b: rtc_clk_cal at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/esp32/rtc_time.c line 125
0x40086bdc: rtc_time_get at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/esp32/rtc_time.c line 129
0x4008202c: lock_release_generic at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./locks.c line 189
0x400d1460: __ltdf2 at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S line 1785
0x400d14ab: __unorddf2 at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S line 1828
0x400d1591: std::get_new_handler() at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_handler.cc line 65
0x400d0fca: __muldf3_aux at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S line 679
0x400d0c3e: uartBegin at C:\Users\ADMIN\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-uart.c line 212
0x400ddb24: intrusive_list ::push_back(nvs::Page*) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/intrusive_list.h line 126
:  (inlined by) nvs::PageManager::load(unsigned int, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_pagemanager.cpp line 40
```

me-no-dev, 2017-06-23:
This looks much like power issue to me. Try external power + maybe larger cap and see if things will improve. ESP32 spikes to 400-500 mA a couple of times while bringing the radio up

androidcho, 2017-06-23:
But it worked the day before.. Even changed the USB port, the microUSB cable is the same..
What about this - `0x400ddb24: intrusive_list ::push_back(nvs::Page*) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/intrusive_list.h line 126
:  (inlined by) nvs::PageManager::load(unsigned int, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_pagemanager.cpp line 40`
Isn't it about the internal storage? How to format the NVS?

me-no-dev, 2017-06-23:
you can use the included esptool to erase everything.
```bash
esptool.exe --chip esp32 --port $SERIAL_PORT --baud 115200 --before default_reset --after hard_reset erase_flash
```

androidcho, 2017-06-24:
Ok, so after some lurking around to find that esptool included in the platformio IDE is not the same as the official one(maybe old version), I've sucessfully erased the flash. After that, compiled and uploaded, the new error log is:
```
Decoding 9 results
0x4008705c: esp_rom_spiflash_read_data at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./spi_flash_rom_patch.c line 412
0x40082594: spi_flash_get_chip_size at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/./flash_ops.c line 99
0x400d1659: quorem at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/dtoa.c line 118 (discriminator 1)
0x400d1214: __divdf3_aux at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S line 1381
0x400d0cda: __adddf3 at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S line 174
0x400de100: __udivmoddi4 at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libgcc/libgcc2.c line 1115
:  (inlined by) __udivdi3 at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libgcc/libgcc2.c line 1288
```
I'm using `WiFi.setAutoConnect(true); WiFi.begin();` to try to connect to the last valid network, and if that fails, code will put the WiFi into softAP mode and start serving HTTP clients.
What about clonning this repo into PlatformIO's ESP32 dir instead of the official one?

androidcho, 2017-06-25:
Problem solved!
For some reason, cleaning the project done the job. 
I've stripped down the problem to a point where I process a GET request(and so working with strings and chars). So it's probably not a WiFi problem.
SOLUTION: Open PlatformIO terminal and run
`platformio run --target clean`
Then rebuild the project and upload. Voila, it works!

closed 2017-06-25 by androidcho