espressif/arduino-esp32 #457: ESP32 WIFI LED CONTROL.....HORRIBLE FACE

sajeelahmed34, 2017-06-20:
The issue is very much DANGEROUS! Everyone is requested to use their skills to solve this problem. Lets start the story…I used code from somewhere else to control led from esp32 over wifi using Arduino IDE but I have stucked, I am able to control led when my esp32 and my phone/laptop are at the same network BUT when my esp32 is connected to my local home network and my friend trying to control it from other part of the world, I am in Pakistan and he is in USA, so we have different networks a Timed out problem occur. I am totally desperate on this situation. Is am doing some configuration mistakes or its my code that creates problems. ?? I have attached the link below from where I used code. I hope to get helpful resource from you. Thanks :) ;) https://www.youtube.com/watch?v=uGUrgOV_6ro

copercini, 2017-06-20:
Hahahahah Don't panic!

You probably are using a local IP only, like 192.168.whatever

For your friend control your LED from the other side of globe, you should you a cloud version https://github.com/marcoschwartz/aREST/blob/master/examples/ESP32_cloud/ESP32_cloud.ino
https://arest.io/cloud

sajeelahmed34, 2017-06-20:
> master/examples/ESP32_cloud/ESP32_cloud.ino

Hey. Thanks for increasing my hope level. Well, in this code somewhere I
need "UNIQUE ID" so what to put in this ??

Thanks once more'

SajeelAhmed

copercini, 2017-06-20:
It's the value which you will use in the control link: https://github.com/marcoschwartz/aREST#cloud-access-esp8266

me-no-dev, 2017-06-21:
Horrible joke incoming: So... you are in Pakistan, your friend is in USA, your issue is DANGEROUS... are you sure you are trying to switch a light bulb and not something else :D

sajeelahmed34, 2017-06-21:
Haha! I like your sense of humor. Well first of all "THANKS" for your kind
help, you have really solved my problem!
Actually, I am working with my friend to innovate an electronic project in
which we have to use APA102 rgb leds and to control them from Wifi and BLE
device. We are in initial stage and just to check whether we are trying to
glow led or not. I have stucked because I have not worked on esp8266/esp32
before in life. I have no any idea about networking as well but I want to
learn things and get hands on experience on this device. The code you have
provided me open ways for me to think out of box. Thanks! Our purpose is to
control rgb colors from BLE and wifi from anywhere in the world and because
of wifi we called we are globally communicated.

Today, I am working on eclipse IDE because Arduino does not support BLE of
esp32. Have you worked on eclipse? Do you know what is MQTT, OpenHab2. ??
Is this provides me cloud storage or not ??

Sajeel Ahmed

lonerzzz, 2017-08-25:
@sajeelahmed34 Is this issue resolved? Can we close it?

copercini, 2017-08-31:
@lonerzzz As I remember this issue was solved then @sajeelahmed34 delete all his posts and change the title, lol

closed 2017-08-31 by copercini