espressif/arduino-esp32 #459: SPIFFS and Preferences

hellodb, 2017-06-20:
Hello, thank you for your project. When will the SPIFFS library be available to Arduino?
Also, what is the maximum capacity of Preferences?

me-no-dev, 2017-06-21:
SPIFFS is to be implemented soon ;) Preferences is not a large storage. The partition for it is 20K but there are other things stored inside also (wifi, system settings)

hellodb, 2017-06-21:
Thanks for the reply and look forward to SPIFFS

lbernstone, 2017-07-13:
Bump. Any ETA from Ivan on SPIFFS?

lonerzzz, 2017-07-15:
A temporary version is available as detailed here (https://github.com/espressif/arduino-esp32/issues/411)

me-no-dev, 2017-07-17:
almost there ;) sorry for the delay but there were some unforseen complications with other modules that I am working on. SPIFFS is definitely the next thing to implement into IDF and VFS

me-no-dev, 2017-09-12:
SPIFFS is in

closed 2017-09-12 by me-no-dev