espressif/arduino-esp32 #462: ESP32 I2C not setting up

architmuchhal12, 2017-06-23:
Hello,

I am using an ADXL345 sensor with ESP32. The issue I am facing is I2C is not being initialized on the Sparkfun ESP32 Thing.
I used the following pins for I2C on ESP32
# SDA - Pin 21
# SCL - Pin 22
I have tried the same code with ESP8266 (NodeMCU 1.0) and it works just fine without any issues.
I have also attached the code and if anyone could explain to me why this is happening, it'd be great!
Thanks!

Code:
```
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_ADXL345_U.h>

// ADXL345 I2C address is 0x53(83)
#define Addr 0x53

//#define SDA_PIN 21
//#define SCL_PIN 22

void setup() {

  Serial.begin(115200);
  Serial.println(" ");
  Serial.println("Serial port initialized ");

  // Initialise I2C communication as MASTER
  Wire.begin();
  // Start I2C Transmission
  Wire.beginTransmission(Addr);
  // Select bandwidth rate register
  Wire.write(0x2C);
  // Normal mode, Output data rate = 100 Hz
  Wire.write(0x0A);
  // Stop I2C transmission
  Wire.endTransmission();

  // Start I2C Transmission
  Wire.beginTransmission(Addr);
  // Select power control register
  Wire.write(0x2D);
  // Auto-sleep disable
  Wire.write(0x08);
  // Stop I2C transmission
  Wire.endTransmission();

  // Start I2C Transmission
  Wire.beginTransmission(Addr);
  // Select data format register
  Wire.write(0x31);
  // Self test disabled, 4-wire interface, Full resolution, Range = +/-2g
  Wire.write(0x08);
  // Stop I2C transmission
  Wire.endTransmission();
  delay(300);
  Serial.println("I2C initialized !");
}

void loop() {
  readAcclData();
  delay(3000);
}

void readAcclData()
{
  unsigned int data[6];
  for (int i = 0; i < 6; i++)
  {
    // Start I2C Transmission
    Wire.beginTransmission(Addr);
    // Select data register
    Wire.write((50 + i));
    // Stop I2C transmission
    Wire.endTransmission();

    // Request 1 byte of data
    Wire.requestFrom(Addr, 1);

    // Read 6 bytes of data
    // xAccl lsb, xAccl msb, yAccl lsb, yAccl msb, zAccl lsb, zAccl msb
    if (Wire.available() == 1)
    {
      data[i] = Wire.read();
    }
  }

  // Convert the data to 10-bits
  int xAccl = (((data[1] & 0x03) * 256) + data[0]);
  if (xAccl > 511)
  {
    xAccl -= 1024;
  }
  int yAccl = (((data[3] & 0x03) * 256) + data[2]);
  if (yAccl > 511)
  {
    yAccl -= 1024;
  }
  int zAccl = (((data[5] & 0x03) * 256) + data[4]);
  if (zAccl > 511)
  {
    zAccl -= 1024;
  }
  // Output data to serial monitor
  Serial.print("Acceleration in X-Axis is : ");
  Serial.println(xAccl);
  delay(1000);
  Serial.print("Acceleration in Y-Axis is : ");
  Serial.println(yAccl);
  delay(1000);
  Serial.print("Acceleration in Z-Axis is : ");
  Serial.println(zAccl);
  delay(1000);
  Serial.println("Please wait...");
  Serial.println("**********************");
  delay(500);
}
```

Serial Output:

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
 
Serial port initialized

```

Jorgen-VikingGod, 2017-06-23:
try to register the I2C by passing the pins:
```Wire.begin(SDA_PIN, SCL_PIN); // 21 & 22 are default on ESP32```

If it still loop within the Wire.begin(...) - try to add 10k pull-ups on SDA and SCL lines.
I had a similar issue on my setup and after add 10k pull-ups to the I2C pins on my master (ESP32) it worked perfect.

architmuchhal12, 2017-06-23:
Thanks, I did that and it was at least getting initialized. But, I think, the readings are probably some garbage values.

Any suggestions?

The serial output after adding pull ups - 

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
 
Serial port initialized 
I2C initialized !
Acceleration in X-Axis is : 0
Acceleration in Y-Axis is : 1073482784
Acceleration in Z-Axis is : -2146597018
Please wait...
**********************
Acceleration in X-Axis is : 0
Acceleration in Y-Axis is : 1073482784
Acceleration in Z-Axis is : -2146597012
Please wait...
**********************
```

closed 2017-07-13 by architmuchhal12

ytemelli, 2018-08-01:
void setup() {
  pinMode(SDA,INPUT_PULLUP);
  pinMode(SCL,INPUT_PULLUP);