espressif/arduino-esp32 #464: SHT21 I2C unable to read

kyjak, 2017-06-24:
I'm trying to get to work the Sensirion SHT21 I2C sensor on ESP32 without luck.
Using pins 21,22 as wire interface.
I tried all libraries I could find for SHT2x.
On arduino uno the code works, ESP does not.
It reads wrong sensor data = bogus.
It might be connected to the hal i2c library of esp32.
Do not have oscilloscope to show log :(

architmuchhal12, 2017-06-26:
I am having the same issue #462 with the ADXL345 sensor. Works well with ESP8266 but fails to initialize I2C with ESP32.

kyjak, 2017-06-26:
I did solve it partially by using this library https://github.com/markbeee/SHT21/tree/master/examples/SHT21_Demo
however, that does not answer, why we cannot use ANY library, I still see an I2C issue with esp's.
Something is wrong probably with the timings.

me-no-dev, 2017-06-26:
It's not timings, it's something with the slaves themselves. ESP32's I2C bus is way more complex than the one found in AVR arduinos, so something somewhere is going wrong ;)
So I tested with 8 slaves on the I2C bus and all but one responded every time. That one was the SHT21.

architmuchhal12, 2017-06-26:
I have switched to SPI communication for time being and it is working okay. I am using just one slave (ADXL345) and still, it does not respond. 
So, is there a solution to counter ESP32's I2C @me-no-dev ?

kyjak, 2017-06-26:
Hmm, even if I have only 1 sensor, so it's not a slave?

stonie08, 2017-07-20:
Are you using the SHT21 in the "hold master mode"? I'm not using the arduino core, but on my esp32 I get a correct reading from the SHT21 if I'm using the "no-hold-master" mode.
So for me the ESP32 ignores that the SHT21 is holding down the SCL line to force into a wait state.

stickbreaker, 2017-11-16:
@kyjak @architmuchhal12 @stonie08 I have posted a fix for the I2C subsystem that may be at the root of your problems,  See #839 .

@stonie08 The ESP32's I2C subsytem enforces a bit TIMEOUT, at maximum it is around 12ms or 2^20 cpu clocks @ 80mhz.  How long is the hold period during this conversion?  If it is anywhere near 5ms the ESP32 will react badly.

Chuck.

kyjak, 2017-11-17:
Thanks, will try out
And let you know ;)
Kyjak

manymany-many, 2018-06-19:
Thank you stickbreaker, I installed you version of [arduio-esp32](https://github.com/stickbreaker/arduino-esp32) and I can read my SHT21, it was almost not working with espressif.

stickbreaker, 2018-06-19:
@manymany-many your welcome.

Chuck.

copercini, 2018-07-01:
The I2C core was changed to @stickbreaker code officially after https://github.com/espressif/arduino-esp32/commit/13dcfe5c13bc882f8923d137ed604aa65b6db057 (thanks @stickbreaker), Let's close issues with old code and if have problems with the new code, open new issues =)

closed 2018-07-01 by copercini