espressif/arduino-esp32 #469: Equivalent to ESP.getVcc() on ESP8266?

kaofishy, 2017-06-28:
Hi,

I'm wondering if there is an equivalent to the ESP.getVcc() function that's available in the ESP8266 Arduino core. I see that in the ROM library there is rom_phy_get_vdd33() but the name suggests it's used by the (WiFi? BT) PHY so would it be safe to use when WiFi or BT is on? From some quick testing it seems to work the same as ESP.getVcc() from ESP8266.

Tony

rmhomecouk, 2017-08-28:
I don't suppose you have an example for this please?

rummyr, 2017-10-04:
Corking!
extern "C" int rom_phy_get_vdd33();

Using arduino IDE I find that this call returns valid values if WiFi or Bluetooth are "ON"...
Not tried it in anger yet, so can't say if it interferes with proper operation though it seems to only take ~45 microSec, so it's not doing much.

FWIW if you don't need Wifi/BT then btStart and btStop seem to be MUCH faster than Arduino WiFi.mode(WIFI_STA) and WiFi.mode(WIFI_OFF), though that could well be Arduino getting in the way.

I believe that even after turning bt or wifi off will still consume a higher baseline power than if you never turn them on.

AusUser5, 2018-02-03:
Do you have an example? on my esp32 I looped the wifi scanner sketch to turn on wifi and still get a reading of 0

kosso, 2018-03-17:
I'm confused why there is no ESP32 equivalent to `ESP.getVcc()`

This seems like something many people would need.

Or am I missing something obvious why the ESP32 doesn't have this basic method?

ldab, 2018-10-01:
Especially for battery powered setups.

closed 2019-07-30 by me-no-dev

marcelstoer, 2019-07-30:
@me-no-dev closing because of what exactly?

atanisoft, 2019-07-30:
@marcelstoer unsupported as there is no built-in functionality for this.

AndreKR, 2019-07-30:
For people who come here from searches, we should mention the equivalent code:
```c
extern "C" int rom_phy_get_vdd33();
float voltage = ((float)rom_phy_get_vdd33()) / 1000;
```

atanisoft, 2019-07-30:
@AndreKR Unfortunately that function is not exposed in the supported APIs and is not guaranteed to return any valid data.

AndreKR, 2019-07-30:
It does return valid data on all ESP32 I own.

I don't think it matters if it's official or not, it's not like the Espressif documentation is very accurate.

atanisoft, 2019-07-30:
@AndreKR while it may work on the boards you have there is absolutely no guarantee that it will always work. It is not recommended to use internal / non-exposed APIs for any purpose...

nutfrey, 2020-02-01:
Obviously there must be some way to read this, otherwise how are brownouts detected?

paolovr70, 2020-03-04:
Unfortunately sometime it works in other cases not. For instance, in my case when ESP32  is connected to wifi it reports 0.5v and when it is disconnect volts are is 5.6v .

`extern "C" int rom_phy_get_vdd33();
float voltage = ((float)rom_phy_get_vdd33()) / 1000;
`

TheRealDJ, 2020-04-03:
Couldn't you just jump Vcc to an analog input are read said analog input?

lbernstone, 2020-04-03:
Run it through a voltage divider to 1V1- but yes, you should read this through the ADC.  The internal meter was unreliable, that is why the function was not supported, and ultimately removed.

jockm, 2021-11-07:
> Couldn't you just jump Vcc to an analog input and read said analog input?

Assuming you have control of the schematic, but if you are trying to add a feature to an existing board that cannot be changed.  Being able to read the internal voltage is a very useful feature if it can be implemented