espressif/arduino-esp32 #470: can't upload with I2C connection

ShawnXuan, 2017-06-28:
Hi,

My hardware connection:
 MPU9250 Breakout --------- ESP32
 VDD ---------------------- 3.3V
 SDA ----------------------- 21
 SCL ----------------------- 22
 GND ---------------------- GND

A fatal error is raised when I upload code:
```
Sketch uses 131278 bytes (10%) of program storage space. Maximum is 1310720 bytes.
Global variables use 10324 bytes (3%) of dynamic memory, leaving 284588 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.0-beta3
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```

I remove `SCL` and `SDA` connection, the code is uploaded successfully. 
then I connect `SCL` and `SDA`, and reset, ESP32 can't boot this time, following is the information from serial monitor:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
user code done
```

the same MPU9250 Breakout board works very well with my other stm32 MCU.

Thanks!

me-no-dev, 2017-06-28:
this has nothing to do with you MPU9250. From the log above, you are unable to even flash the firmware to the device. Try holding the BOOT button while trying to upload

Jorgen-VikingGod, 2017-06-28:
@ShawnXuan maybe you stuck on setup() on the Wire.begin function.
I had similar issue on my setup. After adding 10k pull-ups to SDA and SCL everything was working fine again.

ShawnXuan, 2017-06-29:
Thanks both of you!
I add two pull-ups, it seems better than before, at least I can upload right now, but it keeps reset again and again.
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

```

me-no-dev, 2017-06-29:
What board are you using?

ShawnXuan, 2017-06-30:
[NodeMCU-32S](http://wiki.ai-thinker.com/doku.php/demoboard/esp32/nodemcu_32s)
[image]
I thought it was ESP32-Dev before.

ShawnXuan, 2017-07-01:
solved.
I switched i2c port from pin 21&2()2 to pin 25&26(P0 and P4). 
pin 21&22 are SD_DATA0(SD0) and SD_DATA1(SD1), they can't be connected to an i2c device(SDA, SCL) in my case.

then the code should be modified too:
```c
Wire.begin(0, 4, 400000); //SDA-P0-GPIO0-PIN25, SCL-P4-GPIO4-PIN26 , 400 kHz I2C clock
```

closed 2017-07-01 by ShawnXuan

me-no-dev, 2017-07-02:
I am a bit confused? I use 21 and 22 for I2C and do not have problems.

ShawnXuan, 2017-07-03:
I guess you connect the right pins,` P21` for SDA and `P23` SCL, refer to previous figure for pin definition.
But I connected to `SD0` and `SD1`.
My conclusion is: **using the `GPIO` number instead of `pin` number for args of `Wire.begin()`.**