espressif/arduino-esp32 #472: Public roadmap for ESP32 development

lonerzzz, 2017-06-28:
Hi,

I have been seeing hints of the development strategy in response to numerous enhancement related issues but was wondering if someone was actually tracking the intended development path and strategy. This would help for those looking for specific features and those wanting to contribute. So far I have seen:

1) SPIFFs is coming - not sure when or who is working on it or how it will work in relation to Preferences
2) WebServer is being or will be generalized to work for both the ESP32 and ESP8266 and housed outside either project - again not sure who is working it or where it will be placed
3) Threading is not being worked on so far - options are to use underlying RTOS directly, write a wrapper or make it compliant with modern C++ standards - no strategy chosen at this point
4) I2C is synchronous for now - not sure of any plans to make it asynchronous

These are the issues of interest to me personally. There are likely others that other folks have encountered. I would happily track these efforts if nobody else has a written strategy or roadmap. I would just need some summary input from the different contributors.

Thanks,

Jason

igrr, 2017-06-29:
My 2c: it would be good to start tracking this indeed. Github "projects" feature is good for this, i think, if we create a board with lanes such as "looking for contributor", "assigned", "in progress", "done", etc. If you are willing to take part, I can add you as a collaborator to the project.

ESP8266WebServer has been copied to a new repository: https://github.com/esp8266/ESPWebServer. We need to update the ESP8266 core project to include that as a submodule. Also need to set up CI there to build against ESP8266 and ESP32 cores. Once this is done, some changes can be implemented (such as renaming of header files and the library, ESP32-specific adaptation and so on).

Regarding threading: the current idea is to implement pthreads port on top of FreeRTOS, and then enable gthread support in libstdc++. That will allow C++11 functions to be used. Relevant issue is https://github.com/espressif/esp-idf/issues/690.

bbx10, 2017-06-29:
Here is my attempt at making ESP8266WebServer work on ESP8266 and ESP32.

https://github.com/bbx10/webserver_tng

Similarly, for DNSServer and Hash libraries.

https://github.com/bbx10/DNSServer_tng
https://github.com/bbx10/Hash_tng

With these 3 libraries, I forked the following external libraries and got them working on ESP32. Be sure to use the esp32 branches because the master branches is just clones.

https://github.com/bbx10/arduinoWebSockets/tree/esp32
https://github.com/bbx10/WiFiManager/tree/esp32

There are some TODOs, some examples that do not work, and some code I do not really understand so this code is not ready for release. On the other hand, I ported many of my personal ESP8266 projects to ESP32 with these libraries.

lonerzzz, 2017-06-29:
I am happy to contribute. Please add me to the project and I will start a tracking page for different pieces. Once we have a shared view of who is doing what, I will see where I can contribute code.

lonerzzz, 2017-06-29:
@igrr One follow up question that arises - are you intending to keep the shared functionality in the ESP8266 or is this a temporary location?

igrr, 2017-06-29:
Basically we have three options where to keep shared functionality:
- esp8266 organization (owned by @themindfactory)
- espressif group
- as a personal repository (mine or @me-no-dev's)

Hosting these libraries in espressif org would send the wrong message, in my opinion, implying that these libraries are something that Espressif supports as a product. ESP8266 organisation on the other hand is not affiliated with Espressif and is an known place for ESP8266-related projects. Hosting on a personal page would not be as flexible (bus factor). Overall i think that hosting on esp8266 org is a good compromise but i am open to suggestions.

lonerzzz, 2017-06-29:
I am fine with that too, just wanted to capture the intent.

me-no-dev, 2017-06-29:
Personal or Espressif account will send the wrong message, I agree. If @themindfactory has no issues, esp8266 org is the place to go :)

copercini, 2017-06-29:
Here are my 2c for a better repository:

-	BLE/Bluetooth (At least say that there is a discussion here: https://github.com/espressif/arduino-esp32/issues/423 on index or projects)
-	WiFiSecureClient DN verification implementation
-	Tone/noTone functions
-	Move installation instructions to another pages keeping index clean
-	List of others working IDEs (VSCode, Sloeber and others), maybe make a quick guide
-	Insert a minimal structure example when open an issue, like ESP8266, ie: “your board model, some code, decode your logs with ExceptionDecoder and etc”
-	Maybe apply workarounds for deep sleep reset on Arduino Core by default 
-	How to start documentation for all libraries, like https://github.com/espressif/arduino-esp32/tree/master/libraries/SD
-	Installation via boards manager (can use the same ESP8266 package json file, so the people that already uses ESP8266 don’t need to include another link, just install ESP32 on boards manager)

lonerzzz, 2017-06-30:
I have tried to capture an initial set of outstanding tasks under Projects based on the feedback so far. Update it as needed or send me any items that you think should be added and I will add them in a standard manner there. If you are working on something, please tag the note with your id.

lonerzzz, 2017-06-30:
@copercini - can you please expand on the bullets: WiFiSecureClient DN verification implementation and Tone/noTone functions

lonerzzz, 2017-07-01:
There is a now a project capturing what has been outlined so far (https://github.com/espressif/arduino-esp32/projects/1)

bbx10, 2017-07-04:
@lonerzzz I am working on "continuous integration of ESPWebServer"

I created a .travis.yml that builds the WebServer library for esp8266 (stable and devel) and esp32 (devel).

https://github.com/bbx10/WebServer_tng/blob/master/.travis.yml

I never used Travis so seeing 3 cases build in parallel is a pleasant surprise.

lonerzzz, 2017-07-06:
@me-no-dev Are you able to outline what you are working on so we can track under: https://github.com/espressif/arduino-esp32/projects/1

You seem to have many items in progress.

lonerzzz, 2017-07-06:
@copercini thanks for the reference to SPIFFS. I have captured it in the project.

tedder, 2017-07-28:
So, is there a roadmap for "publishing" this, so the general Arduino community can use it without external tooling?

lonerzzz, 2017-07-28:
@tedder The roadmap is my attempt to capture the known open items and ones mentioned in the bugs. I have received limited responses from active developers regarding making this more formal. So as of now it serves as informal tracking.

copercini, 2017-08-01:
@lonerzzz Issue template was merged!
I think you have the first "Development Complete" item =)

lonerzzz, 2017-08-01:
@copercini Cool, thanks for the heads up.

pbecchi, 2017-08-02:
I am interested to WiFiManager and Ping libraries!
There is anybody working on this? 
Where I can find actual status info?
If you like, you can add EEprom library to the done list!

lonerzzz, 2017-08-02:
@pbecchi My exact intent in putting together the Road Map was so people could see who was working on what. I have assigned you to the WiFiManager and Ping libraries for now. If others come forward, then we can resolve otherwise. If someone has a better approach, I am happy to makes changes to help align the efforts from different people.

pbecchi, 2017-08-02:
Thanks for the assignement.....but I havent time...may be I will start looking to Ping!

lonerzzz, 2017-08-02:
@pbecchi Fair enough, I will leave you only on Ping for now unless you change your mind.

copercini, 2017-09-17:
Updated!

closed 2019-08-15 by stale[bot]