espressif/arduino-esp32 #474: OTA and analogRead

szerwi, 2017-06-29:
Hi,

When I have OTA enabled in my code, the analog reads are always 4095. I use pins 14, 4 and 2 and all of them aren't working.

me-no-dev, 2017-06-29:
I do not see why OTA would mess with anything. It's only network and flash access.

szerwi, 2017-06-29:
I don't understand it to, but it's not working.

Raienryu97, 2017-06-30:
Hey @szerwi  since you said you have OTA code I'm  assuming you obviously have WiFi on. 
With reference to [igrr's comment](https://github.com/espressif/arduino-esp32/issues/102#issuecomment-295272351)
Pins 2,4 and 14 all belong to ADC2, they will al give the highest value 4095 (been there, suffered that  :) )
You need to use ADC1 channel when you have WiFi on. 
Check the pinout and use one among ADC0-ADC7 
It helped me out.

me-no-dev, 2017-06-30:
yup... I was confused at thinking it was the other ADC that would go off

closed 2017-06-30 by me-no-dev

szerwi, 2017-06-30:
So all ADC10+ pins belong to ADC2? Thare are only 6 pins that belong to ADC1?

Raienryu97, 2017-06-30:
I think it is a design limitation that ADC2 will not work, I do not know the specifics of why, but ADC0-ADC7 is 8 Analog pins, is that not enough?
If not, then I guess buying external ADCs and connecting to non ADC1 GPIO could help you out.

szerwi, 2017-06-30:
@Raienryu97 but there's no ADC1_1 and ADC1_2, or am I blind?

https://www.google.pl/search?q=ESP32+pins&source=lnms&tbm=isch&sa=X&ved=0ahUKEwiqgMqHluXUAhUBVRQKHX2QCZgQ_AUICigB&biw=1920&bih=950#imgrc=rfUUzNMMCSZiEM:

Raienryu97, 2017-06-30:
[image]
Use one among ADC0 - ADC7 here.

szerwi, 2017-06-30:
I realised that two sensors can be connected to digital pins, so I think 5 analog pins are enough. Will pins that are ADC2 work as digital pins properly when WiFi enabled?

Raienryu97, 2017-06-30:
As digital pins, they should work.

szerwi, 2017-06-30:
Ok. Thanks for help. It's stupid that while WiFi enabled, this analogs don't work... :D

szerwi, 2017-06-30:
I've connected sensor to ADC1 pin and I'm calculating something. With OTA disabled the result is -0.9, but with OTA enabled the result is -1.2. Any ideas? :D

szerwi, 2017-06-30:
Sometimes when the ESP is turning on, and when OTA is enabling I have an error:

Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400fbf44  PS      : 0x00060b30  A0      : 0x8010318f  A1      : 0x3ffd0ab0  
A2      : 0x00000000  A3      : 0x3ffd6d94  A4      : 0x3ffd0b10  A5      : 0x0000000d  
A6      : 0x0000001f  A7      : 0x3ffd6a74  A8      : 0xf0029056  A9      : 0x3000012f  
A10     : 0x0000000d  A11     : 0x00000001  A12     : 0x00000002  A13     : 0x00000025  
A14     : 0x00000006  A15     : 0x00000005  SAR     : 0x00000013  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400fbf44:0x3ffd0ab0 0x4010318f:0x3ffd0b00 0x4010395c:0x3ffd0bd0 0x40093a04:0x3ffd0bf0 0x400952de:0x3ffd0c40

CPU halted.

But it's only from time to time, not always...

Raienryu97, 2017-06-30:
Use the esp exception decoder and paste your backtrace there, that should give you an idea :)

szerwi, 2017-06-30:
There are 3 different errors that are shown randomly:

1.
0x4010bd87: rc_get_trc at ?? line ?
0x4010bd87: rc_get_trc at ?? line ?
0x40092fc5: ppRxProtoProc at ?? line ?
0x400935ca: ppRxPkt at ?? line ?
0x400952de: ppTask at ?? line ?

2.
0x4008c5ce: write_txrate_power_offset at ?? line ?
0x40089589: set_tx_gain_table at ?? line ?
0x4008a511: bb_init at ?? line ?
0x4008acd3: register_chipv7_phy at ?? line ?
0x40118610: esp_phy_rf_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./phy_init.c line 150
0x4011875f: esp_phy_load_cal_and_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./phy_init.c line 294
0x400fd997: wifi_hw_start at ?? line ?
0x400fd9dd: wifi_start_process at ?? line ?
0x400fdf7a: ieee80211_ioctl_process at ?? line ?
0x4009528e: ppTask at ?? line ?

3.
0x400fbf44: ieee80211_parse_beacon at ?? line ?
0x4010318f: sta_input at ?? line ?
0x4010395c: sta_rx_cb at ?? line ?
0x40093a04: ppRxPkt at ?? line ?
0x400952de: ppTask at ?? line ?

me-no-dev, 2017-07-02:
seems to be bad power?

szerwi, 2017-07-02:
@me-no-dev my power module gives 500mA or 1A, so it is enough.

me-no-dev, 2017-07-02:
it really depends on many many things. Cable, actual output power, voltage drop at current spikes. Random errors like those usually mean that