espressif/arduino-esp32 #475: ESP32 Light Sleep Mode

architmuchhal12, 2017-06-29:
Hey all,

I want to explore the sleep modes on ESP32 using Arduino IDE. By referring the ESP8266 guide, I am able to put the ESP32 in `deepSleep` mode.

I want to use my ESP32 in either `lightSleep` or `modemSleep`, both of which are not defined in `#include "esp_deep_sleep.h"`.

Is there a way to use the other sleep modes or they are yet to be explored?
I have implemented the sleep modes in the basic Blink program for which the code is attached.

Thank you!

```
#include "esp_deep_sleep.h"

int ledPin = 5;

// Time to sleep (in seconds):
const int sleepTimes = 15;

void setup()
{
  pinMode(ledPin, OUTPUT);
  Serial.begin(115200);
}

void loop()
{
  Serial.println("Hello, world!");
  digitalWrite(ledPin, HIGH);

  delay(1500);
  digitalWrite(ledPin, LOW);

  delay(1000);
  Serial.println("Please wait...");
  
  Serial.println("Entering deep sleep mode...");
  // deepSleep time is defined in microseconds. Multiply
  ESP.deepSleep(sleepTimes * 1000000);
}

```

copercini, 2017-06-29:
Quote from ESP_igrr https://www.esp32.com/viewtopic.php?f=19&t=716#p3110
> Light sleep is essentially suspending the CPU, this is supported in the ESP-IDF (and happens automatically), but due to Arduino "loop" which runs constantly, this doesn't work in Arduino. This will probably also be supported somehow, but i'm not sure what the API should be. Modem sleep is already supported and enabled by default, unless SoftAP mode is used. There is no API to disable it, but unlike the ESP8266, enabling modem sleep doesn't mess with LwIP timers, so there should be no problem with keeping it enabled.

architmuchhal12, 2017-06-29:
Thanks, that was helpful!
So, I assume I can't use lightSleep in Arduino IDE for now.
Is is possible to use Watchdog Timers in ESP32 to go in sleep mode? Is there any example available for reference?

Thanks!

AndreKR, 2017-07-14:
Yes.
```
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
  Serial.println("This will be printed every 5 seconds");
  esp_deep_sleep_enable_timer_wakeup(5000000);
  esp_deep_sleep_start();
}

void loop() {
  // This will never be reached
}
```

lonerzzz, 2017-08-25:
@architmuchhal12 Is this question answered for you? If yes,we will retain it for reference.

AndreKR, 2017-08-26:
Even if he's happy with deep sleep, you might want to keep this issue around because light sleep doesn't work yet?
FWIW, I don't buy the argument that it doesn't work because of Arduino's loop, because on ESP8266 [it works](https://blog.creations.de/?p=149#auto_light_sleep) as long as you're executing `delay(t)` with `t` > 0.

copercini, 2017-10-18:
There it is:
``` cpp
#include <WiFi.h>

void setup() {
Serial.begin(115200);
}

void loop() {
Serial.println("light_sleep_enter");
esp_sleep_enable_timer_wakeup(10000000); //10 seconds
int ret = esp_light_sleep_start();
Serial.printf("light_sleep: %d\n", ret);
}
```

GeeF, 2017-11-28:
At least the current consumption goes down for me when using delay. Seems to be ok

copercini, 2018-04-21:
Here is some documentation and tests about it: https://github.com/espressif/esp-iot-solution/blob/master/documents/DFS_and_light_sleep/light_sleep_performance_en.md

Will close due it was implemented

closed 2018-04-21 by copercini

wilsonlmh, 2018-08-18:
In Arduino IDE there's no CPU frequency to set(only "Flash frequency"). By default the frequency is 240MHz. You can however change via:

    #include <soc/rtc.h>
    void setup() {
        rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
    }

The power consumption decrease half or more instantly! And to test if the `rtc_clk_cpu_freq_set()` works, you can do this:

    int usCount = 0;
    int lastUs = 0;
    void loop() {
         if (micros() - lastUs > 1000000) {
          Serial.println(usCount);
          usCount = 0;
          lastUs = micros();
        }
        usCount++;

    }

The `usCount` will be different between utilize or not utilize `rtc_clk_cpu_freq_set()`. You can do `Serial.print(usCount)` but will affect precision however.

Vladislavo, 2018-10-05:
In [this section](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html#entering-light-sleep) of ESP32 documentation you can find a way to put it into lite sleep mode.

Vladislavo, 2018-10-05:
In [this section](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html#entering-light-sleep) of ESP32 documentation you can find a way to put it into the light sleep mode.

ligantx, 2021-02-23:
Light sleep doesn't work for me either..
```
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

void setup(){
  Serial.begin(115200);
  delay(1000);
  print_wakeup_reason();
  //my code
  //https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html
  gpio_wakeup_enable(MY_NOT_RTC_GPIO_PIN,GPIO_INTR_LOW_LEVEL);
  //https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html
  esp_sleep_enable_gpio_wakeup(); 
  //Go to sleep now
  Serial.println("Going to sleep now");
  esp_light_sleep_start();
  Serial.println("This will never be printed");
}
void loop(){}
```
It prints  only: 
```
Wa
```
like it has stopped before even running the whole print_wakeup_reason() function

Then....when i connect MY_NOT_RTC_GPIO_PIN to GND (it is already pulled up with external resistor) I what the rest of the message:
```
keup was not caused by deep sleep: 0
Going to sleep now
This will never be printed
```

Very strange

actronc3, 2021-03-22:
> 
> It prints only:
> 
> ```
> Wa
> ```

put a 
delay(100); 
before entering

lbu01, 2022-03-27:
you should use flush() to force print act immediatly .
This code works perfectly for me in Arduino IDE:

#include <WiFi.h>

void setup() {
    Serial.begin(115200);
    Serial.println("light_sleep_mode");
    Serial.flush(); // send immediatly
    esp_sleep_enable_timer_wakeup(1000000); //1 second
}

int unsavedCount=0;
void loop(){
  //This is not going to be called
    unsavedCount++;
    Serial.println((String)"unsavedCount = "+unsavedCount);
    Serial.println("go to sleep");
    Serial.flush(); // send immediatly
    esp_light_sleep_start();  // enter in light sleep mode for 1 second
}

it prints : 
unsavedCount = 1
unsavedCount = 2
unsavedCount = 3

etc.

pipi61, 2022-03-28:
ligantx: try
[https://randomnerdtutorials.com/esp32-deep-sleep-arduino-ide-wake-up-sources/](url)