espressif/arduino-esp32 #477: Second (multiple) TLS HTTP requests on ESP32 result in error (randomly)

electronicsguy, 2017-07-01:
Hello all,
      I am experimenting with TLS on esp8266 and esp32. Trying out the basic _httpclientsecure_ example on both. On esp8266, it works perfectly. Whereas, it randomly gives errors on ESP32 after a few requests. The behaviour is easily reproducible.

   I've slightly modified the example to remove redundant things and make a series of 10 requests. The sketch I'm using can be found here: [test-wificlientsecure](https://pastebin.com/EA21pyvy).

(In order to run in it, uncomment the line at the very beginning only for esp8266 after selecting the correct board. Note: the certificate check has been removed to make the same code run on both boards).

The correct, expected output as seen on esp8266 is here: [output-esp8266](https://pastebin.com/0keG3bnx).

The erroneous output on ESP32 is here: [output-ESP32](https://pastebin.com/tJ71wSHr).

I enabled 'debug' level core-debugging within Arduino IDE and this is corresponding output on esp32: [output-esp32-debug](https://pastebin.com/r0H08wB5)

I think it may be related to this issue reported before: [https://github.com/esp8266/Arduino/issues/3330](https://github.com/esp8266/Arduino/issues/3330). however, the code changes between esp8266 and esp32 doesn't make it possible to apply the patch (at least I couldn't see how).

Has anyone else faced this problem before? Any ideas on what the underlying issue is? @igrr and @me-no-dev 

Testing environment:
software: Arduino IDE 1.8.3 running on Windows 7 64-bit
hardware: esp-01 (esp8266) and ESP32-S mounted on a ESP32-T dev. board

copercini, 2017-07-01:
Interesting...  just for make sure, you are running the last git version os ESP32? after this commit https://github.com/espressif/arduino-esp32/commit/cfde5579161e337c0f9d264a52468698931a81b1 ?

electronicsguy, 2017-07-01:
@copercini   ok, no I was using the stable branch. But I updated ssl_client.cpp with the commit you referenced above, but it still giving the same errors. 

New error log (debug level) located here: [output-esp32-debug-2](https://pastebin.com/Vf0naRqn)

 handle_error() seems to get memory allocation errors.

copercini, 2017-07-01:
What stable branch?

I'm 99% sure it is because of this commit or you are using an outdated version of library...

Try these files: https://github.com/espressif/arduino-esp32/tree/87093368d9b42d8a211d95ff54b8d3d2e26120d2/libraries/WiFiClientSecure

electronicsguy, 2017-07-02:
@copercini   stable branch is the one at: https://github.com/espressif/arduino-esp32

To confirm, yes I am using the commit now you mentioned above. It still giving the errors.

copercini, 2017-07-02:
Looks like you are running the example code from library.

It is EXACTLY this? https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/examples/WiFiClientSecure/WiFiClientSecure.ino

electronicsguy, 2017-07-02:
@copercini   No. As I mentioned in my OP, the cert key check has been commented out. The sketch link is available in the OP. Other than the key check, yes it is the same. I don't see why initial requests would go thru correctly and later ones fail, if there's any problem with the calling sketch. 

There definitely is some problem with the TLS library. I'm looking into it now.

electronicsguy, 2017-07-02:
@copercini  you can put the cert check back in (to make it exactly the same) and it'll give the same error.

copercini, 2017-07-02:
Oh sorry, I had not seen your code in the first post

So, try uncomment the client.stop(); from line 105

electronicsguy, 2017-07-02:
@copercini  So adding to two client.stop() calls as you mentioned above seems to make a difference. It's not erroring out for a count of 10 anymore. I'll do some more tests with this and another server and report.

copercini, 2017-07-02:
Nice! =)

ESP8266 WiFiClientSecure is based on Axtls and ESP32 on Mbedtls, so the whole backend implementation is different.

client.stop() is critical on ESP32 because of asynchronous implementation, so if you don't stop it, the connection will be open forever causing memory leak every new open client, until exhausted.  

Any idea to make it better is welcomed, maybe re-using the same context like IDF example does (https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_request/main/https_request_example_main.c#L309 ) or a hybrid sync/async implementation...

electronicsguy, 2017-07-03:
@copercini   " every new created client," - I didn't get this part. Even if stop() isn't called, isn't the client object still the same? Why would a new client object be created to make another request? Since the client object is defined in the sketch, it would still refer to the same object. Are you saying some dynamically allocated memory within the wificlientsecure implementation is duplicated, leading to memory leaks?

tobozo, 2018-05-19:
Currently testing [this example](https://github.com/espressif/arduino-esp32/blob/e3a5ae439bb94ae13bd970d9484d4665e1df4972/libraries/HTTPClient/examples/BasicHttpClient/BasicHttpClient.ino) where I'm using the second argument in `http.begin(URL, CA_CERT)` to enable TLS.

I've modified the sketch so it downloads **a hundred files in a row** and saves them to a MicroSD Card, no memory leaks observed, no reset, and no noticeable speed difference.

I've set `http.setReuse(true);` in the setup and I call `http.end()` on download complete.

MartinDavidWaller, 2018-11-14:
Hi, I was wondering if you got to the bottom of this as there still seems to be issues with multiple gets. Something as simple as:

<head>
    <title>ESP32 Failure</title>
    <link rel="stylesheet" type="text/css" href="styles.css" media="all" />
    <script type="text/javascript" src="jquery.js"></script>
</head>

Can cause failure with jquery.js simple not being processed?

Many thanks

Martin

waltsteinchen, 2018-12-20:
I tried all samples and suggestion above the last 3 days .... none of them seem to work (latest version from /espressif/esp32 as of dec 20 2018).

In all Instances I see the same behavior that running in  a loop after ~20 successfull loops, I finally get a 'connection refused' on further attempts.  Heap goes a little down and up  but even at the failure I see ~130KByte of freeheap .. 

Has anybody a working sample to share with me which run's in a loop to a https server to fetch a few bytes each time ?? certificate stuff is not important, no server validation needed.

Many Thanks!

waltsteinchen, 2018-12-21:
My fault, duplicate IP address!  (oberlapping dhcp and static range) 
Sorry for inconvenience

closed 2019-08-15 by stale[bot]