espressif/arduino-esp32 #479: client.connect hangs ESP32

dmody, 2017-07-02:
I have implemented a datalogger that connects to google sheets.  Almost all the time it works perfectly.  Intermittently, as far as I can tell, the ESP32 is hanging at the client.connect
`        display.print(".");
        display.display();
        client.connect(server,443);
        display.print("+");
        display.display();`

I can see that every once in awhile, the dot will display and no plus sign will appear. 

I have tried the following:

- Changed the wall wart 
- added a line "client.setTimeout(2000);"  
- checked to make sure the wifi.status was connected prior to making the call.
- put a delay of 2000 millisec between making the wifi connection and calling the client.connect
- I thought perhaps I had a memory leak and created a counter to call ESP.reset(); once every 100 client.connect() calls

Non of the above seems to have fixed the problem.    I believe I can tell from another board (I built) and is running off battery, and I think is having a similar problem (I don't have a display to check the issue on the second board is identical); since the battery on this board is draining, the ESP is stuck in a loop somewhere in the library code.  

I can't seem to trace my way through the library code to see where an infinite loop would be possible, so I can't seem to locate the source of the problem.  

I'd appreciate thoughts on solutions, debugging, etc.

How exactly can I tell if my libraries are up to date?

copercini, 2017-07-02:
Please enable Core debug levels as debug . It hangs on handshake or due memory leak?

dmody, 2017-07-02:
@copercini Thanks for responding so quickly.  

I haven't had the ESP32 connected to a computer when the error was occurring, so I haven't been able to to capture debugging info on the serial or on the arduino IDE.  Sometimes it's taking a day or two for the error to occur, sometimes it's a couple of hours.

dmody, 2017-07-02:
@copercini 
Is there an easy way to capture the debugging information without a connected computer?

copercini, 2017-07-02:
Your issue is probably this one: https://github.com/espressif/esp-idf/issues/630
To solve it is necessary Disable "Enable hardware MPI (bignum) acceleration" on ESP menuconfig if you are using Arduino as IDF component.

dmody, 2017-07-02:
I'm not using the Arduino as IDF component.  I started down that path but got confused by the instructions.  I could use a video tutorial.  

So there's no way around this in the Arduino IDE?

me-no-dev, 2017-07-02:
hang on for a day and it will be working :)

dmody, 2017-07-02:
@me-no-dev ... Very Awesome.  Thank you.

In the mean time, I'll try to get my head wrapped around using the IDF, as I'd really like to have some watchdog timers too.

dmody, 2017-07-27:
Quick Question... is this problem solved now in the arduino ide?

jmwright, 2017-07-28:
@me-no-dev @dmody I'm wondering about this update as well. I'm having similar hangs with the latest master.

me-no-dev, 2017-08-01:
yup all fixed :)

closed 2017-08-01 by me-no-dev

dmody, 2017-08-01:
@me-no-dev  Thank you for working this!!! Much appreciated!
* I have one project testing to see if there are further problems.
* I have a second project won't run.  The following exception is being caused.  Any hints on what could be wrong please?  

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:708
load:0x40078000,len:0
load:0x40078000,len:12168
entry 0x40078a68
2349.00 2879.00 2879.00 2919.00
_____end init____________ 
Attempting to connect to SSID: Guest
.[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 2 - STA_START
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40121e24  PS      : 0x00060230  A0      : 0x801226ed  A1      : 0x3ffd1130  
A2      : 0x3ffcd174  A3      : 0x00000000  A4      : 0x3ffd1150  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffd1110  
A10     : 0x00fb6c2e  A11     : 0x0afcd2a0  A12     : 0x005a5587  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0afcd2a8  LBEG    : 0x4000164d  LEND    : 0x40001667  LCOUNT  : 0xfffffffd  

Backtrace: 0x40121e24:0x3ffd1130 0x401226ea:0x3ffd1150 0x40121a25:0x3ffd11a0 0x40121a5e:0x3ffd11d0 0x40121811:0x3ffd1230 0x40119002:0x3ffd1260 

0x40117d3b:0x3ffd1290 0x40120c09:0x3ffd12c0 0x40118bcf:0x3ffd1320 0x4013193b:0x3ffd1340 0x4013023a:0x3ffd1360 0x40130fde:0x3ffd1380 0x40128382:0x3ffd13b0 

0x4012a35a:0x3ffd13e0

CPU halted.

dmody, 2017-08-03:
@me-no-dev 
For my own education, I would love to know what was wrong/changed in the libraries to fix the hanging problem.  Is there a way to trace back through the code changes?

illxi, 2018-05-20:
I am having the same issue here with the lastest arduino-esp32. It hangs on the code below
`digitalWrite(LED_BUILTIN, HIGH);
    //client.setTimeout(2000);
    if (!client.connect(host, httpPort)) {
      Serial.println("connection failed");
      Serial.println("wait 5 sec...");
      delay(5000);
      digitalWrite(LED_BUILTIN, LOW);
      return;
    }
    digitalWrite(LED_BUILTIN, LOW);`
Error codes were:
`[E][WiFiClient.cpp:121] setSocketOption(): 1006 : 9
[D][WiFiGeneric.cpp:300] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:305] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[D][WiFiGeneric.cpp:300] _eventCallback(): Event: 8 - STA_LOST_IP
`

chris-hanna, 2018-06-01:
@me-no-dev  I am not sure how to reopen this issue but I am getting the same result as @illxi 

[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:309] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 8 - STA_LOST_IP

After commenting out "#CONFIG_MBEDTLS_HARDWARE_MPI=" in sdkconfig I get the following:

[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:309] _eventCallback(): Reason: 200 - BEACON_TIMEOUT

My program basically sets up a bluetooth server, connects to wifi, and then sends a message to ifttt. If I comment out the bluetooth setup then everything works fine. I am happy to provide more information if needed!