espressif/arduino-esp32 #481: mbedtls_ssl_handshake: Hang Up

mgo-tec, 2017-07-04:
I will post it for the first time.
I'm sorry, my English is not good.

I am building an Arduino sketch that sends a GET request to the SSL web page every 30 seconds.
WiFiClientSecure.h is included, but it hangs about 10 to 100 times at the client.connect function.

It stops at line 153 of ssl_client.cpp and freezes.

Also debug messages

`[I] [ssl_client.cpp: 153] start_ssl_client (): Performing the SSL / TLS handshake ...`
Freeze here.

It seems to be hanging up with 
`mbedtls_ssl_handshake (& ssl_client -> ssl_ctx)`.

It will not time out.
With reference to [esp-idf issue# 630](https://github.com/espressif/esp-idf/issues/630), I commented out the following parts of sdkconfig.h.

Line 48: `// # define CONFIG_MBEDTLS_MPI_USE_INTERRUPT  1`
Line 126:` // # define CONFIG_MBEDTLS_HARDWARE_MPI  1`

However, it hangs up.

I also commented out the following lines on a trial.

Line 81:` // # define CONFIG_MBEDTLS_HARDWARE_AES 1`
Line 106: `// # define CONFIG_MBEDTLS_HARDWARE_SHA 1`

Still no effect.

Even using the setTimeout function does not work at all.

By the way, all client.stop (); is executed after one connection.

Is there a way to not hang or reliably time out?

copercini, 2017-07-07:
Try just update the IDF to last git version, now these thing are disabled by default https://github.com/espressif/esp-idf/commit/c35aa1d9a24c0a58c2d950866a6ebea025af3d22

mgo-tec, 2017-07-10:
I understood the method of ESP-IDF.
Thank you very much.
However, I do not know how to turn off hardware acceleration of Arduino IDE's mbettls library.
How can I do it?

copercini, 2017-07-10:
Most of the Arduino Core is pre-compiled (to not take ages to run a new skecth), only [using as IDF component](https://github.com/espressif/arduino-esp32#using-as-esp-idf-component) or platform.io it will full compile again, so if you are using otherwise it depends of @me-no-dev to recompile with theses options disabled.

mgo-tec, 2017-07-12:
I was able to solve the problem on my own.
I compiled Arduino-ESP32 libmbedtls.a file with ESP-IDF (using Arduino component).
I disabled mbedtls hardware acceleration with menuconfig.
And I set the stack size of core main.cpp to 8192.
Then, it started to work smoothly.
I think that it is better to modify the Arduino-ESP32 library.

Thank you for your answer so far.

mgo-tec, 2017-08-01:
The libmbedtls.a file was finally fixed.
Now it works only by changing the stack size of _main.cpp_ file.
Thank you very very very much!!

closed 2017-08-01 by mgo-tec