espressif/arduino-esp32 #485: Failed to connect to ESP32: Invalid head of packet ('w')

chinmoyrick, 2017-07-06:
Hi 
after fallowing all step of esp 32 windows installation i got this type error !

`Build options changed, rebuilding all
Sketch uses 108230 bytes (8%) of program storage space. Maximum is 1310720 bytes.
Global variables use 9552 bytes (3%) of dynamic memory, leaving 285360 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.0-beta3
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('w')
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('w')

`
plz help

bestpika, 2017-07-06:
May be you can use esptool to erase flash.

chinmoyrick, 2017-07-06:
sir i don't understand . i have little information about esp 32 . its only esp 32 chip , i connect usb to ttl connector . and fallow all step of esp32 arduino windows installation . and i open serial monitor 115200 bd and its print board information . 

i still unable upload any code ..
plz help me how to flash . i am using windows computer.

copercini, 2017-07-06:
@chinmoyrick After it appers "Connecting.........." try press boot the button of your board for few seconds or connect the pin 0 to the ground.

chinmoyrick, 2017-07-07:
@copercini  it's nor working now it show new error
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

copercini, 2017-08-05:
Got same issue with a new board and a cheap USB cable, changing the cable, everything works fine

lonerzzz, 2017-08-15:
@chinmoyrick Is there any update to this issue? Did you work through it? Can we close it?

greanthai420, 2017-12-09:
on DOIT esp32 devkit1 i also have this issue, messing around with BOOT and EN buttons doesn't do anything.

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('x') <---- the x is random. changes everytime i try to upload

me-no-dev, 2017-12-11:
check and maybe resolder all pads from the module to the board. I have seen reports of bad reflow jobs

sharatkanthi, 2018-11-02:
Press the "Boot" button on your ESP, Start uploading your code from Arduino IDE and keep "Boot" pressed till upload completes.

roysinurat, 2019-01-29:
i press boot button but still error.. or boot+ eneble.

a fatal error occured: failed to connect to esp32: invalid head of packet (0x00)

please help me.. 

i tray upload  with atom with PlatformIO but message is same.. 

my board is new.
DOIT ESP32 DEVKIT V1 (ESPRESSIF ESP32 -WROOM32)

please

sschepis, 2019-02-05:
ya I have this problem too, @everslick maybe if you're going to close an issue maybe give a link to something that isn't a chat room (data permanence and all that...) . I will try erase_flash as per @bestpika 's suggestion

kebner, 2019-03-24:
Hi
I stumbled upon that same issue and found that at least some GPIO pins are causing this. For me, I had a one wire temp sensor on GPIO2 with a 4.7K pullup. No way to upload until I unpluged that one first.

beegee-tokyo, 2019-03-24:
[Reserved GPIO’S](https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/)

glassum, 2019-04-18:
Had this message in Ubuntu but not in Windows 
Solved by going to Tools -> Upload Speed and changing the speed from 921600 to 115200.

rpnogue, 2019-08-24:
I got the same error. Pressing RST button anytime didn't do anything.

My circuit had a LED connected to GPIO0 (pin 6 on my board). Changed GPIO and the problem was fixed.

Hope it helps

Gabriel38g, 2019-12-13:
Thanks for this. I had this same problem and this thread probably saved me hours of frustration! I checked GPIOs but it was changing upload speed that did the trick.

k5avl, 2019-12-23:
I have the same problem.  I have tried every suggestion plus I am now attempting to load a necked board. Incidentally, I even get the message without being connected to a board at all.  Here is my setup:

OS: Linux DebianDT 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11) x86_64 GNU/Linux

Board: DO IT ESP32 DEVKIT V1  or Firebeatle ESP32  (or any board for that matter)
Load Speed: 115200
Port: /dev/ttyUSB0

Program:
/*
   DuinoVOX. Ver.3.1.1  Arduino based PTT control for radio interfaces.
   Written Aug. 2016 by Kevin Loughin.  Originally for use
   in a clone of the Tigertronics Signalink design idea.
   Modified by Garland Sparks for ESP-WROOM-32, 12/6/2019.  Use the "Firebeatle ESP32" board manager
*/

//  Naming the pins I'm using for clarity.
const int audioInPin = 12;  // Audio sense
const int tailsetInPin = 13;  // delay time adjustment POT
const int sensitivity = 14; // threshold set POT
const int PTToutPin = 4; // output to keying transistor

//  declaring variables that we'll use
int delayvalue = 0; // amount of time in hundreths of a second before dropping PTT
int threshold = 0; // audio trigger level initial value
int ptt = 0;  // variable for holding current PTT delay countdown
int audio = 0;  // variable that will hold audio sense
int PTT_ON; // Flag to indicate current PTT status

//Serial.println("delay = " delayvalue, "threshhold = " threshold, "PTT = " PTT_ON);

void setup() {
  // Turn off PTT right away so we're not keying on startup

  pinMode (PTToutPin, OUTPUT);
  digitalWrite(PTToutPin, HIGH); //LOW);
  PTT_ON = 0;
}

void loop()
{
  // Here we go. First read the pots and set variables

  delayvalue = analogRead(tailsetInPin) / 5; // yields 0 to 204
  delayvalue = delayvalue + 2; // adjusted to no less than 2, 1 after first pass through loop
  threshold = analogRead(sensitivity) * .8; // yields 0 to 819
  threshold = threshold + 70; // adjusted 70 to 889

  // Check for audio. Set delay if present
  audio = analogRead(audioInPin);
  if ( audio > threshold ) // if audio in exceeds sesitivity threshold.
  {
    ptt = delayvalue; // set ptt to current delay in hundredths of a second
  }

  // PTT control check
  if ( ptt > 0 ) // if PTT is positive (loop is counting down)
  {
    if ( PTT_ON == 0 ) // check that we havn't already turned it on
    {
      digitalWrite(PTToutPin, LOW); // HIGH); // turn on PTT if it's off
      PTT_ON = 1; // so we don't waste time writing on next pass
    }

    ptt = ptt - 1; // count down in hundredths of a second
  }
  else // the counter reached zero.
  {
    if ( PTT_ON == 1 )  // check if we already turned it off
    {
      digitalWrite(PTToutPin, HIGH); // LOW); // turn off PTT
      PTT_ON = 0; // Remeber that we've turned it off
    }
  }
  delay(10);  // wait 10ms before looping again.  loop 100 times/second
}

I hope someone can help me.

ds18b20, 2020-04-12:
> May be you can use esptool to erase flash.
> 
> ```shell
> python esptool --port com1 erase_flash
> ```

This worked for me.
If you are using microPython like me, you can try this answer.

meinkea, 2020-05-01:
> Just had this error in Ubuntu 14.04. Windows 10 has no such problems.
> It is the fresh ESP32 board with no connected devices at all. All the pins are empty.
> Solved by going to Tools -> Upload Speed... and changing the speed from 921600 to 115200.
> upload speed 460800 works too.
> But 921600 fails.

I have the same problem as glassum.
I am using a fresh Wrover-B module on a TTGO Koala board.

Something as simple as,
```
void setup() {
  Serial.begin(2000000);
  Serial.println("\n\n ----- ----- ----- BEGIN ----- ----- -----");
}

void loop() {

}
```
does not correctly upload with an upload speed of 921600
and with the Tools->Board option set to ESP32 Dev Module,

the output comes out to
```
Build options changed, rebuilding all
Sketch uses 215285 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 15436 bytes (4%) of dynamic memory, leaving 312244 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting......
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 84:0d:8e:d2:8f:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header
```

Changing the selected Tools->Board option from ESP32 Dev Module to ESP32 Wrover Module runs into the same issue...

However, I noticed if you toggle these settings
Tools->Flash Mode (I tried QIO and DIO)
Tools->Flash Frequency (I tried 80Mhz and 40Mhz)

Sometimes you will get
```
Build options changed, rebuilding all
Sketch uses 259157 bytes (19%) of program storage space. Maximum is 1310720 bytes.
Global variables use 15428 bytes (4%) of dynamic memory, leaving 312252 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting......
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 84:0d:8e:d2:8f:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...

A fatal error occurred: Invalid head of packet (0xE0)
A fatal error occurred: Invalid head of packet (0xE0)
```
if you recompile, you get the original error.

and sometimes if you toggle both after getting the 2nd out put (with 0xE0) you _might_ get this...
```
Build options changed, rebuilding all
Sketch uses 259157 bytes (19%) of program storage space. Maximum is 1310720 bytes.
Global variables use 15428 bytes (4%) of dynamic memory, leaving 312252 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting......
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 84:0d:8e:d2:8f:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...

A fatal error occurred: Invalid head of packet (0xA6)
A fatal error occurred: Invalid head of packet (0xA6)
```

_Without changing the settings_...
If you re-upload with the error output 3 (with 0xA6), it gives the 2nd output (with 0xE0)
If you re-upload again with the previous error (with 0xE0) without changing the settings, the output goes back to the original Timed out error.

When the errors for the 2nd and third outputs are rather inconsistent... changing the settings in a certain way did not always guarantee which error you are going to produce.

Changing Flash Mode and Flash Frequency options will do it, but also switching the Tools->Board from ESP32 Wrover Module to ESP32 Dev Module (or vice versa) also yielded a timeout error or the Invalid packet (0xE0) error.

The only procedure that I found that will reliably produce the 2nd output (with 0xE0) is to first change the Flash Frequency -> upload yielding the first output -> change Tools->Board -> upload yielding the 0xE0 error output.

To rule out something along the lines of having a bad board, I tried a second board with a Wrover-B module and it yielded similar results.

Both options for Tools->Boards, ESP32 Wrover Module and ESP32 Dev Module, seem to give all three of the previously mentioned outputs.

Going to Tools -> Upload Speed... and changing the speed from 921600 to another option (115200, 230400 or 460800) fixes the issue for _Both_ Tools->Board options, on both of the physical board.  
The fix works for other configurations for Tools->Flash frequency, Tools->Flash mode and Tools->Board options.

I am guessing something is not being timed right when flashing at 921600.
*Really* no idea what it could be at the moment.  For now I am using 460800, but maybe later I might investigate my /dev/ttyUSB0 connection and see if there is anything to suggest a OS setup problem.  I doubt that's the issue though... my ESP32 Wroom module and other boards flash fine at speed of 921600.  

Not sure why the 921600 setting is breaks things, but it is certainly not playing nice.

patrickjane, 2021-04-10:
I encountered the same issue with a FireBeetle ESP32. 

I can confirm that setting the Upload Speed to 115200 fixed the issue for me.

OuttaPencil, 2021-06-19:
In addition to taking into account all of the above advice, I was having this issue because I was powering my esp32 using an external power source rather than through my USB-serial (FTDI) adapter. Connecting just the TX and RX pins is not enough.

Hopefully this helps at least one other person who is currently making the same mistake and scouring the internet for a solution :P

nuacz1, 2021-09-13:
I had the same problem and error showing up, however it was a simple mistake. I had my ESP32-C3 already flashed from a different program. I had to cancel the unwanted program (control+] on mac) and reflash the wanted program. That worked for me, let me know if that helped.

csesar123, 2022-01-11:
i got the same error,and the board is new.through many times ,my program upload in it successfully.the device need the external power and usb to ttl. press the Boot 、RST button on your esp and then start uploading your code from Arduino IDE and keep the button pressed until upload completes.try more times

aprogrotess, 2022-05-27:
I buy a 'esp32-cam' to develop AI project, i see the same error. My teacher tell me, you should connect GPIO0 to GND, that's work, may it can help you.