espressif/arduino-esp32 #487: How do I run at 160 or 240MHz?

hellodb, 2017-07-06:
How do I run at 160 or 240MHz?

Raienryu97, 2017-07-06:
The current libs used sets ESP32 at 240 MHz. You need to manually compile using esp-idf and choose 80/160/240 MHz (using settings in make menuconfig) and then copy over those libs. It's a bit tedious, but it can be done. 
I am not sure if there is an easier way.

tobozo, 2017-07-06:
I'm quite sure the 160/240Mhz choice was available in the Boards menu of the Arduino IDE sometime in the past, only it was replaced by 80/40Mhz at some stage.

Are those values real ?

everslick, 2017-07-07:
does anybody know, if setting CPU clock at runtime, is even possible on the ESP32 (like it is on ESP8266)?

igrr, 2017-07-07:
@everslick No, this isn't possible on the ESP32.

everslick, 2017-07-07:
thanx, then i can safely remove the setting from the code.

fesch, 2017-07-16:
After having tested my ESP32-EVB board from Olimex with the script on this page

https://hilo90mhz.com/arduino-esp32-esp8266-101-speed-test-comparison-chart/

I conclude that whatever you chose in the menu, the ESP32 runs at 240MHz. Maybe someone else could try too and confirm this ...

tobozo, 2017-07-17:
On ESp32-Wrover-Kit my sketches refuse to work if I select 80Mhz before uploading, it'll only work when 40Mhz is selected, so it does affect *something* but what would it be? The SPI clock maybe?

fesch, 2017-07-17:
Strange ... @me-no-dev told me in some other post that the core of ESP-IDF is compiled with "240MHz" settings and than the speed of the clock cannot yet be changed using the Arduino IDE.

igrr, 2017-07-17:
40MHz/80MHz is the SPI flash frequency setting. It is not related to CPU frequency setting.
CPU frequency is set at the time this library is compiled, and can not be changed afterwards.
80MHz flash setting is indeed not working on ESP-WROVER-KIT, due to some signal integrity issues introduced by flash lines broken out to pin header.

me-no-dev, 2017-07-17:
But I do run all of my WroverKits at 80MHz flash? never had an issue. What revision is the kit in question?

igrr, 2017-07-17:
This only applies to WROVER-KIT with a WROVER module mounted on it. The (older) ones with WROOM are not affected. There may be some boards which do work, but we have observed the issue on some boards. This was also reported on esp32.com by loboris. Removing resistors which connect module pins to flash header can be used to work around the issue.

me-no-dev, 2017-07-17:
Ahh ;) do not have one of those

ricdijk, 2017-09-10:
If i change the 'f_cpu' setting in boards.txt for my board (lolin32) from 240000000L to 160000000L or 80000000L the board seems to run at lower speeds. 
However I don't know the consequences of this, there must be a reason the speed is fixed to 240MHz in the IDE.

me-no-dev, 2017-09-10:
it does not run at lower speed actually, but will mess up some calculations. Libs are build for 240MHz and dynamic freq change is not yet available, but I know it's in the works. Stick to 240MHz until you see the option in the menu to change it :)

sansillusion, 2017-09-10:
@me-no-dev I just want to say : YOU ROCK !!!!
I have been new to arduino/esp8266/esp32 since this summer and I have been following esp32 development lately and every time I need to do or know something you end up working on it or having just finished what I need to use it. I love the fast development and finding out new things I can do with these WONDERFUL esp32 devices.
Thank you so much ! I could never have dreamed this to be possible at such low cost ! I can now afford my passion and have such a great community witch have made everything so easy for me.

I know that coding just the base connection management must be very complicated but to me the user it's just a matter of using a few library and some little code and I have a wifi connected device with web server and etc... just WOW !

You rock and everyone who made this possible !

closed 2017-12-20 by HelloDB

WoderfulIOT, 2018-11-20:
Whats the status of dynamic speed change as on today?

everslick, 2018-11-20:
i use the following snipped, it seems to do something. :-D

```
uint32_t system_cpu_frequency(void) {
#ifdef ESP32
  return (rtc_clk_cpu_freq_get() * 80);
#endif

#ifdef ESP8266
  return (ESP.getCpuFreqMHz());
#endif

  return (0);
}

bool system_cpu_speed(uint8_t step) {
  bool ret = false;

#ifdef ESP32
  if ((step >= 1) && (step <= 3)) {
    rtc_clk_cpu_freq_set_fast((rtc_cpu_freq_t)step);

    ret = true;
  }
#endif

#ifdef ESP8266
  if ((step >= 1) && (step <= 2)) {
    system_update_cpu_freq(step * 80);

    ret = true;
  }
#endif

  if (ret) {
    log_print(F("SYS:  switching to %i MHz CPU clock frequency"), step * 80);
  }

  return (ret);
}
```