espressif/arduino-esp32 #489: Use esp_intr_alloc to register interrupt handlers

igrr, 2017-07-07:
Putting this on the issue list so this doesn't get forgotten...
Currently ISRs are registered manually to hard-coded interrupt numbers.
Interrupt allocator in IDF should be used to register interrupt handlers instead, which will resolve issues with interrupts running while flash cache is disabled (like #488).

cranphin, 2017-08-18:
What is the workaround for this issue until this is fixed? I have no idea to make my interrupt work now :)

igrr, 2017-08-19:
Your interrupt will still cause an exception whenever it happens at time when flash is accessed. A workaround is to mark the ISR with IRAM_ATTR attribute to place it into RAM (instead of flash, where every function goes by default). And make sure all functions you call from ISR also have an IRAM_ATTR.

Testato, 2017-10-15:
I use preferences library to backup some data to flash and have two ISR on two GPIO.
Often this error appear:

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x400d18f0  PS      : 0x00060034  A0      : 0x40082328  A1      : 0x3ffc0bf0  
A2      : 0x00000010  A3      : 0x3ffc3608  A4      : 0x00000000  A5      : 0x00010000  
A6      : 0x00000000  A7      : 0x3ffd5440  A8      : 0x80080df0  A9      : 0x00000001  
A10     : 0x00060723  A11     : 0x00000004  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000003  A15     : 0x00060b23  SAR     : 0x00000015  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400d18f0:0x3ffc0bf0 0x40082325:0x3ffc0c10 0x4008f4eb:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14

Is it related to this issue ?
May point me to an arduino examples that use the IRAM_ATTR like your tips ?
Thanks

InfiniteYuan1, 2018-08-07:
@igrr what is the reason of cause the error?

atanisoft, 2019-08-01:
@me-no-dev any remaining ISRs not using esp_intr_alloc?

VojtechBartoska, 2022-03-16:
Discussed on Team meeting. Not relevant now, closing.

closed 2022-03-16 by VojtechBartoska