espressif/arduino-esp32 #490: compilation error on dhcpserver.h

pbecchi, 2017-07-07:
I have downloaded last esp32 core and I am compiling with IDE 1.8.1 and I am getting following compilation error:

tcpip_adapter.h:48: In file included from
esp_event.h:23: from
esp_event_loop.h:22: from
WiFiGeneric.h:28: from
WiFiSTA.h:28: from
WiFi.h:32: from
OpenSprinkler_ESP8266_V_2_1_6.ino:123: from
 
dhcpserver.h: 133:13: error: expected identifier before '=' token
   MESSAGE = 56
 
dhcpserver.h: 133:13: error: expected '}' before '=' token
 
dhcpserver.h: 133:13: error: expected unqualified-id before '=' token
 
dhcpserver.h: 154:1: error: expected declaration before '}' token
   } dhcp_msg_option
 
can anybody help me?
Thanks

igrr, 2017-07-07:
I guess your ino file has `#define MESSAGE  (..something..)`. Try using a different name.

pbecchi, 2017-07-07:
Thanks Ivan.
You are right ...I  have changed my define and now it compile ok!

closed 2017-07-07 by pbecchi