espressif/arduino-esp32 #491: Fatal error: waiting for packet content

DreamingRaven, 2017-07-07:
## The Setup
OS: Windows 8.1
Using: Arduino IDE
Sketch: Example "WiFiClient"
Board: "ESP32 Dev Module"
Flash Frequency: "80MHz"
Upload Speed: "115200"
Core Debug: "None"
Port: "COM5"
Board: ESP-WROOM-32
Board Status in IDE: [image]

ESP32 Module Soldered and Wired on:
[image]

USB to TTL Wiring:
Black-Ground
Red-3.3V
Yellow -TXD pin on adapter (USB-2-TTL), RXD pin on ESP32 
Blue -RXD pin on adapter, TXD pin on ESP32
[image]

## The Error
Sequence of Events: Compile>Upload Start>Wait for "Connecting">Put ESP32 into Bootloader Manually>Wait>Error
[image]
"A fatal error occured: timed out waiting for packet content"

## Attempted and Failed Solutions
In each attempted solution, settings were reverted back to those in "The Setup" section, and were not changed unless specified in numbered bullet point.
1. Set Upload Speed to 921600.
2. Set Flash Frequency to 40MHz.
3. Using a Different USB to TTL adapter.
4. Using a Different Sketch.

## Query 
I cannot find the specific documentation on this error message, so that I could properly debug the problem. Since it gets past the packet header stage, I assume everything prior is working. Should I be doing something differently? What could be causing this problem?

copercini, 2017-07-07:
You need to put 3.3v on EN (Enable) pin 😉

copercini, 2017-07-07:
It's your first upload?

DreamingRaven, 2017-07-07:
Yeah its my first time with ESP32 and flashing it like this, although in the past ive used ESP8266 Feather, but that was so streamlined for stuff like this. Now i have the horror of finding EN, since i soldered the labels between the breakout and vero board XD

copercini, 2017-07-07:
I checked your adapter board on google and it have an internal VCC to EN connection, so no worry about it

DreamingRaven, 2017-07-07:
Il try the EN pin anyway in a sec.

DreamingRaven, 2017-07-07:
My quick test with a breadboard did not work, with EN, but I need to double check i have the right pin since i cant see the labels.

DreamingRaven, 2017-07-07:
Just checked, and yeah it was the right pin. Im going to make sure this new connection is providing a voltage diff, before I continue.

DreamingRaven, 2017-07-07:
Also was looking through espressif/esptool and found https://github.com/espressif/esptool/issues/213 still looking through the issues over there for anything similar, havent found one that is after compression stage.

DreamingRaven, 2017-07-07:
I think the next best possible approach is to create the Binaries and use esptool directly, and hopefully this will be more verbose than the Arduino IDE.

lonerzzz, 2017-08-15:
@DreamingRaven Are there any updates to this issue? Is there still some information you need or can we close this?

DreamingRaven, 2017-08-16:
No luck as yet, still getting the same error as previously, and ive been slowly trying esptool.py directly where I can.

DreamingRaven, 2017-08-16:
[image]
 I've also been using an oscilloscope and multimeter to verify that the power is not the problem. Which it does not appear to be.

DreamingRaven, 2017-08-16:
I have been slowly attempting different things, I'm just praying it is not a manufacturing defect because that would be a pain to diagnose. Otherwise the chip seems to be communicating fine, stub uploads and runs, its just after compression, it times out "waiting for packet content". :/
[image]

DreamingRaven, 2017-08-16:
[image]
Suppressing compression using esptool leads to the same result above.

DreamingRaven, 2017-08-16:
Ok so now i'm just gonna go through the gotchas in https://nodemcu.readthedocs.io/en/master/en/flash/ and eliminate anything that fits. Can anyone see anything im doing wrong so far?

me-no-dev, 2017-08-17:
Make sure GPIO12 is pulled LOW. I have the feeling yours is not and its trying to boot the flash at 1.8V

DreamingRaven, 2017-08-17:
@me-no-dev What makes you say that? What is the significance of IO12? Looking through the data sheet, i see no significant mentions of IO12, it just seems to be a normal IO pin, but im a newcomer to this specific chip, I dont know its ins and outs yet. Il check in a second the Vdif across IO12 in a moment, its a pain to find the pins due to labels being between module and vero board :/

igrr, 2017-08-17:
If you are looking at the latest datasheet, that is the first row of table
2 in section 2.4, Strapping Pins.

DreamingRaven, 2017-08-17:
I must be looking at the wrong data sheet. (http://akizukidenshi.com/download/ds/espressifsystems/esp_wroom_32_datasheet_en.pdf)
There is no section 2.4 on this one:
[image]
Let me just find the latest one.

DreamingRaven, 2017-08-17:
Yeah just got August 10 2017 datasheet (https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf), 
I assume GPIO12 is MTDI

DreamingRaven, 2017-08-17:
Yeah looks like it is high, 
[image]
Excuse the imprecision of the oscilloscope, but multi meter confirms it.

DreamingRaven, 2017-08-17:
Let me just confirm that this is the correct pin.

DreamingRaven, 2017-08-17:
No i had the wrong one the first time, although it is the last pin, the trace actually takes it to the second to last pin, rather than the last pin which i had connected.
[image]

DreamingRaven, 2017-08-17:
I must admit however I am strugling to understand the pinout diagrams in the latest data sheet, as it looks nothing like my current module: 
[image]

Compared to the older data sheet:
[image]

The problem with the older data sheet is that it puts the TxD and RxD on the other side of the board to which this chinese breakout board has labeled as TxD and RxD.

DreamingRaven, 2017-08-17:
Also my apologies to everyone, i'm posting relentlessly so that in the future both I and others can see the mistakes I made, and my (erratic) thought process.

DreamingRaven, 2017-08-17:
Ok after trace following and relying soley on the modules labels, I have found that this breakout board was not mis labelled, but instead 3.3v and gnd are connected to pins on the opposite side.
Where pink is connected to 3.3v, black is grnd and blue, green are TxD and RxD respectiveley.
[image]

DreamingRaven, 2017-08-17:
@me-no-dev I have "strapped" IO12 to ground, but there was no change in error message:
[image]

DreamingRaven, 2017-08-17:
I don't know where to go now. What is causing this error, timeout; packet content? I'm really excited to make stuff with the ESP32 but its been weeks with minimal progress made trying to solve this issue.

me-no-dev, 2017-08-17:
I have had the same exact issue a couple of days back and it came to be the bootstrapping of pin12 (MTDI) and I mean like 100% the same exact problem. Can you confirm that GPIO12 is low at boot/into-download modes?

DreamingRaven, 2017-08-17:
Yeah il double check tomorrow, its getting pretty late here.

DreamingRaven, 2017-08-18:
Ok, to streamline things a bit, and because I wanted to, Instead of spamming photos, i'l use slightly abstracted diagrams. E.G
[image]

After testing IO12 Vdiff using:
[image]

Vdiff is 34mV.

DreamingRaven, 2017-08-18:
hmm something has happened, maybe a faulty jumper cable connection, but now the module keeps resetting.

DreamingRaven, 2017-08-27:
I have since enabled the auto-bootloader by connecting:
[image]

This still results in the same error "timed out waiting for packet content".

I further tested the voltage supplied to the module, on the modules side, and found it to be 3.45v, within the acceptable range. I cant find any reason why this module should not be working. As such I ordered a new one, to reduce the chances that this is a module side problem. If this does not work then I know that I have either gotten unlucky, or that it is a more systemic problem. These problems could be my soldering, unintended connections, windows, or possibly even esptool.py.

me-no-dev, 2017-08-27:
Above you show GPIO tied to GND and that is not the right configuration!!! Needs to be tied to VCC

me-no-dev, 2017-08-27:
use 1K resistor between IO12 and VCC

DreamingRaven, 2017-08-27:
To confirm what I am testing, you are saying: 
[image]

DreamingRaven, 2017-08-27:
First test, made no change, let me just check that everything is correctly implemented.

DreamingRaven, 2017-08-29:
Ok so I got a second ESP32 and soldered it (less fancy than last). I used the example ESP32 sketch GetChipID, with no alterations. Compiled and attempted upload. Low and behold, the exact same error message. 
[image]

[image]

[image]

DreamingRaven, 2017-08-29:
What could this mean, in terms of source of the problem. 

- Soldering Connection Issue; Less likely to occur in both boards soldered seperateley and differently.

- Chip Manufacturing Issue; Less likely due to the problem being highly improbable to occur twice, on two separate chips and batches.

- Wrong Connections; Same probability since, both chips are wired to the same (above) circuit diagram.

- Faulty UART bridge; Unlikely, since two separate and different boards were used, of different designs and manufacturers.

- Sketch Issue; very low probability, as many different example sketches have been tested so far, with many different configurations.

- Tool Issue; Highly likely, as bugs are frequent in most programs, and exhaustive testing likely has not occurred, as unreasonable/ impossible.

Have I missed anything?

DreamingRaven, 2017-08-29:
I think, it may be best to scrap using arduino and esptool.py all together, and use something else, as the two most likely causes are wrong connections or a tool issue. Since I heuristically assume that between these two a tool based bug, is the most probable (debian, windows 8.1, arduino, esptool.py).

me-no-dev, 2017-08-29:
There surely is nothing wrong with Arduino and esptool.py ;) Arduino is on top of IDF and IDF and esptool are the official tools to work on esp32. Something totally hardware is going on

DreamingRaven, 2017-08-29:
I'm just not sure what more I can try hardware wise. I have read through so much documentation, and followed so many guides, I don't know what other options I have left.

DreamingRaven, 2017-08-29:
:O 
IT WORKED!

DreamingRaven, 2017-08-29:
[image]
HOLY WHUT

DreamingRaven, 2017-08-29:
I just took out the resistor, and it works. ~_~
[image]

DreamingRaven, 2017-08-29:
let me just repeat with a different more advanced sketch, to make sure this is not just some weird fluke.

JustAmaze, 2017-08-29:
well done !  Did you boot up your module and see it run properly with wifi sketch?

DreamingRaven, 2017-08-29:
Sorry for the delay, I went away and made some creme brulees. I did but im getting "brownout detector was triggered" in serial console. Which to my understanding means, the resistance in the jumper cables/ breadboard, is too high (or the UART bridge doesn't supply enough).

DreamingRaven, 2017-08-29:
Maybe il power it from something a little more expandable, so I can put in some quality of life features, like power LED, and battery power etc etc. Kill two birds with one stone. Maybe I should flash it with an OTA sketch first, just in case. But il need to check that ESP32 can update OTA, I may be mis-remembering / watched something outdated that said it couldn't, just need to confirm that.

elucas2uy, 2017-09-28:
Hi DreamingRaven, I´m seeing the same behavior here... exactly the same one... 2x esp32 with the same problem, after "backs and forwards" the first one back to life (no much explanation how it happened), and after that no problem at all... the second one still not working.
There should be some kind of explanation.

What I cant see (and is some kind of strange to me) is that when the problem is present, Testing resistance between 3v3 and GND in too  low (much lower than the usual one in the working boards)... 
**After some hours disconnected this resistance start "going" to the normal one ! **..... and reconnecting it to power it works fine for some seconds, and after that the problem start again.

Also, when the problem is present and the ESP energized (strong 3,3v soldered and stable), the EN (that is poll UP with a 10K resistor) shows almost no voltage.

Apparently, at this state something is happening inside the module or the chip itself (I did not uncover the module to take measures at the chip pins)

I don´t care about the this ESP32 itself, but I definitely care about know what is the logic around this (or how to avoid the problem in the future).... 
So my questions:
1) Did you figure it out how did you solve it ?  ...as you said, You previously try the exactly same configuration that finally worked... wasn´t it?
2) Do you remember if you change something else when it start working (the second ESP 1st time, and the first ESP 2nd time). What was it?

AS you previously said 
Tks

elucas2uy, 2017-09-28:
#flash read err, 1000 #113

elucas2uy, 2017-09-28:
Additionally, @me-no-dev  wrote ...
first: "Make sure GPIO12 is pulled LOW. I have the feeling yours is not and its trying to boot the flash at 1.8V" and "Can you confirm that GPIO12 is low at boot/into-download modes?"

later: "Above you show GPIO tied to GND and that is not the right configuration!!! Needs to be tied to VCC" and "use 1K resistor between IO12 and VCC".

Just to clarify, the first one (ie GPIO12 pulled LOW) is the correct one, right?

Tks

elucas2uy, 2017-09-29:
Hi again, 
Following the @igrr comment on #https://github.com/espressif/esp-idf/issues/958

> If you have flash read err, 1000 error continuously after reset, please check that you don't have any external signals connected to SPI flash pins of the ESP32 (pins 6-11). If you don't have anything connected to these pins, it means that your SPI flash chip may have been damaged.

I have found some flux not cleaned in both boards and ESPs. After cleaning them correctly both start working fine!
Hope this could help somebody else with this same or similar issue..

Note: I do not understand why the module have this behavior but surely is something related to unexpected conductivity between some pins (I could not conclude which of them).

DreamingRaven, 2017-10-31:
Hey @elucas2uy , sorry for the late response, I think It may actually have been the breakout board I was using, because well.. I cant seem to replicate the failure with my other boards. And this is why the same configuration suddenly worked, but on a different board, from a different manufacturer, glad it worked for you in the end. Also happy Halloween!

DreamingRaven, 2017-10-31:
@elucas2uy, If you need me or if anyone else needs me to list my pin voltages for a working configuration I will. Its just a pain in the rear now that its integrated into other stuff. Although I may do it for my own reference and post it anyway.

closed 2018-09-06 by DreamingRaven

LouDnl, 2019-07-26:
I have the MH-ET LIVE (http://mh.nodebb.com/topic/2/mh-et-live-esp32-devkit).
It is impossible to program it through USB, haven't figured out why yet.
But your post made me hook up an FT232R to the boards pins as you describe and low and behold I can program it. Thanks a lot!!

Now to figure out why the USB program doesnt work.

> 
> 
> I just took out the resistor, and it works. ~_~
> [image]

DreamingRaven, 2019-07-26:
Nice one! I hope it all goes well! If I remember correctly I had issue with my program when I flashed it as well, although I'm afraid I cant remember how I overcame it as it was two years ago now, wow time flies. Good luck!

Daemach, 2019-07-26:
Might consider pulling up the EN pin with a 10K resistor.  Dev board also
uses a .1uf cap to ground.

LouDnl, 2019-08-05:
@DreamingRaven Yeah it took me a while to figure out what was wrong, but it works now. 
@Daemach I tried with a capacitor from EN to ground, haven't tried a resistor from EN to 3.3v. Might give that a go.

darkbull, 2019-08-20:
Connect to GPIO12 to GND save my problem. More info can found: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection