espressif/arduino-esp32 #492: Support for eMMC ?

trash-anger, 2017-07-08:
> Then we can also support MMC/eMMC cards via MMC protocol, if there is request for that.

Has it been implemented now ?
Else, when will it be, or still exist another library for that ?

Thank you very much ! :)

copercini, 2017-07-08:
Theoretically this both should works:
https://github.com/espressif/arduino-esp32/tree/master/libraries/SD
https://github.com/espressif/arduino-esp32/tree/master/libraries/SD_MMC

skateone, 2017-07-09:
just a note on the MMC protocol. i have ben playing with it for a while now and its for some reason corrupting my SD cards. am still debugging but thought i'de share.
Will post more once i find the cause

CogPy, 2017-10-10:
Is there any news on this issue? Especially I'm thinking of a timeline?

copercini, 2017-10-19:
There is a PR implementing this: https://github.com/espressif/esp-idf/pull/590

Soon it will be in Arduino too

ESP32DE, 2017-12-09:
i saw [james PR](https://github.com/espressif/esp-idf/pull/590) - does some one works on this?
x-mas comes - comes eMMC now?

me-no-dev, 2017-12-11:
seems development on the PR has been halted. maybe one of you can resume it?

trash-anger, 2019-08-05:
Up ?

lbernstone, 2019-08-05:
I don't have any hardware to test this, but there was a PR implemented upstream. https://github.com/espressif/esp-idf/pull/1941

closed 2019-10-19 by stale[bot]