espressif/arduino-esp32 #493: HTTPClient returns "connection refused" when using a static IP and a domain name

4lloyd, 2017-07-08:
I'm using HTTPClient to access a webpage via a URL with a domain name in it. When the ESP32 gets it's IP via DHCP it does work, when I set a static IP is doesn't work anymore and I get the error “connection refused”. I can 100% confirm the IP addresses are right, because I can reach the ESP32 via the network when using the static IP and if I set them in my PC they also work. I also tested the URL in a browser and that worked.

I tested with Wireshark. This showed that the DNS request is completed and the response is correct. After that nothing showed up anymore which suggest the ESP32 does not even try to connect to the server and just gives the error. If I change the domain name to the IP address of the server everything works fine. What could be the cause of this problem?

Example code which has the problem:
```c
#include <WiFi.h>
#include <HTTPClient.h>

const char* ssid = "SSID";
const char* password = "PASSWORD";

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.config(IPAddress(192, 168, 1, 223), IPAddress(192, 168, 1, 1), IPAddress(255, 255, 255, 0), IPAddress(192, 168, 1, 1)); //When commenting out this line it works
  WiFi.begin(ssid, password);
  //If I put WiFi.config here it also does not work
  while (WiFi.status() != WL_CONNECTED) {
    delay (500);
  }
  Serial.println(doGet());
}

void loop() {
}

int doGet() {
  Serial.println("Start doGet");
  HTTPClient http;
  http.begin("http://server.test/test.html"); //This has to be a url with a domain name in it, so no IP!
  int httpCode = http.GET();
  if (httpCode > 0) {
    if (httpCode == HTTP_CODE_OK) {
      String response = http.getString();
      Serial.println(response);
      return 0;
    } else {
      Serial.printf("Not ok, error: %s\n", http.errorToString(httpCode).c_str());
      http.end();
      return 2;
    }
  } else {
    Serial.printf("Wrong code, error: %s\n", http.errorToString(httpCode).c_str());
    http.end();
    return 1;
  }
}
```

Debug/serial print:
```
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 7 - STA_GOT_IP
Start doGet
[D][HTTPClient.cpp:138] beginInternal(): [HTTP-Client][begin] url: http://server.test/test.html
[D][HTTPClient.cpp:179] beginInternal(): [HTTP-Client][begin] host: server.test port: 80 url: /test.html
[D][HTTPClient.cpp:831] connect(): [HTTP-Client] failed connect to server.test:80
[D][HTTPClient.cpp:1109] returnError(): [HTTP-Client][returnError] error(-1): connection refused
Wrong code, error: connection refused
[D][HTTPClient.cpp:229] end(): [HTTP-Client][end] tcp is closed
1
```

copercini, 2017-07-09:
Looks like a DNS problem...
Try without set the DNS server
`WiFi.config(IPAddress(192, 168, 1, 223), IPAddress(192, 168, 1, 1), IPAddress(255, 255, 255, 0)`
or with the google DNS:
`WiFi.config(IPAddress(192, 168, 1, 223), IPAddress(192, 168, 1, 1), IPAddress(255, 255, 255, 0), IPAddress(8, 8, 8, 8));`

4lloyd, 2017-07-09:
Thank you for the reply.

I tried both. When no DNS server is configured no connection seems to work (tested with a URL with a domain name and one with a IP adress as host). With Google DNS the problems stays te same (only the URL with a domain name doesn't work).

I also first thought the problem is with DNS, but when I checked the connection with Wireshark I saw the DNS request and response with the right IP address. So that should work. The strange this was that when the DNS response came in directly after that the ESP32 gave the error.

everslick, 2018-04-19:
Is this still an issue for the original poster? Static IP works for other users, so I'm inclined to close this issue due to no recent activity.

trevor-sonic, 2019-03-17:
I have the same issue ;
````
    http.begin("http://10.0.0.151/");
    int httpCode = http.GET();

````
Return: connection refused with
````  
      if (WiFi.status() == WL_CONNECTED)
````

but works with,
````
    if ((WiFiMulti.run() == WL_CONNECTED)) 

````

So we can't use WiFiClient.h with static IP

Ratthanin, 2019-05-03:
also have the same issue.  Config IP make http request fail, but DHCP is fine.

//  if (!WiFi.config(localip, gateway, subnet)) {
//  Serial.println("STA Failed to configure");
//  }

/*connect to AP by list compare to given ssid */
 WiFi.begin(ssid[SSIDindex].c_str(), password[SSIDindex].c_str());
 /* string.c_str() function convert string to pchar */
 Serial.print("\r\nConnecting to ");
 Serial.println(ssid[SSIDindex]);

closed 2019-08-14 by stale[bot]

DM10, 2023-12-30:
it's still there...