espressif/arduino-esp32 #494: FreeRTOS error

mitchsf, 2017-07-08:
Hi All, 

I'm not using any FreeRTOS features in my application. It runs fine, but always crashes after about two minutes. Help is appreciated.  

Thanks in advance.

`Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x40084059  PS      : 0x00060034  A0      : 0x8008564b  A1      : 0x3ffc0590  
A2      : 0x3ffc1510  A3      : 0x00060021  A4      : 0x00060023  A5      : 0x00000020  
A6      : 0x00000020  A7      : 0x00060023  A8      : 0xb33f0001  A9      : 0xb33fffff  
A10     : 0x00060021  A11     : 0x00000000  A12     : 0x00060021  A13     : 0x11f27861  
A14     : 0x00000020  A15     : 0x00000002  SAR     : 0x00000020  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40084059:0x3ffc0590 0x4008564b:0x3ffc05b0 0x40083fb6:0x3ffc05d0 0x40086184:0x3ffc05f0 0x40081e7d:0x3ffc0600

CPU halted.
`
`ecoding 6 results
0x40084059: vPortCPUAcquireMutex at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./port.c line 334
0x40084059: vPortCPUAcquireMutex at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./port.c line 334
0x4008564b: xTaskIncrementTick at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4420
0x40083fb6: xPortSysTickHandler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./port.c line 275 (discriminator 1)
0x40086184: _frxt_timer_int at ?? line ?
0x40081e7d: _xt_lowint1 at xtensa_vectors.o line ?
`

mitchsf, 2017-07-10:
No ideas on this?

copercini, 2017-07-10:
> Interrupt wdt timeout on CPU0
> timerAttachInterrupt(timer, &processFrame, true);

My guess is that your function `processFrame()` is taking much time to run causing a wdt timeout.

mitchsf, 2017-07-10:
Thanks! I can eliminate 7 of 14 digital write statements, and 3 serial.print() statements used for debugging. Hopefully that will fix the problem.

copercini, 2017-07-10:
Another thing: `timerAlarmWrite(timer, 10000, true);` 
10ms is too fast, your project allow a bigger value for this?

mitchsf, 2017-07-10:
Is that really too fast? Others have discussed shorter ISR intervals. I can probably change to 30ms without impacting critical timing issues. I'll try that. 

I do enable and disable the ISR every minute, to be sure it starts at 0 seconds. I'm not sure if that matters.
`timerAlarmDisable(timer);
 timerAlarmEnable(timer);`

Thanks again.

forthlightning, 2017-07-11:
you may be able to fill a queue with pointers to your frames and process them by waiting on that queue. that way you dont have to process each one fully inside the isr, will require setting up some freertos machinery though.

mitchsf, 2017-07-11:
Thanks for the suggestions. The ISR has to run every 10ms, but I did remove some of the code leaving only three digitalWrites, and two variables to increment, both declared volatile. It has been running perfectly now for several hours.

closed 2017-07-11 by mitchsf