espressif/arduino-esp32 #495: Cannot run SD / SDMMC example on Olimex Esp32 EVB

pbecchi, 2017-07-09:
I have just downloaded last commit and I am trying to run SD example and / or SDMMC example on Olimex EVB board (I am unsure witch one is applicable).
Build is OK but I get "Card Mount Failed" message at run time.
Do I need to change any  #define for Olimex  EVB board?

williamesp2015, 2017-07-09:
I have Maple ESP32 R1 and I had your problem but II could use SD Card after initializing customized SPI Pins
SPI.begin(14,2,15,13);//SCK,MISO,MOSI,ss
delay(10);
if (!SD.begin(13, SPI, 40000000, "/sd")) {
Serial.println("Card Mount Failed");
return;
}

pbecchi, 2017-07-09:
Thanks but I have tried your suggested SPi configuration but it doesn't work....I think that Maple and Olimex EVB have different SD connections.
Since EVB board is supported I was thinking that SD example could run with no modifications.

melvinpmathew, 2017-07-12:
Did you get this to work?

pbecchi, 2017-07-13:
Finally with the help of Olimex this has been sorted out!
They have added one example onhttps://github.com/OLIMEX/ESP32-EVB/tree/master/SOFTWARE
This example is similar to SdMMc core example with following patch SD_MMC.cpp 
Modifiing line 45 46

  | host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
  | //host.flags = SDMMC_HOST_FLAG_1BIT; //use 1-line SD mode

To

host.max_freq_khz = SDMMC_FREQ_PROBING;
  | host.flags = SDMMC_HOST_FLAG_1BIT; //use 1-line SD 

Can this be done without changing library code?

fesch, 2017-07-16:
I also have the Olimex board and as far as I was able to test, only decommenting the "host.flags" line was enough to make the example run.

pbecchi, 2017-07-18:
You have to change both lines

melvinpmathew, 2017-07-18:
I got it to work only with  host.flags = SDMMC_HOST_FLAG_1BIT: The code is as follows:

    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
    host.flags = SDMMC_HOST_FLAG_1BIT; 

    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = false,
        .max_files = 5
    };

Staubgeborener, 2017-07-31:
i try to get this to work for over a month now. Still no success. I think my wiring is incorrect (especially the pull ups). Can someone please post a "real" picture of his wiring? @melvinpmathew ?
Everytime when my pull ups are connected to 3.3V i'll get error messages (waiting of rpacket handler). When in uncouple vcc from pull ups, i'll get the "normal" 

>  Failed to initialize the card (264). Make sure SD card lines have pull-up resistors in place.

message. So something has to be wrong (with my circuit)[http://imgur.com/a/VmgY8]. But like i said, a "real" picture would help me a lot.

fesch, 2017-07-31:
The "real picture" of my board (Olimex ESP32-EVB) would not help you much. But maybe the circuit will, doesn't it?

[image]

Staubgeborener, 2017-07-31:
Yeah, i also found this example. But thats no really help me. Is this only for 1 bit? Only IO15, IO2 and IO14 are connected. DAT1, DAT2 and DAT3  are only with 10k pull ups at 3.3V? Actual i'm a little deperate after weeks of a not well working sd card. So [here](http://i.imgur.com/pWyG6cq.png) you can see my setup. Not very beauty, but it shows perfectly my circuit. May you see at first attempt the issue.

fesch, 2017-07-31:
No. I suppose there are reference to other GPIO's. I also found this:
[image]

Staubgeborener, 2017-07-31:
Maybe my pinout is not the same in this case. I've got an esp32 from doit.am. Can not imagine it, but i just wrote doit.am to send me the pinout. I use everytime [this](https://raw.githubusercontent.com/gojimmypi/ESP32/master/images/myESP32%20DevKitC%20pinout.png) one and its also works everytime like a charm. But maybe not in this case...?

fesch, 2017-08-01:
Just here:
https://github.com/OLIMEX/ESP32-EVB/blob/master/HARDWARE/REV-B/ESP32-EVB_Rev_B.pdf

fesch, 2017-08-01:
You own this board, right?

https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESP32

me-no-dev, 2017-08-01:
Done ;)
```cpp

void mountSdMmcCard(){
    if(!SD_MMC.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD_MMC.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD_MMC card attached");
        return;
    }

    Serial.print("SD_MMC Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD_MMC.cardSize() / (1024 * 1024);

    Serial.printf("SD_MMC Card Size: %lluMB\n", cardSize);
}
```

me-no-dev, 2017-08-01:
well make sure you select the proper board from the menu

me-no-dev, 2017-08-01:
oh... and no SPI while using MMC... unfortunately Olimex did not attach SS pin for the card (looking at schematic), so we can not use it with SPI

closed 2017-08-01 by me-no-dev

Staubgeborener, 2017-08-01:
Well, it worked. I also have to add
``host.max_freq_khz = SDMMC_FREQ_PROBING;``
and also to connect IO0 to IO2. 1 line mode is working, 4 line mode not. It's a start.

melvinpmathew, 2017-08-02:
@Staubgeborener  I just saw this now. Were you able to make this work? The connection were exactly the same as you have shown, Just that I had a 1k pullup on my IO2 line(10k pullup should also not matter in this case), rest all were 10k pullups. Also only the 1 line mode works, the 4 line mode has some issues coming from the driver level and hence I don't think it worth investigating that now. Incase you find that its working share here too.Let me know if you still need help.

Staubgeborener, 2017-08-02:
Still only 1 line is working here. I have the issue, that i'll get a brownout error when using sdmmc example together with esp_start_wifi.

In my case: 10k pull ups for: D3, CMD, CLK, D0, D1 + Jumper between IO0 and IO2.
Normaly i shouldn't need to connect IO0 to IO2 for my board. But i have to - when i don't, the example won't work. Also tried to pull down IO2 while flashing and disconnect the jumper after/while flashing, also no success.

me-no-dev, 2017-08-02:
Guys I did no modification to my board! EVB board + sdcard + ethernet (did not test WiFi). Selected the appropriate board in the menu and gave it a go. All worked fine (with the latest commits to this repo). No modification or pin jumping was necessary.

pbecchi, 2017-08-02:
@me-no-dev
Can you tell us the examples or the code you have used?
Thanks

me-no-dev, 2017-08-02:
@pbecchi a few comments above: https://github.com/espressif/arduino-esp32/issues/495#issuecomment-319402745

pbecchi, 2017-08-02:
I know that there is a problem of bad contact between the holder and the SD card in the EVB board

me-no-dev, 2017-08-02:
I did not experience those (yet), but anyway :) issues like that are out of the scope of Arduino ;) should be handled by the manufacturer :)

pbecchi, 2017-08-02:
My EVB, after moving the SD , worked!

melvinpmathew, 2017-08-02:
@pbecchi  when you say worked, do you mean on the -line MMC mode? or just 1?

melvinpmathew, 2017-08-02:
@Staubgeborener  thats correct. I have IO2 pulled HIGH on normal operation with a 1k\10k pullup. And when we need to flash the firmware, IO2 is pulled LOW. Just to be clear, you still have it working on 1-line mode correct. Its only the 4-line modewhich is not working.

pbecchi, 2017-08-02:
Yes 1 linea MMC mode, no other way!

melvinpmathew, 2017-08-02:
@pbecchi  Ok thanks. 
@pbecchi @Staubgeborener We need to keep an eye on anyone who has 4-line MMC mode working :)

Staubgeborener, 2017-08-02:
@melvinpmathew It's weird on my setup. I have the whole time IO0 connected to IO2 to flash **AND** to run SDMMC (1 line). When i try to pull IO2 low when flashing and after that high its not working. So to be clear: I have to connect IO0 to IO2 and IO2 to SD (with 10k pull up). Sure, with this setup my blue led is always glowing, but no other setup worked here.

Staubgeborener, 2017-08-02:
It's a never ending "love"story. It's not working here anymore.Without doing anything (nothing at code nor circuit), i'll get brownout detection errors while IO0 is connected to IO2. I use alls of the guides i found (connect IO2 to GND while flashing and after flashing disconnect this jumper, etc) but i'll get only ``E (11518) sdmmc_cmd: sdmmc_card_init: send_scr returned 0x109``
It's so annoying.

melvinpmathew, 2017-08-02:
@Staubgeborener  The connection from IO2 goes to 3.3v with a jumber in between. I remove the jumper when flashing only. I have seen this error when connection is bad. Forgot what the exact issue was.  I know its apain to get it working. I had to rewire and make sure the connections were proper multiple times to finally have it working.

Staubgeborener, 2017-08-03:
@melvinpmathew Wait. With a jumper cable between IO2 and 3.3V you mean the pull up resistor between or truly just a jumper?

is esp_start_wifi() working in the Main of the sdmmc example on your side? I'll get a brownout error. Everytime i try to use wifi + sdmmc there is brownout.

Staubgeborener, 2017-08-03:
Can someone please try SDMMC + wifi + create a task on second core (for example with a while true loop and some calculations like increment an integer or so).
I will decompile the guru meditation error today to view what exactly happens here.

melvinpmathew, 2017-08-03:
@Staubgeborener  I mean a true jumper between IO2 and 3.3v. A pullup is just to achieve logic level!:

In electronic logic circuits, a pull-up resistor is a resistor connected between a signal conductor and a positive power supply voltage to ensure that the signal will be a valid logic level if external devices are disconnected or high-impedance is introduced.(Pull-up resistor - Wikipedia)

Staubgeborener, 2017-08-03:
@melvinpmathew So your circuit looks like this 
[image]

When flashing you disconnect the jumper from IO2 -> 3.3V and after flashing (befor ``make monitor``) you connect it? 

If your setup looks different, please edit this and show. Thanks.

fesch, 2017-08-03:
@Staubgeborener I've written a program that lists every 10 seconds the root of the SD card add publishes a message to my MQTT broker over WiFi. Using xTaskCreate, a second task is running, which increments and prints every second a counter. Add this runs fine!

My SDMMC config in the library is this one, so only 1-bit mode:
```
host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
host.flags = SDMMC_HOST_FLAG_1BIT;
```

Staubgeborener, 2017-08-04:
My new ESP32 arrived and SPI and 1 line works fine. I will test 4 line in the next hours. In the meantime:

Can someone please test [this](https://gist.github.com/Staubgeborener/83f06f83ef43e38112b0b6d8d50db3a8) Script from me? You don't need to put all of those periphals for that. Just flash the ESP with this, enable wifi and bluetooth (there is a bug, so bluetooth is necessary for promisc) and start this? it will first initialize your sd card in 1 line mode (that works) and after that enable wifi and start a thread on second core. And there i'll get 
``Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.``

It would help me a lot if someone could test this and share his experience!

Staubgeborener, 2017-08-05:
Nevermind, it's also working here now! Found out, that the writing intervals on my sd card where to short (some hundred times in a second, don't ask). This cause a ``guru meditation error``.

diskgokey, 2019-11-13:
much easier with latest update
use SD_MMC.begin("/sdcard", true) in setup step