espressif/arduino-esp32 #500: Setting static IP, NW & GW

fesch, 2017-07-11:
Hi

The ESP-IDF now contains a pull request which fixes the problem that static IP settings cannot be applied. I wanted to make the changes to my local Arduino-ESP sources, but can't find the corresponding source files. I only find the header files ... Can anyone point me where to find the sources?

Pull-request: https://github.com/espressif/esp-idf/pull/657/files#diff-6a4183dba634c596fdfa03b23c74a8d2L746
File to look for: tcpip_adapter_lwip.c
Method to modify: esp_err_t tcpip_adapter_dhcpc_stop(tcpip_adapter_if_t tcpip_if)

copercini, 2017-07-12:
Most of the Arduino Core is pre-compiled (to not take ages to run a new skecth), only [using as IDF component](https://github.com/espressif/arduino-esp32#using-as-esp-idf-component) or platform.io it will full compile again.

fesch, 2017-07-12:
OK, so I would need to pre-compile the modified esp-idf files and then copy them over to the esp32 hardware library right? 

Which leads to my next question: how to recompile esp-idf and where to copy what files. 

Until now, I did not find any documentation ... :-/

fesch, 2017-07-12:
OK, I found where the pre-compiled files are stored to. Now I need to compile the given tcpip-adapter component ...

fesch, 2017-07-12:
OK, found how to compile an ESP-IDF component and copy it over. So far so good. 

Next: check how to set a static IP ;)

fesch, 2017-07-16:
Modifying the "right" file helped a lot. The top-mentionned patch works fine!

closed 2017-07-16 by fesch