espressif/arduino-esp32 #501: Ethernet library not working for esp32

Nakul93, 2017-07-11:
I am getting this following errors:

In file included from C:\Program Files (x86)\Arduino\libraries\Ethernet\src\Dhcp.cpp:4:0:

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h: In static member function 'static void W5100Class::initSS()':

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:353:64: error: 'digitalPinToPort' was not declared in this scope

       *portModeRegister(digitalPinToPort(ETHERNET_SHIELD_SPI_CS)) |= digitalPinToBitMask(ETHERNET_SHIELD_SPI_CS);

                                                                ^

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:353:65: error: 'portModeRegister' was not declared in this scope

       *portModeRegister(digitalPinToPort(ETHERNET_SHIELD_SPI_CS)) |= digitalPinToBitMask(ETHERNET_SHIELD_SPI_CS);

                                                                 ^

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:353:112: error: 'digitalPinToBitMask' was not declared in this scope

       *portModeRegister(digitalPinToPort(ETHERNET_SHIELD_SPI_CS)) |= digitalPinToBitMask(ETHERNET_SHIELD_SPI_CS);

                                                                                                                ^

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h: In static member function 'static void W5100Class::setSS()':

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:356:66: error: 'digitalPinToPort' was not declared in this scope

       *portOutputRegister(digitalPinToPort(ETHERNET_SHIELD_SPI_CS)) &= ~digitalPinToBitMask(ETHERNET_SHIELD_SPI_CS);

                                                                  ^

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:356:67: error: 'portOutputRegister' was not declared in this scope

       *portOutputRegister(digitalPinToPort(ETHERNET_SHIELD_SPI_CS)) &= ~digitalPinToBitMask(ETHERNET_SHIELD_SPI_CS);

                                                                   ^

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:356:115: error: 'digitalPinToBitMask' was not declared in this scope

       *portOutputRegister(digitalPinToPort(ETHERNET_SHIELD_SPI_CS)) &= ~digitalPinToBitMask(ETHERNET_SHIELD_SPI_CS);

                                                                                                                   ^

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h: In static member function 'static void W5100Class::resetSS()':

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:359:66: error: 'digitalPinToPort' was not declared in this scope

       *portOutputRegister(digitalPinToPort(ETHERNET_SHIELD_SPI_CS)) |= digitalPinToBitMask(ETHERNET_SHIELD_SPI_CS);

                                                                  ^

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:359:67: error: 'portOutputRegister' was not declared in this scope

       *portOutputRegister(digitalPinToPort(ETHERNET_SHIELD_SPI_CS)) |= digitalPinToBitMask(ETHERNET_SHIELD_SPI_CS);

                                                                   ^

C:\Program Files (x86)\Arduino\libraries\Ethernet\src\utility/w5100.h:359:114: error: 'digitalPinToBitMask' was not declared in this scope

       *portOutputRegister(digitalPinToPort(ETHERNET_SHIELD_SPI_CS)) |= digitalPinToBitMask(ETHERNET_SHIELD_SPI_CS);

Do help me out

fesch, 2017-07-16:
I think that the stock Arduino Ethernet Library will not work for ESP32.

I've coded a new one, which can be found here: http://www.fisch.lu/junk/ESP32-Ethernet.zip

I has been written to work with the Olimex ESP32-EVB development board, which uses the LAN8720 chip.

You need to copy the library into the "libraries" folder of your expressif/esp32 installation folder. Do **not** copy to the regular libraries folder of Arduino!

What Ethernet chip do you use?

Nakul93, 2017-07-16:
Thanks
Do you have an example code for arduino.

fesch, 2017-07-16:
Of course. Of integrated some into the package:

http://www.fisch.lu/junk/ESP32-Ethernet.zip

Nakul93, 2017-07-17:
I am unable to use EthernetServer function. I have tried including the libraries associated with it from arduino lib but getting errors. Is there any custom function for EthernetServer and EthernetClient.
TIA

fesch, 2017-07-17:
I did not implement those. Maybe @me-no-dev knows more about this?

me-no-dev, 2017-07-17:
there will be no such thing ;) Ethernet is connected to LwIP so it functions the same as WiFi. You open a client/server and you are good to go. (WiFi Client/Server in this case)

Nakul93, 2017-07-17:
Ok thanks...and one more thing do we have to do those tweaks on LAN8720 board as described here https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

fesch, 2017-07-17:
I didn't have to in the library I wrote ...

Nakul93, 2017-07-17:
Its not working ...In the serial monitor it just shows the basic chip info

Nakul93, 2017-07-18:
@me-no-dev did you made any tweaks in the connection for the LAN8720 board

WALLTECH, 2017-07-26:
@fesch From the examples it seems like the Ethernet library extends the functionality of the base wifi library, would this mean that anything written for wifi would work for ethernet too simply added? If so this is amazing and very much appreciated.

fesch, 2017-07-26:
@WALLTECH That's a good question. I did not try this, but yes, I think that all stuff from within the application layer of wifi may work. In fact, only the underlying network laying is different, but the service calls should be the same.

But I have to admit, that I'm not too deep into all this to really be able to confirm.

WALLTECH, 2017-07-26:
@fesch Awesome. To test this, would I attach the hardware, then add from the examples to include ethernet and start it, then try something that's usually wifi and see if it works? Would I need to add Ethernet to a library that uses wifi to use that? I'll test this

fesch, 2017-07-26:
No idea ... but I will give it a try tomorrow ... if I find the time to ;)

fesch, 2017-07-28:
I yesterday successfully used my Ethernet library with the WiFiClient as well as with PubSubClient in order to connect to an MQTT broker. As said, this worked quite out of the box.

BUT ... using the secured version, thus WiFiClientSecure, on port 8883 did not work and resulted in some nice core panic :(

```
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered 
Register dump:
PC      : 0x40139501  PS      : 0x00060136  A0      : 0x800fb809  A1      : 0x3ffd72b0  
A2      : 0x3ffd72e0  A3      : 0x3ffd73e2  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000080  A7      : 0x00000080  A8      : 0x00000087  A9      : 0x3ffd7290  
A10     : 0x00000002  A11     : 0x3ffd72f0  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000010  A15     : 0x3ffd7330  SAR     : 0x00000018  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x40139501:0x3ffd72b0 0x400fb809:0x3ffd72e0 0x400ef784:0x3ffd73e0 0x400ef7fc:0x3ffd7450 0x400ef88f:0x3ffd7500 0x400ef5da:0x3ffd7560 0x400ef66c:0x3ffd7700 0x400ef68d:0x3ffd7750 0x400d2628:0x3ffd7770 0x400d2428:0x3ffd79d0 0x400d2498:0x3ffd79f0 0x400d24b6:0x3ffd7a20 0x400d2a94:0x3ffd7a40 0x400d2c0d:0x3ffd7a80 0x400d18e4:0x3ffd7ab0 0x400d19b5:0x3ffd7ad0 0x400d2044:0x3ffd7b00 0x400db7a8:0x3ffd7b20 0x400db7d0:0x3ffd7b40

CPU halted.
```

fesch, 2017-07-30:
I'm soooo stupid!!!

In my example I was calling the connect() method in the asynchronously called WiFiEvent callback, which can't be handled by the WiFiClientSecure. So by leaving the connection code inside the main loop, the use of the secured MQTT version also works!

rammichael1, 2017-10-18:
hey here i am using USR TS232T2  i am trying to connect with my esp32 but i am very confused how to do please help me with the code and connections

me-no-dev, 2017-10-18:
uhmm we are not here to help you with that :) we are here to help with bugs with the code. What you show above is serial to ethernet thing... you get UART to Serial adapter, connect the UART side to ESP32, start Serial there and talk to the UART.
Better yet, get an ESP32 board with Ethernet and do not go through this.

closed 2017-10-18 by me-no-dev

rammichael1, 2017-10-20:
please give me examples so that i can understand

danyg, 2018-03-22:
@fesch Could you expose the ESP32 - Ethernet library on github so can be added to the platform.io repository and can be easily added to a esp32 project with arduino
Thanks!

fesch, 2018-03-22:
I taught @me-no-dev  already integrated something?

dvvrao, 2018-08-07:
Hi fesch,

Can you please give the example code that uses "PubSubClient" that uses this Ethernet library.
I do not how to  use this Ethernet library with "PubSubClient".

Thanks & Regards,
Vshnuvardhan

zekageri, 2018-09-04:
Hello guys!
I have a problem with esp-Wroom-32 with ETH_Lan8720 board. I'am using an arduino example code from here :

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/ETH_LAN8720/ETH_LAN8720.ino

and i also tryed with your librari Fesch.

https://github.com/fesch/arduino-esp32/tree/master/libraries/Ethernet/examples

I did the wireing like this:

GPIO17 - PHY_POWER : NC - Osc. Enable - 4k7 Pulldown
GPIO22 - EMAC_TXD1 : TX1
GPIO19 - EMAC_TXD0 : TX0
GPIO21 - EMAC_TX_EN : TX_EN
GPIO26 - EMAC_RXD1 : RX1
GPIO25 - EMAC_RXD0 : RX0
GPIO27 - EMAC_RX_DV : CRS
GPIO00 - EMAC_TX_CLK : nINT/REFCLK (50MHz) - 4k7 Pullup
GPIO23 - SMI_MDC : MDC
GPIO18 - SMI_MDIO : MDIO
GND : GND
3V3 : VCC

from the site :
https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

I soldered the NC PIN on the lan8720 board to the enable pin of the oscillator. I added pull down resistor to nc pin and pull up to gpio00.

LAN 8720 board : https://fr.aliexpress.com/item/Smart-Electronics-LAN8720-module-network-module-Ethernet-transceiver-RMII-interface-development-board-for-arduino/32825173408.html?spm=a2g0s.9042311.0.0.40696c370I7nRH

ESP32: https://fr.aliexpress.com/item/ESP-WROOM-32-WiFi-Bluetooth-4-2-dual-core-CPU-MCU-low-power-Bluetooth-240MHZ-ESP32/32829367382.html?spm=a2g0s.9042311.0.0.15636c37H4la82

I get the following errors:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
E (2256) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
E (3256) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000
E (4256) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000

I dont know what to do , can anyone help me?

sauttefk, 2018-09-04:
@zekageri You are hijacking this issue ticket.
Besides that:
* How long is your wiring? These are 50MHz signals that will most probably **not work** on breadbords and long cables.
* Did you configure the correct PHY-ID in your software?
* Have you checked the that the 50MHz clock is being turned on?
* Send us a picture of your hardware setup, so we might spot the error.

zekageri, 2018-09-04:
First of all thank you for the answer. And sorry for hijacking. I dont know where else should i put this issue.

My cables are long A.F and iam using a breadboard.
I think i configure the PHY-ID right. I defined it like this: #define ETH_PHY_ADDR 1 in my sketch.
I checked the clock. Its turned on.
I try to send a picture from it after i shorten the cables

zekageri, 2018-09-04:
[image]
[image]
I shortened the cables. The Serial output still the same. Timed out. I use the pin defines in the sketch and the PHY address define too like this:

#define ETH_PHY_ADDR 1
#define CONFIG_PHY_LAN8720
#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN
#define ETH_POWER_PIN   17
#define ETH_TYPE        ETH_PHY_LAN8720
#define ETH_ADDR        1 // 0
#define ETH_MDC_PIN     23
#define ETH_MDIO_PIN    18

zorce, 2019-07-15:
> I'm soooo stupid!!!
> 
> In my example I was calling the connect() method in the asynchronously called WiFiEvent callback, which can't be handled by the WiFiClientSecure. So by leaving the connection code inside the main loop, the use of the secured MQTT version also works!

Hi Fesch,
Do you happen to have an example code to get the esp32-poe to work with mqtt?
Have managed to run the example https://github.com/OLIMEX/ESP32-POE/tree/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_Arduino
But when trying to add MQTT via PubSubClient to it I fail as I need a reference to WiFi to get in running.
Have also tried with EspMQTTClient library without any luck as it also depend on WiFi.

fesch, 2019-07-16:
No, I never tried the PoE as my board has none ... sorry :-/

KVAR-NEW-DEV, 2024-05-10:
I have being try to set get request to my nodejs Server. On the upload first time the connection with the server is extablished, later if reboted the server fails to connect 

```
#include <EthernetENC.h>
#include "esp_system.h"
#include <Update.h>
#include <EthernetENC.h>
#include <SPI.h>
#include <EthernetClient.h>

uint8_t mac[6] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
IPAddress ip = {192,168,131,183};
IPAddress bc = {255,255,255,255};
IPAddress subnet = {255,0,0,0};
uint8_t myDns[] = {192,168,131,1};
char buf[100];

char serverAddress[] = "60.254.18.10"; // Server IP address
int serverPort = 5000;           // Server port
char temp[]="5000";
char path[] = "/?AUTH=adwdscsdc342423423rw@$@";
char host[] = "192.168.131.100:5000";

void setup(){
  Serial.begin(115200);
  Serial.println("Hello Esp32");
  Ethernet.init(2);
  Ethernet.begin(mac, ip, myDns, myDns, subnet);
  int checkLink = Ethernet.linkStatus();
  int checkHwd = Ethernet.hardwareStatus();

    //Cabel & Hardware Check
    while (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("ENC: not found");
      delay(1000);
    }

    while (Ethernet.linkStatus() != LinkON){
       Serial.println("ENC: cable not found");
       delay(1000);
    }
    Serial.println(Ethernet.localIP());
    GET(serverAddress,serverPort,"/checkOtaM/KVAR-1");
    
  
}
int count =0;
void loop(){
  Ethernet.maintain();
  delay(1000);
  if(count==5){
    ESP.restart();
  }else{
    count = count +1;
  }
}

void GET(const char* host, int port, const char* path){

  Serial.println("Check Status : Init");
  EthernetClient ethClient2;
  if (!ethClient2.connected()) { // Check if ethClient2 is not connected
    //ethClient2.flush();
    ethClient2.stop(); // Disconnect any previous connection
  }
   if (ethClient2.connect(host, port)) {
    Serial.println("Connected to server");

    // Send HTTP request
    ethClient2.print("GET ");
[... 116 lines cut ...]
10:59:48.486 -> load:0x3fff0030,len:1184
10:59:48.486 -> load:0x40078000,len:13220
10:59:48.486 -> ho 0 tail 12 room 4
10:59:48.486 -> load:0x40080400,len:3028
10:59:48.486 -> entry 0x400805e4
10:59:48.672 -> Hello Esp32
10:59:48.719 -> ENC: cable not found
10:59:49.687 -> 192.168.131.183
10:59:49.687 -> Check Status : Init
11:00:00.726 -> ets Jun  8 2016 00:22:57
11:00:00.726 -> 
11:00:00.726 -> rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
11:00:00.726 -> configsip: 0, SPIWP:0xee
11:00:00.726 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:00:00.726 -> mode:DIO, clock div:1
11:00:00.726 -> load:0x3fff0030,len:1184
11:00:00.726 -> load:0x40078000,len:13220
11:00:00.726 -> ho 0 tail 12 room 4
11:00:00.726 -> load:0x40080400,len:3028
11:00:00.726 -> entry 0x400805e4
11:00:00.912 -> Hello Esp32
11:00:00.960 -> ENC: cable not found
11:00:01.941 -> 192.168.131.183
11:00:01.941 -> Check Status : Init
"